nextcloud/nextcloudpi #709: Update hangs. update now impossible through the webUI

hueppinr, 2018-11-05:
Previous Version 0.60.3 running fine for weeks, nextcloud 13.0.6

I wanted to update today to a new version as I have seen a new build for download was released yesterday.

It just hangs after some time . please see below output from the webfrontend

_______________________________________________
[ nc-update ]
Downloading updates
Performing updates
Installing nc-hdd-monitor
Installing nc-hdd-test
Get:1 http://security.debian.org stretch/updates InRelease [94.3 kB]
Ign:3 http://cdn-fastly.deb.debian.org/debian stretch InRelease
Get:4 http://cdn-fastly.deb.debian.org/debian stretch-updates InRelease [91.0 kB]
Get:5 http://security.debian.org stretch/updates/main arm64 Packages [542 kB]
Get:6 http://cdn-fastly.deb.debian.org/debian stretch-backports InRelease [91.8 kB]
Get:7 http://security.debian.org stretch/updates/main armhf Packages [548 kB]
Get:8 http://security.debian.org stretch/updates/contrib armhf Packages [1761 B]
Get:9 http://security.debian.org stretch/updates/contrib arm64 Packages [1762 B]
Get:10 http://security.debian.org stretch/updates/non-free arm64 Packages [1208 B]
Get:11 http://security.debian.org stretch/updates/non-free armhf Packages [1208 B]
Get:12 http://cdn-fastly.deb.debian.org/debian stretch Release [118 kB]
Get:13 http://cdn-fastly.deb.debian.org/debian stretch Release.gpg [2434 B]
Get:2 https://apt.armbian.com stretch InRelease [19.0 kB]
Get:14 http://cdn-fastly.deb.debian.org/debian stretch-updates/main armhf Packages [5432 B]
Get:15 http://cdn-fastly.deb.debian.org/debian stretch-updates/main arm64 Packages [5479 B]
Get:16 http://cdn-fastly.deb.debian.org/debian stretch-updates/main armhf Contents (deb) [19.5 kB]
Get:17 http://cdn-fastly.deb.debian.org/debian stretch-updates/main arm64 Contents (deb) [19.5 kB]
Get:18 http://cdn-fastly.deb.debian.org/debian stretch-backports/main arm64 Packages [619 kB]
Get:19 http://cdn-fastly.deb.debian.org/debian stretch-backports/main armhf Packages [611 kB]
Get:20 http://cdn-fastly.deb.debian.org/debian stretch-backports/main arm64 Contents (deb) [5468 kB]
Get:21 http://cdn-fastly.deb.debian.org/debian stretch-backports/main armhf Contents (deb) [5701 kB]
Get:22 http://cdn-fastly.deb.debian.org/debian stretch-backports/contrib arm64 Packages [4714 B]
Get:23 http://cdn-fastly.deb.debian.org/debian stretch-backports/contrib armhf Packages [6601 B]
Get:24 http://cdn-fastly.deb.debian.org/debian stretch-backports/contrib armhf Contents (deb) [19.9 kB]
Get:25 http://cdn-fastly.deb.debian.org/debian stretch-backports/contrib arm64 Contents (deb) [19.4 kB]
Get:26 http://cdn-fastly.deb.debian.org/debian stretch-backports/non-free arm64 Packages [6977 B]
Get:27 http://cdn-fastly.deb.debian.org/debian stretch-backports/non-free armhf Packages [6977 B]
Get:28 http://cdn-fastly.deb.debian.org/debian stretch-backports/non-free arm64 Contents (deb) [54.1 kB]
Get:29 http://cdn-fastly.deb.debian.org/debian stretch-backports/non-free armhf Contents (deb) [54.1 kB]
Get:30 http://cdn-fastly.deb.debian.org/debian stretch/main armhf Packages [9277 kB]
Get:31 http://cdn-fastly.deb.debian.org/debian stretch/main arm64 Packages [9287 kB]
Get:32 https://apt.armbian.com stretch/main arm64 Packages [139 kB]
Get:33 https://apt.armbian.com stretch/main armhf Packages [287 kB]
Get:34 https://apt.armbian.com stretch/stretch-utils arm64 Packages [4903 B]
Get:35 https://apt.armbian.com stretch/stretch-utils armhf Packages [4890 B]
Get:36 https://apt.armbian.com stretch/stretch-desktop armhf Packages [1440 B]
Get:37 http://cdn-fastly.deb.debian.org/debian stretch/main arm64 Contents (deb) [31.0 MB]
Get:38 http://cdn-fastly.deb.debian.org/debian stretch/main armhf Contents (deb) [31.0 MB]
Get:39 http://cdn-fastly.deb.debian.org/debian stretch/contrib armhf Packages [49.6 kB]
Get:40 http://cdn-fastly.deb.debian.org/debian stretch/contrib arm64 Packages [47.2 kB]
Get:41 http://cdn-fastly.deb.debian.org/debian stretch/contrib arm64 Contents (deb) [77.3 kB]
Get:42 http://cdn-fastly.deb.debian.org/debian stretch/contrib armhf Contents (deb) [78.0 kB]
Get:43 http://cdn-fastly.deb.debian.org/debian stretch/non-free armhf Packages [71.2 kB]
Get:44 http://cdn-fastly.deb.debian.org/debian stretch/non-free arm64 Packages [59.8 kB]
Get:45 http://cdn-fastly.deb.debian.org/debian stretch/non-free arm64 Contents (deb) [732 kB]
Get:46 http://cdn-fastly.deb.debian.org/debian stretch/non-free armhf Contents (deb) [735 kB]
Fetched 97.0 MB in 53s (1811 kB/s)
Reading package lists...

Reading package lists...
Building dependency tree...

Reading state information...
Suggested packages:
gsmartcontrol smart-notifier
Recommended packages:
mailx | mailutils
The following NEW packages will be installed:
[... 429 lines cut ...]
Setting up python3.5-minimal (3.5.3-1+deb9u1) ...
Setting up php7.0-zip (7.0.32-2+0~20181015120718.6+stretch~1.gbpa6b8cf) ...
debconf: unable to initialize frontend: Dialog
debconf: (Dialog frontend will not work on a dumb terminal, an emacs shell buffer, or without a controlling terminal.)
debconf: falling back to frontend: Readline
Setting up libmagickcore-6.q16-3-extra:arm64 (8:6.9.7.4+dfsg-11+deb9u6) ...
Setting up php7.0-json (7.0.32-2+0~20181015120718.6+stretch~1.gbpa6b8cf) ...
debconf: unable to initialize frontend: Dialog
debconf: (Dialog frontend will not work on a dumb terminal, an emacs shell buffer, or without a controlling terminal.)
debconf: falling back to frontend: Readline
Setting up libpython2.7:arm64 (2.7.13-2+deb9u3) ...
Setting up python3.5 (3.5.3-1+deb9u1) ...
Setting up php7.0-cli (7.0.32-2+0~20181015120718.6+stretch~1.gbpa6b8cf) ...
debconf: unable to initialize frontend: Dialog
debconf: (Dialog frontend will not work on a dumb terminal, an emacs shell buffer, or without a controlling terminal.)
debconf: falling back to frontend: Readline
Setting up php7.0-fpm (7.0.32-2+0~20181015120718.6+stretch~1.gbpa6b8cf) ...
debconf: unable to initialize frontend: Dialog
debconf: (Dialog frontend will not work on a dumb terminal, an emacs shell buffer, or without a controlling terminal.)
debconf: falling back to frontend: Readline
NOTICE: Not enabling PHP 7.0 FPM by default.
NOTICE: To enable PHP 7.0 FPM in Apache2 do:
NOTICE: a2enmod proxy_fcgi setenvif
NOTICE: a2enconf php7.0-fpm
NOTICE: You are seeing this message because you have apache2 package installed.

ovpc, 2018-11-05:
Try running ncp-update from cli (again). 
Current version is v0.66.1 for NCP and 14.0.3 for NC, It seems you did not have autoupdate enabled yet. It gets enabled by default as from v0.62 or so.
Then run nc-update-nextcloud from either ncp-config or ncp-web.

nachoparker, 2018-11-06:
It seems that you haven't updated in more than a month. You should enable autoupdates, or remember to update every week or two. We cannot support outdated stuff by more than a month, as this is a rolling release style of system

closed 2018-11-20 by hueppinr