nextcloud/nextcloudpi #710: nc-hdd-test not displaying results

zedocrob, 2018-11-06:
After doing a nc-hdd-test "short test", i can't see the results in the windows.

(Should i remove my previously installed Smartmontools or can i keep it alongside the new nc-hdd-test fonction without problems?)

[image]

**System information**

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v0.66.1
NextCloudPi image    NextCloudPi_09-29-18
distribution         Debian GNU/Linux 9 \n \l
automount            yes
USB devices          sda 
datadir              /media/DATAcloud/ncdata
data in SD           no
data filesystem      btrfs
data disk usage      346G/3.7T
rootfs usage         3.9G/30G
swapfile             /var/swap
dbdir                /var/lib/mysql
Nextcloud check      ok
Nextcloud version    14.0.3.0
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            enx001e0632be42
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               32days

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "5": "nextcloudpi.local",
            "1": "192.168.0.9",
            "3": "robserver.mycloud.bz",
            "4": "robserver.mycloud.bz",
            "7": "nextcloudpi",
            "8": "nextcloudpi.lan"
        },
        "datadirectory": "***REMOVED SENSITIVE VALUE***",
        "dbtype": "mysql",
        "version": "14.0.3.0",
        "overwrite.cli.url": "https:\/\/robserver.mycloud.bz\/",
        "dbname": "***REMOVED SENSITIVE VALUE***",
        "dbhost": "***REMOVED SENSITIVE VALUE***",
        "dbport": "",
[... 54 lines cut ...]
```

{"reqId":"lO3rnwFVxHUliZkGwmrO","level":1,"time":"2018-10-16T06:28:38+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Repair::step: Repair step: Fix potential broken mount points","userAgent":"--","version":"14.0.1.1"}
{"reqId":"lO3rnwFVxHUliZkGwmrO","level":1,"time":"2018-10-16T06:28:38+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Repair::info: Repair info: No mounts updated","userAgent":"--","version":"14.0.1.1"}
{"reqId":"lO3rnwFVxHUliZkGwmrO","level":1,"time":"2018-10-16T06:28:38+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Repair::step: Repair step: Repair invalid paths in file cache","userAgent":"--","version":"14.0.1.1"}
{"reqId":"lO3rnwFVxHUliZkGwmrO","level":1,"time":"2018-10-16T06:28:38+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Repair::step: Repair step: Add log rotate job","userAgent":"--","version":"14.0.1.1"}
{"reqId":"lO3rnwFVxHUliZkGwmrO","level":1,"time":"2018-10-16T06:28:38+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Repair::step: Repair step: Clear frontend caches","userAgent":"--","version":"14.0.1.1"}
{"reqId":"lO3rnwFVxHUliZkGwmrO","level":1,"time":"2018-10-16T06:28:38+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Repair::info: Repair info: Image cache cleared","userAgent":"--","version":"14.0.1.1"}
{"reqId":"lO3rnwFVxHUliZkGwmrO","level":1,"time":"2018-10-16T06:28:50+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Repair::info: Repair info: SCSS cache cleared","userAgent":"--","version":"14.0.1.1"}
{"reqId":"lO3rnwFVxHUliZkGwmrO","level":1,"time":"2018-10-16T06:28:57+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Repair::info: Repair info: JS cache cleared","userAgent":"--","version":"14.0.1.1"}
{"reqId":"lO3rnwFVxHUliZkGwmrO","level":1,"time":"2018-10-16T06:28:57+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Repair::step: Repair step: Add preview background cleanup job","userAgent":"--","version":"14.0.1.1"}
{"reqId":"lO3rnwFVxHUliZkGwmrO","level":1,"time":"2018-10-16T06:28:57+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Repair::step: Repair step: Queue a one-time job to cleanup old backups of the updater","userAgent":"--","version":"14.0.1.1"}
{"reqId":"lO3rnwFVxHUliZkGwmrO","level":1,"time":"2018-10-16T06:28:57+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Repair::step: Repair step: Repair pending cron jobs","userAgent":"--","version":"14.0.1.1"}
{"reqId":"lO3rnwFVxHUliZkGwmrO","level":1,"time":"2018-10-16T06:28:57+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Repair::info: Repair info: No need to repair pending cron jobs.","userAgent":"--","version":"14.0.1.1"}
{"reqId":"lO3rnwFVxHUliZkGwmrO","level":1,"time":"2018-10-16T06:28:57+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Updater::startCheckCodeIntegrity: Starting code integrity check...","userAgent":"--","version":"14.0.1.1"}
{"reqId":"lO3rnwFVxHUliZkGwmrO","level":1,"time":"2018-10-16T06:29:09+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Updater::finishedCheckCodeIntegrity: Finished code integrity check","userAgent":"--","version":"14.0.1.1"}
{"reqId":"lO3rnwFVxHUliZkGwmrO","level":1,"time":"2018-10-16T06:29:09+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Updater::updateEnd: Update successful","userAgent":"--","version":"14.0.3.0"}
{"reqId":"lO3rnwFVxHUliZkGwmrO","level":1,"time":"2018-10-16T06:29:09+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Updater::maintenanceDisabled: Turned off maintenance mode","userAgent":"--","version":"14.0.3.0"}
{"reqId":"lO3rnwFVxHUliZkGwmrO","level":1,"time":"2018-10-16T06:29:09+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Updater::resetLogLevel: Reset log level to Warning(2)","userAgent":"--","version":"14.0.3.0"}
{"reqId":"W8jYgn8AAQEAAEU5aNMAAABM","level":3,"time":"2018-10-18T19:01:22+00:00","remoteAddr":"185.216.35.67","user":"--","app":"index","method":"GET","url":"\/wp\/wp-login.php","message":{"Exception":"Doctrine\\DBAL\\Exception\\UniqueConstraintViolationException","Message":"An exception occurred while executing 'INSERT INTO `oc_filecache` (`mimepart`,`mimetype`,`mtime`,`size`,`etag`,`storage_mtime`,`permissions`,`parent`,`checksum`,`path_hash`,`path`,`name`,`storage`) SELECT ?,?,?,?,?,?,?,?,?,?,?,?,? FROM `oc_filecache` WHERE `storage` = ? AND `path_hash` = ? HAVING COUNT(*) = 0' with params [\"8\", \"17\", 1539889282, 0, \"547f548ea5ec82f4ecb9d38c1a2dfa33\", 1539889282, 27, 95, \"\", \"98e45a68e12949696cb911da754c2326\", \"appdata_oc65v9pdw1g0\\\/css\\\/core\\\/cd70-be2e-jquery.ocdialog.css\", \"cd70-be2e-jquery.ocdialog.css\", 3, 3, \"98e45a68e12949696cb911da754c2326\"]:\n\nSQLSTATE[23000]: Integrity constraint violation: 1062 Duplicate entry '3-98e45a68e12949696cb911da754c2326' for key 'fs_storage_path_hash'","Code":0,"Trace":[{"file":"\/var\/www\/nextcloud\/3rdparty\/doctrine\/dbal\/lib\/Doctrine\/DBAL\/DBALException.php","line":128,"function":"convertException","class":"Doctrine\\DBAL\\Driver\\AbstractMySQLDriver","type":"->","args":["An exception occurred while executing 'INSERT INTO `oc_filecache` (`mimepart`,`mimetype`,`mtime`,`size`,`etag`,`storage_mtime`,`permissions`,`parent`,`checksum`,`path_hash`,`path`,`name`,`storage`) SELECT ?,?,?,?,?,?,?,?,?,?,?,?,? FROM `oc_filecache` WHERE `storage` = ? AND `path_hash` = ? HAVING COUNT(*) = 0' with params [\"8\", \"17\", 1539889282, 0, \"547f548ea5ec82f4ecb9d38c1a2dfa33\", 1539889282, 27, 95, \"\", \"98e45a68e12949696cb911da754c2326\", \"appdata_oc65v9pdw1g0\\\/css\\\/core\\\/cd70-be2e-jquery.ocdialog.css\", \"cd70-be2e-jquery.ocdialog.css\", 3, 3, \"98e45a68e12949696cb911da754c2326\"]:\n\nSQLSTATE[23000]: Integrity constraint violation: 1062 Duplicate entry '3-98e45a68e12949696cb911da754c2326' for key 'fs_storage_path_hash'",{"errorInfo":["23000",1062,"Duplicate entry '3-98e45a68e12949696cb911da754c2326' for key 'fs_storage_path_has [... cut ...]
{"reqId":"W8@o1X8AAQEAAEU5cCAAAEIk","level":2,"time":"2018-10-23T23:03:49+00:00","remoteAddr":"192.168.0.254","user":"--","app":"core","method":"PROPFIND","url":"\/remote.php\/webdav\/","message":"Login failed: 'ncp' (Remote IP: '192.168.0.254')","userAgent":"Transmit\/5.2","version":"14.0.3.0"}
{"reqId":"W9PUuX8AAQEAAEU5fC0AAABG","level":2,"time":"2018-10-27T03:00:10+00:00","remoteAddr":"172.91.117.94","user":"--","app":"core","method":"PROPFIND","url":"\/remote.php\/webdav\/","message":"Login failed: 'Vincent' (Remote IP: '172.91.117.94')","userAgent":"Mozilla\/5.0 (Macintosh) mirall\/2.3.3 (build 84) (Nextcloud)","version":"14.0.3.0"}

```
</details>

nachoparker, 2018-11-07:
Hi, 

Does it work for you if you run it through `ncp-config`?

if you had installed smartmontools already, then it was already there. We only install it through apt

nachoparker, 2018-11-07:
If it doesn't work without `-d sat`, then smartmontools is not able to recognize your device type as `SAT`.

zedocrob, 2018-11-07:
Yes,  i need `-d sat` argument for Odroid HC2 (USB to Sata Bridge) with `smartctl`.
Means that `nc-hdd-test` and `nc-hdd-monitor` is useless for Odroid HC2?
So i should stick with terminal Smartmontools command with `-d sat` to check the HDD smart?

nachoparker, 2018-11-08:
if it doesn't recognize it automatically, then it will have to be adapted somehow

nachoparker, 2019-01-15:
fixed, thanks for reporting

closed 2019-01-15 by nachoparker