nextcloud/nextcloudpi #712: Internal Server Error

Gabinator9, 2018-11-12:
Ive tried logging in to the web UI multiple times after re etchering my ncp image to my SD card and I cannot get it to work, I login and it just gives me [This](https://imgur.com/a/F6CCFqL) Error, I have not changed anything all 3 times on my pi.

**System information**

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v0.64.12
NextCloudPi image    NextCloudPi_11-04-18
distribution         Raspbian GNU/Linux 9 \n \l
automount            no
USB devices          none
datadir              /var/www/nextcloud/data
data in SD           yes
data filesystem      ext2/ext3
data disk usage      1.8G/59G
rootfs usage         1.8G/59G
swapfile             /var/swap
dbdir                /var/lib/mysql
Nextcloud check      error
HTTPD service        up
PHP service          down
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        closed
port check 443       closed
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               9min

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

/usr/local/bin/ncp-report: line 48:  4588 Illegal instruction     sudo -u www-data php /var/www/nextcloud/occ config:list system

```
</details>

<details>
<summary>HTTPd logs</summary>

```

[Mon Nov 12 04:52:00.889350 2018] [authz_host:error] [pid 2750:tid 2755679280] [client fe80::e006:de0f:22e9:4ecc:54707] AH01753: access check of 'localhost' to /favicon.ico failed, reason: unable to get the remote host name
[Mon Nov 12 04:52:00.889611 2018] [authz_core:error] [pid 2750:tid 2755679280] [client fe80::e006:de0f:22e9:4ecc:54707] AH01630: client denied by server configuration: /var/www/ncp-web/favicon.ico
[Mon Nov 12 04:52:08.645251 2018] [authz_host:error] [pid 2749:tid 2755679280] [client fe80::e006:de0f:22e9:4ecc:54732] AH01753: access check of 'localhost' to / failed, reason: unable to get the remote host name, referer: https://github.com/nextcloud/nextcloudpi/wiki/How-to-access-NextCloudPi
[Mon Nov 12 04:52:08.645651 2018] [authz_core:error] [pid 2749:tid 2755679280] [client fe80::e006:de0f:22e9:4ecc:54732] AH01630: client denied by server configuration: /var/www/ncp-web/, referer: https://github.com/nextcloud/nextcloudpi/wiki/How-to-access-NextCloudPi
[Mon Nov 12 04:52:08.914107 2018] [authz_host:error] [pid 2749:tid 2755679280] [client fe80::e006:de0f:22e9:4ecc:54732] AH01753: access check of 'localhost' to /favicon.ico failed, reason: unable to get the remote host name
[Mon Nov 12 04:52:08.914371 2018] [authz_core:error] [pid 2749:tid 2755679280] [client fe80::e006:de0f:22e9:4ecc:54732] AH01630: client denied by server configuration: /var/www/ncp-web/favicon.ico
[Mon Nov 12 04:54:26.821134 2018] [authz_host:error] [pid 2749:tid 2755679280] [client 192.168.1.7:55443] AH01753: access check of 'localhost' to / failed, reason: unable to get the remote host name
[Mon Nov 12 04:54:33.762770 2018] [authz_host:error] [pid 2749:tid 2755679280] [client 192.168.1.7:55472] AH01753: access check of 'localhost' to / failed, reason: unable to get the remote host name
[Mon Nov 12 04:54:33.763185 2018] [authnz_external:error] [pid 2749:tid 2755679280] [client 192.168.1.7:55472] Invalid AuthExternal keyword (pwauth)
[Mon Nov 12 04:54:34.393688 2018] [authz_host:error] [pid 2749:tid 2755679280] [client 192.168.1.7:55472] AH01753: access check of 'localhost' to /favicon.ico failed, reason: unable to get the remote host name
[Mon Nov 12 04:54:55.398672 2018] [authz_host:error] [pid 2750:tid 2755679280] [client 192.168.1.7:55546] AH01753: access check of 'localhost' to / failed, reason: unable to get the remote host name
[Mon Nov 12 04:54:55.398979 2018] [authnz_external:error] [pid 2750:tid 2755679280] [client 192.168.1.7:55546] Invalid AuthExternal keyword (pwauth)
[Mon Nov 12 04:54:56.848048 2018] [authz_host:error] [pid 2750:tid 2755679280] [client 192.168.1.7:55551] AH01753: access check of 'localhost' to / failed, reason: unable to get the remote host name
[Mon Nov 12 04:54:56.848400 2018] [authnz_external:error] [pid 2750:tid 2755679280] [client 192.168.1.7:55551] Invalid AuthExternal keyword (pwauth)
[Mon Nov 12 04:55:17.330766 2018] [authz_host:error] [pid 2749:tid 2755679280] [client 192.168.1.7:55612] AH01753: access check of 'localhost' to / failed, reason: unable to get the remote host name
[Mon Nov 12 04:55:33.854980 2018] [authz_host:error] [pid 2750:tid 2755679280] [client 192.168.1.7:55627] AH01753: access check of 'localhost' to / failed, reason: unable to get the remote host name
[Mon Nov 12 04:55:33.855247 2018] [authnz_external:error] [pid 2750:tid 2755679280] [client 192.168.1.7:55627] Invalid AuthExternal keyword (pwauth)
[Mon Nov 12 04:55:34.032055 2018] [authz_host:error] [pid 2750:tid 2755679280] [client 192.168.1.7:55627] AH01753: access check of 'localhost' to /favicon.ico failed, reason: unable to get the remote host name
[Mon Nov 12 04:55:35.872256 2018] [authz_host:error] [pid 2750:tid 2755679280] [client 192.168.1.7:55631] AH01753: access check of 'localhost' to / failed, reason: unable to get the remote host name
[Mon Nov 12 04:55:35.872593 2018] [authnz_external:error] [pid 2750:tid 2755679280] [client 192.168.1.7:55631] Invalid AuthExternal keyword (pwauth)

```
</details>

<details>
<summary>Database logs</summary>

```

2018-11-12  4:49:55 3062557904 [Note] InnoDB: Using mutexes to ref count buffer pool pages
2018-11-12  4:49:55 3062557904 [Note] InnoDB: The InnoDB memory heap is disabled
2018-11-12  4:49:55 3062557904 [Note] InnoDB: Mutexes and rw_locks use GCC atomic builtins
2018-11-12  4:49:55 3062557904 [Note] InnoDB: GCC builtin __atomic_thread_fence() is used for memory barrier
2018-11-12  4:49:55 3062557904 [Note] InnoDB: Compressed tables use zlib 1.2.8
2018-11-12  4:49:55 3062557904 [Note] InnoDB: Using Linux native AIO
2018-11-12  4:49:55 3062557904 [Note] InnoDB: Using generic crc32 instructions
2018-11-12  4:49:55 3062557904 [Note] InnoDB: Initializing buffer pool, size = 128.0M
2018-11-12  4:49:55 3062557904 [Note] InnoDB: Completed initialization of buffer pool
2018-11-12  4:49:55 3062557904 [Note] InnoDB: Highest supported file format is Barracuda.
2018-11-12  4:49:56 3062557904 [Note] InnoDB: 128 rollback segment(s) are active.
2018-11-12  4:49:56 3062557904 [Note] InnoDB: Waiting for purge to start
2018-11-12  4:49:56 3062557904 [Note] InnoDB:  Percona XtraDB (http://www.percona.com) 5.6.35-80.0 started; log sequence number 3415595
2018-11-12  4:49:56 2533356352 [Note] InnoDB: Dumping buffer pool(s) not yet started
2018-11-12  4:49:56 3062557904 [Note] Plugin 'FEEDBACK' is disabled.
2018-11-12  4:49:56 3062557904 [Note] Server socket created on IP: '127.0.0.1'.
2018-11-12  4:49:56 3062557904 [Note] /usr/sbin/mysqld: ready for connections.
Version: '10.1.23-MariaDB-9+deb9u1'  socket: '/var/run/mysqld/mysqld.sock'  port: 3306  Raspbian 9.0

```
</details>

<details>
<summary>Nextcloud logs</summary>

```

```
</details>

nachoparker, 2018-11-13:
https://github.com/nextcloud/nextcloudpi/wiki/FAQ#i-get-stuck-at-nc-not-yet-initialized--nextcloud-is-so-slow--failing-randomly--getting-server-errors-even-with-a-fresh-image

Gabinator9, 2018-11-15:
> take it out, copy the whole image to your computer and then copy it back to the SD. Assert that there are no read/write errors. You can also check the md5sum after reading it again and assert that it matches the md5sum of the image that you just copied

"Ive tried logging in to the web UI multiple times after re etchering my ncp image to my SD card and I cannot get it to work" from my previous message

>Check your power supply. Many cheap power supplies are not stable enough for reliable functioning. Also try to have external supplies for your external HDDs.

Everytime I reimaged my SD card, I also disconnected the ethernet cable and power cable and plugged them back in.

closed 2018-11-16 by nachoparker