nextcloud/nextcloudpi #713: NextcloudPi Install on Debian Stretch (AMD64): Installation of a swap file while a swap file/swap partition already exists

LittleAlf, 2018-11-12:
**System information**
* VM with VirtualBox 5.2.22r:
    > 4 GB RAM
    > 29 GB hard disk
    > network bridge connection
    > EFI setup

This is a little issue combined with a feature request: I installed NextCloudPi on Debian using a fresh Debian Stretch install on the above virtual machine. I used 'debian-9.5.0-amd64-DVD-1.iso' with minimal installation and no graphical interface. I followed the guided install, with **one** major exception: I changed the root filesystem to **btrfs**! The guided install proposed the following partitioning:
* 512 MB EFI partition (vfat)
* 23.5 GB root partition (btrfs)
* 4 GB swap partition (swap)

which I accepted and continued with the installation.

After the installation I did `apt-update`, `apt-upgrade`, and `apt-dist-upgrade`.

Then I ran `curl -sSL https://raw.githubusercontent.com/nextcloud/nextcloudpi/master/install.sh | bash`. This installed well without errors. (*Thank you, Nachoparker!*) The only additional message was:

`swap file has holes`

NextCloupPi-Install installed the package `dphys-swapfile` and created a new swap file on the root partition. This does not work well on a btrfs partition. Therefore this message. So I had to manually change `swapon` back to the swap partition, clear the swap file, and deinstall the `dphys-swapfile` package.

**Feature request**
*Before* the installation of `dphys-swapfile` package incorporate a check, if a swap file or a swap partition already exists, and is active.

Only, if no active swap file/partition is found, install `dphys-swapfile` and create the swap file.

LittleAlf, 2018-11-13:
Dear Nachoparker,

thanks for your answer. I use the VM because I am thinking about installing NCP on a more powerful Intel-based hardware (Zotac ZBOX Pico PI335), and want to make sure, that I get it up and running properly. So I am testing the installation procedure on the VM.

I saw your new VM (*thank you!*) and did already a try - but I did not get it running with Virtualbox on Windows yet. If I will not suceed I will ask a question in the Nextcloud forum.

bmbeverst, 2020-04-15:
Good day,

I have found that in a LXC container the update.sh script fails when it tries to install the `nc-swapfile.sh` script due to `dphys-swapfile` usage of fallocate failing since the "operation is not supported". This can be fixed with adding the `-x` arg to fallocate but swapon seems to skip the swap file due to "it appears to have holes" which also causes update.sh to fail.

Steps to duplicate:
* Create an LXC container
* apt install dphys-swapfile

At this point I don't need nc-swapfile and am just going to remove it before the install.

Will report back about how well that goes.

P.S. I did not use the NCP VM due to not wanting a fat VM. I did not use docker since it is missing most of the features of NCP.

bmbeverst, 2020-04-16:
Managed to remove the nc-swapfile.sh with this command, `while true; do find / -type f -name "nc-swapfile.sh" -print -exec rm -f {} \; 2>/dev/null; sleep 5; done`

But now I get a failure trying to setup nc-static-IP:
```
Installing nc-static-IP
/usr/local/etc/library.sh: line 216: : No such file or directory
```
Which is this [line](https://github.com/nextcloud/nextcloudpi/blob/master/etc/library.sh#L216) `source "$script"`

nartahn, 2020-05-11:
@bmbeverst 
Did you manage to get it installed in Proxmox LXC?

bmbeverst, 2020-05-13:
Unluckily no. 

I tried again with `touch /.ncp-image` which causes the setup to act differently see, [ncp-update](https://github.com/nextcloud/nextcloudpi/blob/9b1ecbb8d605ddded41ed011653b213d4b9b03da/bin/ncp-update#L20), [update.sh](https://github.com/nextcloud/nextcloudpi/blob/4a51c1f5e8a33e670a63c4c82f7eae7115b86980/update.sh#L170)

It did get me to a nextcloudpi motd, but pretty sure the rest of the install failed.

Either I am going to play with this more or just try with a VM next.

bmbeverst, 2020-05-13:
I found and got the VM image working!

1. Go to the[ download site](https://ownyourbits.com/downloads/). It was hard to find this is the [only docs of it](https://docs.nextcloudpi.com/en/how-to-install-nextcloudpi/)
1. Download the 1.1 GB file, or torrent but only one person seeding.
1. Extract the img file and upload it (Probably better to copy the compress file 1.1 GB to your Proxmox host and extract there, a 3.5 GB file)
1. Run these command [pve source](https://pve.proxmox.com/wiki/Migration_of_servers_to_Proxmox_VE#Qemu.2FKVM)
```qm create 120 --bootdisk scsi0
qm importdisk 120 NextCloudPi_VM_03-29-20.img someStorage
qm set 120 --scsi0 someStorage:vm-120-disk-0.raw
```

Some notes,
* The importdisk command seems to ignore the `someStorage` so just run it with the path to the file or from the same directory as the file, `cd /someStorage/templates/iso` then run it.
* The VM is created with no network, 1 CPU and only 512 RAM! So add some more resources once it is ready to go.
* Did not document any thing since it is late, let me know if you need any help!

nartahn, 2020-05-17:
I got it working, thanks for your help. 
FYI... If you scroll down a bit further under templates of LXC containers you will find [turnkeylinux-nextcloud](https://www.turnkeylinux.org/nextcloud). Might be worth a look if you have spare time.

hockey6611, 2020-09-11:
Hoping to re-awaken the topic. I am still seeing the above issues with current issues with `dphys-swapfile`, on 1.29.8 of ncp. I am also trying to get NCP working in an LXC container on proxmox.

> I got it working, thanks for your help.
> FYI... If you scroll down a bit further under templates of LXC containers you will find [turnkeylinux-nextcloud](https://www.turnkeylinux.org/nextcloud). Might be worth a look if you have spare time.

I have managed to get the turnkey linux container working, however, it is pretty out of date and needs a good amount of updating. I would still prefer the convenience and features of NCP in an LXC.

nachoparker, 2020-09-12:
Unfortunately NCP expects a clean debian filesystem, you will have to remove swap and dphys-swapfile before installing.

closed 2020-09-12 by nachoparker

bmbeverst, 2020-09-15:
NCP installs `dphys-swapfile` even if you remove it. See this line [here](https://github.com/nextcloud/nextcloudpi/blob/34cba9f52876dd1b0026c1ed6d7ab3796926c79d/bin/ncp/SYSTEM/nc-swapfile.sh#L51). It would be great if we could disable `nc-swapfile.sh` some how.

Please reopen this issue, I can't, or should I create a new one?

hockey6611, 2020-09-16:
Hopefully you'll consider re-opening issue as compatibility with a container would be a great feature.

nachoparker, 2021-09-29:
added experimental support for LXC/LXD

https://github.com/nextcloud/nextcloudpi/commit/212f4e01325978b9bd937171814e564164c5f4af

there will be an image next release