nextcloud/nextcloudpi #714: NeedRestart looks for terminal input during automatic updates

Iolaum, 2018-11-12:
## Issue Description

After unattended-upgrades run I checked the logs in the email and saw the following:

```
Unable to get Terminal Size. The TIOCGWINSZ ioctl didn't work. The COLUMNS and LINES environment variables didn't work. The resize program didn't work. The stty program didn't work. at /usr/share/perl5/NeedRestart/UI.pm line 50. 
Running kernel seems to be up-to-date. 
Unable to get Terminal Size. The TIOCGWINSZ ioctl didn't work. The COLUMNS and LINES environment variables didn't work. The resize program didn't work. The stty program didn't work. at /usr/share/perl5/NeedRestart/UI.pm line 50. 
Services to be restarted: 
  systemctl restart avahi-daemon.service 
  systemctl restart dnsmasq.service 
  systemctl restart mariadb.service 
  systemctl restart php7.2-fpm.service 
  systemctl restart ssh.service 
  systemctl restart triggerhappy.service 
  systemctl restart udisks2.service 
Unable to get Terminal Size. The TIOCGWINSZ ioctl didn't work. The COLUMNS and LINES environment variables didn't work. The resize program didn't work. The stty program didn't work. at /usr/share/perl5/NeedRestart/UI.pm line 50. 
Services being skipped: 
  /etc/needrestart/restart.d/dbus.service 
  systemctl restart systemd-journald.service 
  systemctl restart systemd-logind.service 
Unable to get Terminal Size. The TIOCGWINSZ ioctl didn't work. The COLUMNS and LINES environment variables didn't work. The resize program didn't work. The stty program didn't work. at /usr/share/perl5/NeedRestart/UI.pm line 50. 
No containers need to be restarted. 
Unable to get Terminal Size. The TIOCGWINSZ ioctl didn't work. The COLUMNS and LINES environment variables didn't work. The resize program didn't work. The stty program didn't work. at /usr/share/perl5/NeedRestart/UI.pm line 50. 
No user sessions are running outdated binaries.
```

I m not sure if this is a problem or we can just wait until debian gets to upgrade the package (likely in buster).

The problem may have been fixed in `2.12` according to [this bug report](https://github.com/liske/needrestart/issues/86). However currently we have 2.11:

```
$ apt policy needrestart
needrestart:
  Installed: 2.11-3+deb9u1
  Candidate: 2.11-3+deb9u1
  Version table:
 *** 2.11-3+deb9u1 500
        500 http://raspbian.raspberrypi.org/raspbian stretch/main armhf Packages
        100 /var/lib/dpkg/status
```

Just mentioning this in case there's a quick fix/workaround that I don't know of and for documentation purposes.

**System information**

<details>
<summary>NextCloudPi diagnostics</summary>

```

Cannot load Zend OPcache - it was already loaded
NextCloudPi version  v0.66.1
NextCloudPi image    NextCloudPi_12-04-17
distribution         Raspbian GNU/Linux 9 \n \l
automount            yes
USB devices          sda 
datadir              /media/USBdrive/ncdata
data in SD           no
data filesystem      ext2/ext3
data disk usage      256G/917G
rootfs usage         1.9G/3.6G
swapfile             /media/wd1/swap
dbdir                /media/USBdrive/ncdatabase
Nextcloud check      ok
Nextcloud version    14.0.3.0
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
[... 106 lines cut ...]
```

{"reqId":"WtCUs5w4t8kuyqKL6naz","level":1,"time":"2018-10-12T16:22:35+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Repair::info: Repair info: Image cache cleared","userAgent":"--","version":"14.0.1.1"}
{"reqId":"WtCUs5w4t8kuyqKL6naz","level":1,"time":"2018-10-12T16:22:37+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Repair::info: Repair info: SCSS cache cleared","userAgent":"--","version":"14.0.1.1"}
{"reqId":"WtCUs5w4t8kuyqKL6naz","level":1,"time":"2018-10-12T16:22:41+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Repair::info: Repair info: JS cache cleared","userAgent":"--","version":"14.0.1.1"}
{"reqId":"WtCUs5w4t8kuyqKL6naz","level":1,"time":"2018-10-12T16:22:41+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Repair::step: Repair step: Add preview background cleanup job","userAgent":"--","version":"14.0.1.1"}
{"reqId":"WtCUs5w4t8kuyqKL6naz","level":1,"time":"2018-10-12T16:22:41+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Repair::step: Repair step: Queue a one-time job to cleanup old backups of the updater","userAgent":"--","version":"14.0.1.1"}
{"reqId":"WtCUs5w4t8kuyqKL6naz","level":1,"time":"2018-10-12T16:22:41+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Repair::step: Repair step: Repair pending cron jobs","userAgent":"--","version":"14.0.1.1"}
{"reqId":"WtCUs5w4t8kuyqKL6naz","level":1,"time":"2018-10-12T16:22:41+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Repair::info: Repair info: No need to repair pending cron jobs.","userAgent":"--","version":"14.0.1.1"}
{"reqId":"WtCUs5w4t8kuyqKL6naz","level":1,"time":"2018-10-12T16:22:41+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Updater::startCheckCodeIntegrity: Starting code integrity check...","userAgent":"--","version":"14.0.1.1"}
{"reqId":"WtCUs5w4t8kuyqKL6naz","level":1,"time":"2018-10-12T16:23:19+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Updater::finishedCheckCodeIntegrity: Finished code integrity check","userAgent":"--","version":"14.0.1.1"}
{"reqId":"WtCUs5w4t8kuyqKL6naz","level":1,"time":"2018-10-12T16:23:19+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Updater::updateEnd: Update successful","userAgent":"--","version":"14.0.3.0"}
{"reqId":"WtCUs5w4t8kuyqKL6naz","level":1,"time":"2018-10-12T16:23:19+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Updater::maintenanceDisabled: Turned off maintenance mode","userAgent":"--","version":"14.0.3.0"}
{"reqId":"WtCUs5w4t8kuyqKL6naz","level":1,"time":"2018-10-12T16:23:19+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Updater::resetLogLevel: Reset log level to Warning(2)","userAgent":"--","version":"14.0.3.0"}
{"reqId":"2DEOlocwS6UptdGqB95L","level":3,"time":"2018-10-13T04:00:09+00:00","remoteAddr":"","user":"--","app":"cron","method":"","url":"--","message":{"Exception":"Error","Message":"Call to undefined method OCP\\AppFramework\\Http\\DataResponse::succeeded()","Code":0,"Trace":[{"file":"\/var\/www\/nextcloud\/lib\/private\/BackgroundJob\/Job.php","line":61,"function":"run","class":"OCA\\Survey_Client\\BackgroundJobs\\MonthlyReport","type":"->","args":[null]},{"file":"\/var\/www\/nextcloud\/lib\/private\/BackgroundJob\/TimedJob.php","line":55,"function":"execute","class":"OC\\BackgroundJob\\Job","type":"->","args":[{"__class__":"OC\\BackgroundJob\\JobList"},{"__class__":"OC\\Log"}]},{"file":"\/var\/www\/nextcloud\/cron.php","line":123,"function":"execute","class":"OC\\BackgroundJob\\TimedJob","type":"->","args":[{"__class__":"OC\\BackgroundJob\\JobList"},{"__class__":"OC\\Log"}]}],"File":"\/var\/www\/nextcloud\/apps\/survey_client\/lib\/BackgroundJobs\/MonthlyReport.php","Line":43,"CustomMessage":"--"},"userAgent":"--","version":"14.0.3.0"}
{"reqId":"W8Y5YX8AAQEAAHB2qMIAAAAK","level":4,"time":"2018-10-16T19:17:53+00:00","remoteAddr":"192.168.1.7","user":"admin","app":"webdav","method":"PUT","url":"\/remote.php\/dav\/files\/admin\/shared\/A011%20Photos-Marriage\/dialegmenes\/karta4\/_ASH0209.jpg","message":{"Exception":"OCA\\DAV\\Connector\\Sabre\\Exception\\FileLocked","Message":"\"shared\/A011 Photos-Marriage\" is locked","Code":0,"Trace":[{"file":"\/var\/www\/nextcloud\/3rdparty\/sabre\/dav\/lib\/DAV\/Server.php","line":1096,"function":"createFile","class":"OCA\\DAV\\Connector\\Sabre\\Directory","type":"->","args":["_ASH0209.jpg",null]},{"file":"\/var\/www\/nextcloud\/3rdparty\/sabre\/dav\/lib\/DAV\/CorePlugin.php","line":529,"function":"createFile","class":"Sabre\\DAV\\Server","type":"->","args":["files\/admin\/shared\/A011 Photos-Marriage\/dialegmenes\/karta4\/_ASH0209.jpg",null,null]},{"function":"httpPut","class":"Sabre\\DAV\\CorePlugin","type":"->","args":[{"absoluteUrl":"https:\/\/$WEBSITE\/remote.php\/dav\/files\/admin\/shared\/A011%20Photos-Marriage\/dialegmenes\/karta4\/_ASH0209.jpg","__class__":"Sabre\\HTTP\\Request"},{"__class__":"Sabre\\HTTP [... cut ...] lass":"OC\\BackgroundJob\\Job","type":"->","args":[{"__class__":"OC\\BackgroundJob\\JobList"},{"__class__":"OC\\Log"}]},{"file":"\/var\/www\/nextcloud\/cron.php","line":123,"function":"execute","class":"OC\\BackgroundJob\\TimedJob","type":"->","args":[{"__class__":"OC\\BackgroundJob\\JobList"},{"__class__":"OC\\Log"}]}],"File":"\/var\/www\/nextcloud\/apps\/survey_client\/lib\/BackgroundJobs\/MonthlyReport.php","Line":43,"CustomMessage":"--"},"userAgent":"--","version":"14.0.3.0"}

```
</details>

nachoparker, 2018-11-13:
If it's an important issue they will fix it. Do you know the implications? does it just provide a warning and that's it?

thanks for the feedback

Iolaum, 2018-11-13:
I 'm not sure of the implications. Usually when I see this issue appearing I will just `sudo apt update`, `sudo apt upgrade` and then reboot. Next time it happens I will try to investigate it a little bit more and post what I found.

Needrestart has been updated in [Sid](https://packages.debian.org/sid/needrestart) and [Buster](https://packages.debian.org/buster/needrestart) but I 'm not sure if it will be updated in stretch. At the very least this means that once we upgrade to Buster next year this will go away.

closed 2018-11-27 by nachoparker