nextcloud/nextcloudpi #715: Access denied for user 'ncadmin'@'localhost' (using password: YES)

shaan7, 2018-11-17:
I have been running NCP successfully for a while using `docker-compose-armhf.yml` on Asus Tinkerboard. This morning I had to reboot the device and now NextCloud shows me a white page.

Digging more, this seems problematic-
```
==> var/log/apache2/nc-error.log <==
[Sat Nov 17 03:10:04.906205 2018] [proxy_fcgi:error] [pid 90:tid 2765812768] [client 192.168.1.1:59946] AH01071: Got error 'PHP message: PHP Fatal error:  Uncaught Doctrine\\DBAL\\DBALException: Failed to connect to the database: An exception occured in driver: SQLSTATE[HY000] [1045] Access denied for user 'ncadmin'@'localhost' (using password: YES)
```

`config.php` says this-
```
  'dbtype' => 'mysql',
  'version' => '13.0.6.1',
  'dbname' => 'nextcloud',
  'dbhost' => 'localhost',
  'dbport' => '',
  'dbtableprefix' => 'oc_',
  'mysql.utf8mb4' => true,
  'dbuser' => 'ncadmin',
  'dbpassword' => 'default',
```

Is it supposed to be `"default"`? Or something else is wrong?

**System information**

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v0.66.4
NextCloudPi image    NextCloudPi_docker_09-21-18
distribution         Debian GNU/Linux 9 \n \l
USB devices          sda 
datadir              /data/app/data
data in SD           yes
data filesystem      ext2/ext3
data disk usage      77G/541G
rootfs usage         77G/541G
swapfile             /dev/zram4
dbdir                /data/database
Nextcloud check      error
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        closed
port check 443       open
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               3:03

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

An unhandled exception has been thrown:
Doctrine\DBAL\DBALException: Failed to connect to the database: An exception occured in driver: SQLSTATE[HY000] [1045] Access denied for user 'ncadmin'@'localhost' (using password: YES) in /data/app/lib/private/DB/Connection.php:64
Stack trace:
#0 /data/app/3rdparty/doctrine/dbal/lib/Doctrine/DBAL/Connection.php(429): OC\DB\Connection->connect()
#1 /data/app/3rdparty/doctrine/dbal/lib/Doctrine/DBAL/Connection.php(389): Doctrine\DBAL\Connection->getDatabasePlatformVersion()
[... 115 lines cut ...]
```

{"reqId":"W@2Jr6wSAAIAAABuM3sAAEIP","level":3,"time":"2018-11-15T14:58:55+00:00","remoteAddr":"192.168.1.1","user":"shantanu","app":"gallery","method":"GET","url":"\/index.php\/apps\/gallery\/thumbnails?ids=7677%3B7230%3B6969%3B9356%3B7678%3B7679&scale=1&square=0&requesttoken=fP2QeafrgMAuhM6tqCRd8ILSA2uAZfGErgNVfvS3vmA%3D%3ANL%2FhTcWR76ZJ8JzJxBYZlOO7UTLlD6XjxDFgDsbRyFQ%3D","message":"Exception: Preview generation has failed","userAgent":"Mozilla\/5.0 (X11; Ubuntu; Linux x86_64; rv:63.0) Gecko\/20100101 Firefox\/63.0","version":"13.0.6.1"}
{"reqId":"W@2Zt6wSAAIAAABuM-YAAABB","level":2,"time":"2018-11-15T16:07:19+00:00","remoteAddr":"184.105.247.194","user":"--","app":"core","method":"GET","url":"\/","message":"Trusted domain error. \"184.105.247.194\" tried to access using \"122.178.217.28\" as host.","userAgent":"--","version":"13.0.6.1"}
{"reqId":"W@4RfqwSAAIAAABuN7cAAABQ","level":2,"time":"2018-11-16T00:38:22+00:00","remoteAddr":"139.162.113.204","user":"--","app":"core","method":"GET","url":"\/","message":"Trusted domain error. \"139.162.113.204\" tried to access using \"122.178.217.28\" as host.","userAgent":"HTTP Banner Detection (https:\/\/security.ipip.net)","version":"13.0.6.1"}
{"reqId":"W@5VFawSAAIAAABtLGwAAAAX","level":4,"time":"2018-11-16T05:26:45+00:00","remoteAddr":"192.168.1.1","user":"manijha","app":"webdav","method":"PUT","url":"\/remote.php\/dav\/files\/manijha\/Documents\/College\/Placement\/August\/%20WTM%20Scholarship%20Program%0AAPAC%202018.pdf","message":"Exception: {\"Exception\":\"OCA\\\\DAV\\\\Connector\\\\Sabre\\\\Exception\\\\InvalidPath\",\"Message\":\"File name contains at least one invalid character\",\"Code\":0,\"Trace\":\"#0 \\\/data\\\/app\\\/3rdparty\\\/sabre\\\/dav\\\/lib\\\/DAV\\\/Tree.php(76): OCA\\\\DAV\\\\Connector\\\\Sabre\\\\Directory->getChild(' WTM Scholarshi...')\\n#1 \\\/data\\\/app\\\/apps\\\/dav\\\/lib\\\/Connector\\\/Sabre\\\/LockPlugin.php(59): Sabre\\\\DAV\\\\Tree->getNodeForPath('files\\\/manijha\\\/D...')\\n#2 [internal function]: OCA\\\\DAV\\\\Connector\\\\Sabre\\\\LockPlugin->getLock(Object(Sabre\\\\HTTP\\\\Request), Object(Sabre\\\\HTTP\\\\Response))\\n#3 \\\/data\\\/app\\\/3rdparty\\\/sabre\\\/event\\\/lib\\\/EventEmitterTrait.php(105): call_user_func_array(Array, Array)\\n#4 \\\/data\\\/app\\\/3rdparty\\\/sabre\\\/dav\\\/lib\\\/DAV\\\/Server.php(466): Sabre\\\\Event\\\\EventEmitter->emit('beforeMethod', Array)\\n#5 \\\/data\\\/app\\\/3rdparty\\\/sabre\\\/dav\\\/lib\\\/DAV\\\/Server.php(254): Sabre\\\\DAV\\\\Server->invokeMethod(Object(Sabre\\\\HTTP\\\\Request), Object(Sabre\\\\HTTP\\\\Response))\\n#6 \\\/data\\\/app\\\/apps\\\/dav\\\/lib\\\/Server.php(293): Sabre\\\\DAV\\\\Server->exec()\\n#7 \\\/data\\\/app\\\/apps\\\/dav\\\/appinfo\\\/v2\\\/remote.php(35): OCA\\\\DAV\\\\Server->exec()\\n#8 \\\/data\\\/app\\\/remote.php(164): require_once('\\\/data\\\/app\\\/apps\\\/...')\\n#9 {main}\",\"File\":\"\\\/data\\\/app\\\/apps\\\/dav\\\/lib\\\/Connector\\\/Sabre\\\/Directory.php\",\"Line\":227}","userAgent":"Mozilla\/5.0 (Linux) mirall\/2.3.3 (Nextcloud)","version":"13.0.6.1"}
{"reqId":"W@5VFawSAAIAAABtLG0AAAAK","level":4,"time":"2018-11-16T05:26:45+00:00","remoteAddr":"192.168.1.1","user":"manijha","app":"webdav","method":"PUT","url":"\/remote.php\/dav\/files\/manijha\/Documents\/College\/Placement\/August\/HSBC%0ATechnology.pdf","message":"Exception: {\"Exception\":\"OCA\\\\DAV\\\\Connector\\\\Sabre\\\\Exception\\\\InvalidPath\",\"Message\":\"File name contains at least one invalid character\",\"Code\":0,\"Trace\":\"#0 \\\/data\\\/app\\\/3rdparty\\\/sabre\\\/dav\\\/lib\\\/DAV\\\/Tree.php(76): OCA\\\\DAV\\\\Connector\\\\Sabre\\\\Directory->getChild('HSBC\\\\nTechnology...')\\n#1 \\\/data\\\/app\\\/apps\\\/dav\\\/lib\\\/Connector\\\/Sabre\\\/LockPlugin.php(59): Sabre\\\\DAV\\\\Tree->getNodeForPath('files\\\/manijha\\\/D...')\\n#2 [internal function]: OCA\\\\DAV\\\\Connector\\\\Sabre\\\\LockPlugin->getLock(Object(Sabre\\\\HTTP\\\\Request), Object(Sabre\\\\HTTP\\\\Response))\\n#3 \\\/data\\\/app\\\/3rdparty\\\/sabre\\\/event\\\/lib\\\/EventEmitterTrait.php(105): call_user_func_array(Array, Array)\\n#4 \\\/data\\\/app\\\/3rdparty\\\/sabre\\\/dav\\\/lib\\\/DAV\\\/Server.php(466): Sabre\\\\Event\\\\EventEmitter->emit('beforeMethod', Array)\\n#5 \\\/data\\\/app\\\/3rdparty\\\/sabre\\\/dav\\\/lib\\\/DAV\\\/Server.php(254): Sabre\\\\DAV\\\\Server->invokeMethod(Object(Sabre\\\\HTTP\\\\Request), Object(Sabre\\\\HTTP\\\\Response))\\n#6 \\\/data\\\/app\\\/apps\\\/dav\\\/lib\\\/Server.php(293): Sabre\\\\DAV\\\\Server->exec()\\n#7 \\\/data\\\/app\\\/apps\\\/dav\\\/appinfo\\\/v2\\\/remote.php(35): OCA\\\\DAV\\\\Server->exec()\\n#8 \\\/data\\\/app\\\/remote.php(164): require_once('\\\/data\\\/app\\\/apps\\\/...')\\n#9 {main}\",\"File\":\"\\\/data\\\/app\\\/apps\\\/dav\\\/lib\\\/Connector\\\/Sabre\\\/Directory.php\",\"Line\":227}","userAgent":"Mozilla\/5.0 (Linux) mirall\/2.3.3 (Nextcloud)","version":"13.0.6.1"}
{"reqId":"W@5VFawSAAIAAABuOjIAAABE","level":4,"time":"2018-11-16T05:26:45+00:00","remoteAddr":"192.168.1.1","user":"manijha","app":"webdav","method":"PUT","url":"\/remote.php\/dav\/files\/manijha\/Documents\/College\/Placement\/August\/Incture%0ATechnologies.pdf","message":"Exception: {\"Exception\":\"OCA\\\\DAV\\\\Connector\\\\Sabre\\\\Exception\\\\InvalidPath\",\"Message\":\"File name contains at least one invalid character\",\"Code\":0,\"Trace\":\"#0 \\\/data\\\/app\\\/3rdparty\\\/sabre\\\/dav\\\/lib\\\/DAV\\\/Tree.php(76): OCA\\\\DAV\\\\Connector\\\\Sabre\\\\Directory->getChild('Incture\\\\nTechnol...')\\n#1 \\\/data\\\/app\\\/apps\\\/dav\\\/lib\\\/Connector\\\/Sabre\\\/LockPlugin.php(59): Sabre\\\\DAV\\\\Tree->getNodeForPath('files\\\/manijha\\\/D...')\\n#2 [internal function]: OCA\\\\DAV\\\\Connector\\\\Sabre\\\\LockPlugin->getLock(Object(Sabre\\\\HTTP\\\\Request), Object(Sabre\\\\HTTP\\\\Response))\\n#3 \\\/data\\\/app\\\/3rdparty\\\/sabre\\\/event\\\/lib\\\/EventEmitterTrait.php(105): call_user_func_array(Array, Array)\\n#4 \\\/data\\\/app\\\/3rdparty\\\/sabr [... cut ...] )\\n#18 \\\/data\\\/app\\\/apps\\\/dav\\\/lib\\\/Server.php(293): Sabre\\\\DAV\\\\Server->exec()\\n#19 \\\/data\\\/app\\\/apps\\\/dav\\\/appinfo\\\/v2\\\/remote.php(35): OCA\\\\DAV\\\\Server->exec()\\n#20 \\\/data\\\/app\\\/remote.php(164): require_once('\\\/data\\\/app\\\/apps\\\/...')\\n#21 {main}\",\"File\":\"\\\/data\\\/app\\\/lib\\\/private\\\/Files\\\/Cache\\\/QuerySearchHelper.php\",\"Line\":185}","userAgent":"Mozilla\/5.0 (iOS) Nextcloud-iOS\/2.22.5","version":"13.0.6.1"}

```
</details>

ovpc, 2018-11-17:
**Default** doesn't seem right!
Mysql user password should also be stored in /root/.my.cnf
That is in a normal install, not sure where it is in a docker container.

shaan7, 2018-11-17:
Wow, its `default` even there!
```
$ docker exec nextcloudpi_nextcloudpi_1 cat  /root/.my.cnf
[client]
password=default
```

nachoparker, 2018-11-17:
hi,

make sure you are on the latest version. This was fixed a long time ago. Update through ncp-web or `sudo ncp-config`

shaan7, 2018-11-17:
@nachoparker Cool, that worked! I didn't try updating manually because I was scared it will make matters worse :P

nero120, 2019-02-25:
@nachoparker I've been seeing a similar issue to this, except in my case the `dbpassword` value is being set to an incorrect salted value and is only fixed temporarily by setting it to the correct cleartext password value.

I've posted on the [forum](https://help.nextcloud.com/t/dbpassword-issues-in-config-php/48188) but I found this issue and thought I'd mention here in case you knew a quick fix?