nextcloud/nextcloudpi #720: LOG2RAM : No place on RAM anymore, fallback on the disk

funkmastergst, 2018-11-22:
Hi there,
I´ve got a couple of questions regarding log2ram and logrotate.

First the basics (from GUI/System Info):
System			ODroid HC2
NextCloudPi version	v0.66.4
NextCloudPi image	NextCloudPi_06-29-18
distribution		Debian GNU/Linux 9 \n \l
nc-ramlogs		active

After enabling nc-ramlogs I got an error message as e-mail:
"LOG2RAM : No place on RAM anymore, fallback on the disk"

So I checked /var/log and it seems to be located in RAM with sufficent space left:
$ df /var/log
Filesystem     1K-blocks  Used Available Use% Mounted on
/dev/zram0         49584  8084     37916  18% /var/log

On the other hand there is /var/log.hdd which is located on my SD-card:
$ df /var/log.hdd
Filesystem     1K-blocks    Used Available Use% Mounted on
/dev/mmcblk1p1  15138112 4489720  10460844  31% /var/log.hdd

1.) Why does log2ram thinks there is no place in RAM?

When checking "https://github.com/nextcloud/nextcloudpi/commit/e912749" I found these lines:
grep -q "Logs in RAM" /etc/fstab || cat >> /etc/fstab <<EOF
tmpfs /var/log tmpfs defaults,noatime,mode=1777,size=100M 0 0 # Logs in RAM
tmpfs /tmp tmpfs defaults,noatime,mode=1777 0 0 # /tmp in RAM

In my /etc/fstab there is only:
UUID=3f515314-8080-4acc-8e68-4fd603f56814 / ext4 defaults,noatime,nodiratime,commit=600,errors=remount-ro 0 1
tmpfs /tmp tmpfs defaults,nosuid 0 0
UUID=755e28bc-e031-401f-b39f-0c137174aed8 /media/wdred btrfs rw,nosuid,nodev,relatime,space_cache,subvolid=5,subvol=/,uhelper=udisks2 0 2
# a swapfile is not a swap partition, no line here
#   use  dphys-swapfile swap[on|off]  for that

First line is my SD, third my attached HDD. /var/log and /tmp are missing as tmpfs.

Additionally I checked logrotate. In /var/log.hdd the logs seem to rotate, as there are gzipped files with numbers ending. All configuration files in /etc/logrotate.conf lead to logs in /var/log.hdd

2.) So logrotate doesn´t work with /var/log out of the box? Or is my configuration somewhat broken?

Any help with the two questions is greatly appreciated. Thanks a lot.

Kind regards, funkmastergst

nachoparker, 2018-11-22:
Hi,

> When checking "e912749" I found these lines:
grep -q "Logs in RAM" /etc/fstab || cat >> /etc/fstab <<EOF
tmpfs /var/log tmpfs defaults,noatime,mode=1777,size=100M 0 0 # Logs in RAM
tmpfs /tmp tmpfs defaults,noatime,mode=1777 0 0 # /tmp in RAM

That is the old implementation, now we use log2ram. It is normal that you don't have anything like that in `fstab`.

- to check log2ram space, just do `df -h` and look for the `logs2ram` entry.
- ramlogs is configured to use only 40 MiB. You might be above that, in which case it falls back to SD card storage. You can clean up `/var/log.hdd` and `/var/log`.

- logrotate should be working, but it would be interesting to see what's taking all that log space in order we need to tweak it.

- maybe at the moment you activated log2ram your logs folder was already above 40MiB? in that case we should warn about this, or cleanup logs or something upon activating `nc-ramlogs`

cheers!

funkmastergst, 2018-11-23:
Hi nachoparker,
thanks for your information. Indeed log2ram never seemed to work correctly on my machine.
What I did now was first disabling nc-ramlogs from WebGUI, after rebooting I cleaned up /var/log and /var/log.hhd, i.e. I removed all compressed and rotated stuff. Both directories then contained less then 10MB. Now I enabled nc-ramlogs again and rebooted.
df -h now stated:
pi@nextcloudpi:~$ df -h
Filesystem      Size  Used Avail Use% Mounted on
udev            932M     0  932M   0% /dev
tmpfs           200M  6.6M  194M   4% /run
/dev/mmcblk1p1   15G  4.3G   11G  30% /
tmpfs           999M     0  999M   0% /dev/shm
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
tmpfs           999M     0  999M   0% /sys/fs/cgroup
tmpfs           999M     0  999M   0% /tmp
/dev/sda1       3.7T  168G  3.5T   5% /media/wdred
/dev/zram0       49M  9.5M   36M  22% /var/log
log2ram          40M   12M   29M  29% /var/log.hdd
tmpfs           200M     0  200M   0% /run/user/1001

I´ll have a look at logrotate in the future and will report back if something is odd. Regarding your question about huge logs: postfix was configured incorrect by me. I had some trouble with the auth against the smart host. These logs (8 in total, mail.* and .mail.*.swp) each were above 20MB quit fast. Your last assumption is correct, too. I strongly believe that the directories were above the limit when enabling ram2log and the fallback option kicked in from the start. A hint on the GUI would be helpful, automatic cleaning up even more. But in some other forum I read that not all applications automatically generate a new logfile if the old one gets deleted and perhaps show signs of error?!? Just emptying the files might be a solution.

So the issue is closed for me, thanks again. If I can help further please let me know.

Kind regards, funkmastergst

closed 2018-11-23 by nachoparker