nextcloud/nextcloudpi #722: Upstream pgp provider packages incompatible armhf

CoolSilver, 2018-11-23:
https://github.com/oerdnj/deb.sury.org/issues/1006

Bricks SSH and SSL and unstable system. Happens with Rasbian and the latest nextcloudpi image when updating.

nachoparker, 2018-11-23:
thanks for reporting @CoolSilver. Indeed they don't have their stuff very well put together with their naming of _armhf_ architecture.

This is only with pi1/zero right? (armv6)

CoolSilver, 2018-11-23:
This was on a Pi 3 I believe. If not a Pi 2. I will verify tonight

CoolSilver, 2018-11-29:
You're right it is a B+ 1.2. No wonder why I was thinking it should perform better.

nachoparker, 2018-11-30:
maybe next summer we'll support it again once Raspbian is on buster

closed 2018-11-30 by nachoparker

nachoparker, 2019-07-07:
armv6 pi versions, like 1b and zero are supported again.

Also NCP can work with different PHP versions, even though PHP7.2 or greater is required for NC16