nextcloud/nextcloudpi #728: Nextcloud - NC-info show error

vashalejandro, 2018-11-29:
Hello,

I've setup Nextcloud on my Pi 3B+ and everything works fine until I move my data to my external drive using `nc-datadir` after that my instance just start giving me error on Nextcloud Check and I'm not able to access my Cloud.

This is what `nc-info` shows: 

> Launching nc-info
Gathering information...
{"reqId":"2nS0bYmWDo9OlkNkxJ9g","level":2,"time":"2018-11-29T03:59:00+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Temporary directory \/media\/myCloudDrive\/ncdata\/\/tmp is not present or writable","userAgent":"--","version":"14.0.4.2"}
NextCloudPi version  v0.67.2
distribution         Raspbian GNU/Linux 9 \n \l
automount            yes
USB devices          sda sdb
datadir              /media/myCloudDrive/ncdata/
data in SD           no
data filesystem      btrfs
data disk usage      1.5G/932G
rootfs usage         7.9G/30G
swapfile             /var/swap
dbdir                /var/lib/mysql
Nextcloud check      error
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   **********
gateway              *********
interface            eth0
certificates         ************
NAT loopback         yes
uptime               15:26

Also as you can see I get an error at first.

Please, if you someone could guide me with this issue that would be great.

Thanks!

ovpc, 2018-11-29:
Have you tried to run

 nc-fix-permissions

from ncp-config?

vashalejandro, 2018-11-29:
I've tried that too.

What I did was moving the data directory back to the SD card. Then I formatted the drive, cleaned up all the files and moved the data directory back to the drive. Now I will copy my files from a manual backup I have to see if everything keeps working normal.

closed 2018-11-30 by nachoparker