nextcloud/nextcloudpi #729: "Referrer-Policy" HTTP header

funkmastergst, 2018-11-29:
Hi there,

after updating Nextcloud from 14.0.3 to 14.0.4 the following warning is shown under Administration/Overview for admins:
"The "Referrer-Policy" HTTP header is not set to "no-referrer", "no-referrer-when-downgrade", "strict-origin", "strict-origin-when-cross-origin" or "same-origin". This can leak referer information."

There was no such warning prior to the upgrade. No header information has been changed by me seperatly, so this part of NCP should be out of the box. In some other threads elsewhere the issue is also known. Suggestions range from checking .htaccess files (Header set Referrer-Policy "no-referrer") to editing
/etc/apache2/sites-enabled/*.conf (adding "Header set Referrer-Policy "no-referrer"" between <VirtualHost... :port… and </VirtualHost.... Nothing worked for me so far.

Another hint is that with the update a check produces the warning if both options from above are set, i.e. in webserver conf and in .htaccess. For me it is not quit clear which .htaccess is clearly meant (/var/www/ncp-web, /var/www/nextcloud, /var/www/html or /media/wdred/data (which is my data dir) or all of them).

Anybody else with this issue?

Regards, funkmastergst

**System information**

NextCloudPi version  v0.67.2
NextCloudPi image    NextCloudPi_06-29-18
distribution         Debian GNU/Linux 9 \n \l
automount            yes
USB devices          sda sdb
datadir              /media/wdred/ncdata
data in SD           no
data filesystem      btrfs
data disk usage      172G/3.7T
rootfs usage         4.3G/15G
swapfile             /var/swap
dbdir                /media/wdred/ncdatabase
Nextcloud check      ok
Nextcloud version    14.0.4.2
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            enx001e0632c276
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               22:26

Further logs upon request

zedocrob, 2018-11-29:
i had the same issue, 
after rebooting my Odroid-HC2 the message was gone and it was OK in "Admin/overview check".

nachoparker, 2018-11-30:
Yes, it is fixed in the update, but need a reboot or restart apache to take effect.

closed 2018-11-30 by nachoparker

funkmastergst, 2018-11-30:
Hi nachoparker,

indeed I found the comment about the fix in Releases/v0.67.1. But I rebooted several times since the update and the warning is persistent. Unfortenately I am not able to read the provided source code. So please could you share some information in noobish about the fix, so that I can check manually if the configuration is correct?

Thanks in advance and kind regards
funkmastergst

funkmastergst, 2018-12-01:
I just found a fix for my problem, but its kind of weird. After double checking the content of /var/www/nextcloud/.htaccess (and all other .htaccess files of the system) I manually restarted apache with "sudo /etc/init.d/apache2 restart" followed by another reboot. Now the warning in Admin/Overview switched to all checks passed. I didn´t change anything in configuration files at all.
So it seemed to me, that not only a reboot is needed but **additionally** a manual apache restart.
Does apache has some kind of cache to survive a reboot without reloading its conf?

Issue closed for me, too.