nextcloud/nextcloudpi #735: freeDNS update does not work

KUGA2, 2018-12-04:
In the WebUI I enter mydomain and my hash and press run. It gets green, but the ip in the freeDNS web interface does not change. I cannot acces via my domain.

The log shows:
[ DDNS_freeDNS ]
System config value trusted_domains => 3 set to string ***
System config value overwrite.cli.url set to string https://***/
FreeDNS client is enabled

My setup:
NextCloudPi version | v0.67.5
-- | --
NextCloudPi image | NextCloudPi_11-22-18
distribution | Raspbian GNU/Linux 9 \n \l
automount | yes
USB devices | sda
datadir | /media/USBdrive/ncdata
data in SD | no
data filesystem | btrfs
data disk usage | 100M/4.6T
rootfs usage | 1.9G/7.2G
swapfile | /var/swap
dbdir | /var/lib/mysql
Nextcloud check | ok
Nextcloud version | 14.0.4.2
HTTPD service | up
PHP service | up
MariaDB service | up
Redis service | up
Postfix service | up
internet check | ok
port check 80 | open
port check 443 | open
IP | 192.168.178.10
gateway | 192.168.178.1
interface | eth0
certificates | nextcloudpi1.mooo.com
NAT loopback | no
uptime | 23:49

nachoparker, 2018-12-05:
cc @Pant

nachoparker, 2019-02-06:
asked in the Telegram group, people are using it fine so it must be a network configuration issue on your end.

Thanks for reporting

closed 2019-02-06 by nachoparker