nextcloud/nextcloudpi #737: Installation failed on Debian 9

leomanfredini, 2018-12-07:
I am having the following error in setup through script
_curl -sSL https://raw.githubusercontent.com/nextcloud/nextcloudpi/master/install.sh | bash_
```
NextCloudPi updated to version v0.67.5
Activating nc-init
Setting up a clean Nextcloud instance... wait until message 'NC init done'
Setting up database...
Setting up Nextcloud...
Nextcloud was successfully installed
Set mode for background jobs to 'cron'
System config value tempdirectory => /var/www/nextcloud/data/tmp set to empty string
sed: could not read /etc/php//cli/php.ini: File or directory not found
```
**System information**

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v0.67.5
distribution         Debian GNU/Linux 9 \n \l
automount            no
USB devices          none
datadir              /var/www/nextcloud/data
data in SD           yes
data filesystem      ext2/ext3
data disk usage      4,1G/9,0G
rootfs usage         4,1G/9,0G
swapfile             /var/swap
dbdir                /var/lib/mysql
Nextcloud check      ok
Nextcloud version    14.0.4.2
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        closed
port check 443       closed
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            enp0s3
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               11min

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": [
            "localhost"
        ],
        "datadirectory": "***REMOVED SENSITIVE VALUE***",
        "dbtype": "mysql",
        "version": "14.0.4.2",
        "overwrite.cli.url": "http:\/\/localhost",
        "dbname": "***REMOVED SENSITIVE VALUE***",
        "dbhost": "***REMOVED SENSITIVE VALUE***",
        "dbport": "",
[... 68 lines cut ...]
2018-12-06 21:40:13 139822338396608 [Note] InnoDB: 128 rollback segment(s) are active.
2018-12-06 21:40:13 139822338396608 [Note] InnoDB: Waiting for purge to start
2018-12-06 21:40:13 139822338396608 [Note] InnoDB:  Percona XtraDB (http://www.percona.com) 5.6.41-84.1 started; log sequence number 1616727
2018-12-06 21:40:13 139822338396608 [Note] Plugin 'FEEDBACK' is disabled.
2018-12-06 21:40:13 139821720758016 [Note] InnoDB: Dumping buffer pool(s) not yet started
2018-12-06 21:40:13 139822338396608 [Note] Server socket created on IP: '127.0.0.1'.
2018-12-06 21:40:13 139822338396608 [Note] /usr/sbin/mysqld: ready for connections.
Version: '10.1.37-MariaDB-0+deb9u1'  socket: '/var/run/mysqld/mysqld.sock'  port: 3306  Debian 9.6

```
</details>

<details>
<summary>Nextcloud logs</summary>

```

{"reqId":"Dy0TBGNfSQSKPfFZN4Vn","level":3,"time":"2018-12-06T23:45:01+00:00","remoteAddr":"","user":"--","app":"cron","method":"","url":"--","message":{"Exception":"Exception","Message":"Not installed","Code":0,"Trace":[{"file":"\/var\/www\/nextcloud\/lib\/base.php","line":660,"function":"checkInstalled","class":"OC","type":"::","args":[]},{"file":"\/var\/www\/nextcloud\/lib\/base.php","line":1068,"function":"init","class":"OC","type":"::","args":[]},{"file":"\/var\/www\/nextcloud\/cron.php","line":41,"args":["\/var\/www\/nextcloud\/lib\/base.php"],"function":"require_once"}],"File":"\/var\/www\/nextcloud\/lib\/base.php","Line":277,"CustomMessage":"--"},"userAgent":"--","version":"14.0.4.2"}
{"reqId":"4FBHm7vxwRdzf1QS9yvS","level":3,"time":"2018-12-06T23:49:14+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"--","message":"is_writable() expects parameter 1 to be a valid path, array given at \/var\/www\/nextcloud\/lib\/private\/TempManager.php#260","userAgent":"--","version":"14.0.4.2"}
{"reqId":"4FBHm7vxwRdzf1QS9yvS","level":2,"time":"2018-12-06T23:49:14+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Temporary directory {\"\/var\/www\/nextcloud\/data\/tmp\":\"\"} is not present or writable","userAgent":"--","version":"14.0.4.2"}
{"reqId":"3jAC4DLJsppkUhz4Kt0f","level":3,"time":"2018-12-06T23:49:20+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"--","message":"is_writable() expects parameter 1 to be a valid path, array given at \/var\/www\/nextcloud\/lib\/private\/TempManager.php#260","userAgent":"--","version":"14.0.4.2"}
{"reqId":"3jAC4DLJsppkUhz4Kt0f","level":2,"time":"2018-12-06T23:49:20+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Temporary directory {\"\/var\/www\/nextcloud\/data\/tmp\":\"\"} is not present or writable","userAgent":"--","version":"14.0.4.2"}

```
</details>

nachoparker, 2018-12-07:
Fixed in `v0.67.6`, thanks for reporting!

closed 2018-12-07 by nachoparker

leomanfredini, 2018-12-07:
After installing version 0.67.6, access to the web interface is generating HTTP 500 error
<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v0.67.6
distribution         Debian GNU/Linux 9 \n \l
automount            no
USB devices          none
datadir              /var/www/nextcloud/data
data in SD           yes
data filesystem      ext2/ext3
data disk usage      4,1G/9,0G
rootfs usage         4,1G/9,0G
swapfile             /var/swap
dbdir                /var/lib/mysql
Nextcloud check      ok
Nextcloud version    14.0.4.2
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        closed
port check 443       closed
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            enp0s3
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               1min

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "1": "10.1.1.17",
            "5": "nextcloudpi.local",
            "7": "nextcloudpi",
            "8": "nextcloudpi.lan"
        },
        "datadirectory": "***REMOVED SENSITIVE VALUE***",
        "dbtype": "mysql",
        "version": "14.0.4.2",
        "overwrite.cli.url": "http:\/\/localhost",
        "dbname": "***REMOVED SENSITIVE VALUE***",
        "dbhost": "***REMOVED SENSITIVE VALUE***",
        "dbport": "",
        "dbtableprefix": "oc_",
        "dbuser": "***REMOVED SENSITIVE VALUE***",
        "dbpassword": "***REMOVED SENSITIVE VALUE***",
        "installed": true,
        "instanceid": "***REMOVED SENSITIVE VALUE***",
        "memcache.local": "\\OC\\Memcache\\Redis",
        "memcache.locking": "\\OC\\Memcache\\Redis",
        "redis": {
            "host": "***REMOVED SENSITIVE VALUE***",
            "port": 0,
[... 79 lines cut ...]
```

{"reqId":"fhJwxNOO9kboqjvITGvz","level":2,"time":"2018-12-07T00:54:50+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Temporary directory {\"\/var\/www\/nextcloud\/data\/tmp\":\"\"} is not present or writable","userAgent":"--","version":"14.0.4.2"}
{"reqId":"cN0k41mqOYRrJhkLdo42","level":3,"time":"2018-12-07T00:54:51+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"--","message":"is_writable() expects parameter 1 to be a valid path, array given at \/var\/www\/nextcloud\/lib\/private\/TempManager.php#260","userAgent":"--","version":"14.0.4.2"}
{"reqId":"cN0k41mqOYRrJhkLdo42","level":2,"time":"2018-12-07T00:54:51+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Temporary directory {\"\/var\/www\/nextcloud\/data\/tmp\":\"\"} is not present or writable","userAgent":"--","version":"14.0.4.2"}
{"reqId":"RU0JPOE4gCRwrjqphlha","level":3,"time":"2018-12-07T00:54:51+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"--","message":"is_writable() expects parameter 1 to be a valid path, array given at \/var\/www\/nextcloud\/lib\/private\/TempManager.php#260","userAgent":"--","version":"14.0.4.2"}
{"reqId":"RU0JPOE4gCRwrjqphlha","level":2,"time":"2018-12-07T00:54:51+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Temporary directory {\"\/var\/www\/nextcloud\/data\/tmp\":\"\"} is not present or writable","userAgent":"--","version":"14.0.4.2"}
{"reqId":"VRHMFZ6FhIqUGASrEtBv","level":3,"time":"2018-12-07T01:00:01+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"--","message":"is_writable() expects parameter 1 to be a valid path, array given at \/var\/www\/nextcloud\/lib\/private\/TempManager.php#260","userAgent":"--","version":"14.0.4.2"}
{"reqId":"VRHMFZ6FhIqUGASrEtBv","level":2,"time":"2018-12-07T01:00:01+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Temporary directory {\"\/var\/www\/nextcloud\/data\/tmp\":\"\"} is not present or writable","userAgent":"--","version":"14.0.4.2"}
{"reqId":"VRHMFZ6FhIqUGASrEtBv","level":3,"time":"2018-12-07T01:00:10+00:00","remoteAddr":"","user":"--","app":"core","method":"","url":"--","message":{"Exception":"Doctrine\\DBAL\\Exception\\DriverException","Message":"An exception occurred while executing 'SELECT `uid`, `displayname` FROM `oc_users` `u` LEFT JOIN `oc_preferences` `p` ON (`userid` = `uid`) AND (`appid` = 'settings') AND (`configkey` = 'email') WHERE (`uid`  COLLATE utf8mb4_general_ci LIKE ?) OR (`displayname`  COLLATE utf8mb4_general_ci LIKE ?) OR (`configvalue`  COLLATE utf8mb4_general_ci LIKE ?) ORDER BY `uid_lower` ASC LIMIT 500 OFFSET 0' with params [\"%%\", \"%%\", \"%%\"]:\n\nSQLSTATE[42000]: Syntax error or access violation: 1253 COLLATION 'utf8mb4_general_ci' is not valid for CHARACTER SET 'utf8'","Code":0,"Trace":[{"file":"\/var\/www\/nextcloud\/3rdparty\/doctrine\/dbal\/lib\/Doctrine\/DBAL\/DBALException.php","line":128,"function":"convertException","class":"Doctrine\\DBAL\\Driver\\AbstractMySQLDriver","type":"->","args":["An exception occurred while executing 'SELECT `uid`, `displayname` FROM `oc_users` `u` LEFT JOIN `oc_preferences` `p` ON (`userid` = `uid`) AND (`appid` = 'settings') AND (`configkey` = 'email') WHERE (`uid`  COLLAT [... cut ...]
{"reqId":"nhZpCtjfmulEpbJe93pZ","level":3,"time":"2018-12-07T01:10:05+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"--","message":"is_writable() expects parameter 1 to be a valid path, array given at \/var\/www\/nextcloud\/lib\/private\/TempManager.php#260","userAgent":"--","version":"14.0.4.2"}
{"reqId":"nhZpCtjfmulEpbJe93pZ","level":2,"time":"2018-12-07T01:10:05+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Temporary directory {\"\/var\/www\/nextcloud\/data\/tmp\":\"\"} is not present or writable","userAgent":"--","version":"14.0.4.2"}

```
</details>

ovpc, 2018-12-07:
I tried a fresh install on a spare rpi3b+ using

`curl -sSL https://raw.githubusercontent.com/nextcloud/nextcloudpi/master/install.sh | sudo bash`

Works like charm for me, just for reference output of init below. NOTE I will create a PR against devel later on to edit the message at the very end. In my view it is incomplete at best, should be

`Done.

First: visit https://this_ip/ or https://nextcloudpi.local/ to activate your instance of NC, and save the auto generated passwords. You may view or reset them anytime by using nc-admin and nc-passwd. 

Second: type 'sudo ncp-config' to further configure NCP, or access ncp-web on https://<this_ip>:4443

Note: You will have to add an exception, to bypass your browser warning when you first load the activation and :4443 pages. You can run letsencrypt to get rid of the warning if you have a (sub)domain available. 
`

........................
NextCloudPi updated to version v0.67.6
Activating nc-init
Setting up a clean Nextcloud instance... wait until message 'NC init done'
Setting up database...
Setting up Nextcloud...
Nextcloud was successfully installed
Set mode for background jobs to 'cron'
System config value tempdirectory => /var/www/nextcloud/data/tmp set to empty string
System config value mysql.utf8mb4 set to boolean true
System config value trusted_domains => 1 set to string 192.168.2.59
System config value trusted_domains => 5 set to string nextcloudpi.local
System config value trusted_domains => 7 set to string nextcloudpi
System config value trusted_domains => 8 set to string nextcloudpi.lan
System config value mail_smtpmode set to string sendmail
System config value mail_smtpauthtype set to string LOGIN
System config value mail_from_address set to string admin
System config value mail_domain set to string ownyourbits.com
calendar installed
calendar enabled
contacts installed
contacts enabled
notes installed
notes enabled
tasks installed
tasks enabled
news installed
news enabled
admin_notifications installed
admin_notifications enabled
previewgenerator installed
previewgenerator enabled
calendar enabled
contacts enabled
notes enabled
tasks enabled
news enabled
admin_notifications enabled
previewgenerator enabled
System config value overwriteprotocol set to string https
NC init done
/home/pi
Done.

Type 'sudo ncp-config' to configure NCP, or access ncp-web on https://<this_ip>:4443