nextcloud/nextcloudpi #740: Nextcloud running Php7.0 even though 7.2 was installed via ncp-update

Windyo, 2018-12-10:
### PROBLEMS

Checking the Admin panel reveals nextcloud is running under php 7.0 even though ncp-update supposedly did the migration to php7.2-common.

Update was run via ncp-update.

**System information**

<--! Paste this in GitHub report -->                                                                                                                                                                   
                                                                                                                                                                                                       
<details>                                                                                                                                                                                              
<summary>NextCloudPi diagnostics</summary>                                                                                                                                                             
                                                                                                                                                                                                       
```                                                                                                                                                                                                    
                                                                                                                                                                                                       
Cannot load Zend OPcache - it was already loaded                                                                                                                                                       
NextCloudPi version  v0.67.6                                                                                                                                                                           
NextCloudPi image    NextCloudPi_06-27-18                                                                                                                                                              
distribution         Raspbian GNU/Linux 9 \n \l                                                                                                                                                        
automount            no                                                                                                                                                                                
USB devices          sda sdb                                                                                                                                                                           
datadir              /mnt/raid0/ncdata                                                                                                                                                                 
data in SD           no                                                                                                                                                                                
data filesystem      ext2/ext3                                                                                                                                                                         
data disk usage      1.8T/2.7T                                                                                                                                                                         
rootfs usage         2.0G/15G                                                                                                                                                                          
swapfile             /mnt/raid0/swap                                                                                                                                                                   
dbdir                /mnt/raid0/ncdb                                                                                                                                                                   
Nextcloud check      ok                                                                                                                                                                                
Nextcloud version    15.0.0.10                                                                                                                                                                         
HTTPD service        up                                                                                                                                                                                
PHP service          up                                                                                                                                                                                
MariaDB service      up                                                                                                                                                                                
Redis service        up                                                                                                                                                                                
Postfix service      up                                                                                                                                                                                
internet check       ok                                                                                                                                                                                
port check 80        open                                                                                                                                                                              
port check 443       open                                                                                                                                                                              
IP                   ***REMOVED SENSITIVE VALUE***                                                                                                                                                     
gateway              ***REMOVED SENSITIVE VALUE***                                                                                                                                                     
interface            eth0                                                                                                                                                                              
certificates         ***REMOVED SENSITIVE VALUE***                                                                                                                                                     
NAT loopback         no                                                                                                                                                                                
uptime               24min                                                                                                                                                                             
                                                                                                                                                                                                       
```                                                                                                                                                                                                    
</details>                                                                                                                                                                                             
                                                                                                                                                                                                       
                                                                                                                                                                                                       
[... cut ...]
mbi' (Remote IP: '90.187.91.81')","userAgent":"Mozilla\/5.0 (Android) ownCloud-android\/3.4.0 RC1","version":"15.0.0.10"}                                                                              
                                                                                                                                                                                                       
```                                                                                                                                                                                                    
</details>

Windyo, 2018-12-10:
Applied instructions 

```
sudo su
apt-get purge -y php7.0* php7.2*
rm -rf /etc/php/7.2
ncp-update
```
from #669.

Currently have no version of php-fpm installed as a result, which borks NCP.

Funnily enough NC does rend, though it fails to serve a valid 500 page: output is 
```* @author Joas Schilling * @author JÃ¶rn Friedrich Dreyer * @author Lukas Reschke * @author Morris Jobke * @author Robin Appelman * @author Sergio BertolÃ­n * @author Thomas MÃ¼ller * @author Vincent Petry * * @license AGPL-3.0 * * This code is free software: you can redistribute it and/or modify * it under the terms of the GNU Affero General Public License, version 3, * as published by the Free Software Foundation. * * This program is distributed in the hope that it will be useful, * but WITHOUT ANY WARRANTY; without even the implied warranty of * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE. See the * GNU Affero General Public License for more details. * * You should have received a copy of the GNU Affero General Public License, version 3, * along with this program. If not, see * */ require_once __DIR__ . '/lib/versioncheck.php'; try { require_once __DIR__ . '/lib/base.php'; OC::handleRequest(); } catch(\OC\ServiceUnavailableException $ex) { \OC::$server->getLogger()->logException($ex, array('app' => 'index')); //show the user a detailed error page OC_Template::printExceptionErrorPage($ex, 503); } catch (\OC\HintException $ex) { try { OC_Template::printErrorPage($ex->getMessage(), $ex->getHint(), 503); } catch (Exception $ex2) { try { \OC::$server->getLogger()->logException($ex, array('app' => 'index')); \OC::$server->getLogger()->logException($ex2, array('app' => 'index')); } catch (Throwable $e) { // no way to log it properly - but to avoid a white page of death we try harder and ignore this one here } //show the user a detailed error page OC_Template::printExceptionErrorPage($ex, 500); } } catch (\OC\User\LoginException $ex) { OC_Template::printErrorPage($ex->getMessage(), $ex->getMessage(), 403); } catch (Exception $ex) { \OC::$server->getLogger()->logException($ex, array('app' => 'index')); //show the user a detailed error page OC_Template::printExceptionErrorPage($ex, 500); } catch (Error $ex) { try { \OC::$server->getLogger()->logException($ex, array('app' => 'index')); } catch (Error $e) { http_response_code(500); header('Content-Type: text/plain; charset=utf-8'); print("Internal Server Error\n\n"); print("The server encountered an internal error and was unable to complete your request.\n"); print("Please contact the server administrator if this error reappears multiple times, please include the technical details below in your report.\n"); print("More details can be found in the webserver log.\n"); throw $ex; } OC_Template::printExceptionErrorPage($ex, 500); } ```

when navigating to the page.

Windyo, 2018-12-10:
OK I tried reinstalling php 7.0 but no cigar. Seems i fucked up the instance so bad it's ridiculous.

I'll reimage to my backup from this morning - bringing us back to the original status : instance still using php 7.0

moejoeblue, 2018-12-10:
I have the same problem and end up with the same screen. I have the impression, it has something to do with php7.3 and php7.2 not being (re)installed correctly.

Windyo, 2018-12-10:
yeah no idea; seems it's serving some static page instead of actually going through php. noooooooooo idea how to fix this.

Windyo, 2018-12-17:
Thanks for the help - but same result. Apache serves the PHP file with no interpretation at all

```* @author Joas Schilling * @author JÃ¶rn Friedrich Dreyer * @author Lukas Reschke * @author Morris Jobke * @author Robin Appelman * @author Sergio BertolÃ­n * @author Thomas MÃ¼ller * @author Vincent Petry * * @license AGPL-3.0 * * This code is free software: you can redistribute it and/or modify * it under the terms of the GNU Affero General Public License, version 3, * as published by the Free Software Foundation. * * This program is distributed in the hope that it will be useful, * but WITHOUT ANY WARRANTY; without even the implied warranty of * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE. See the * GNU Affero General Public License for more details. * * You should have received a copy of the GNU Affero General Public License, version 3, * along with this program. If not, see * */ require_once __DIR__ . '/lib/versioncheck.php'; try { require_once __DIR__ . '/lib/base.php'; OC::handleRequest(); } catch(\OC\ServiceUnavailableException $ex) { \OC::$server->getLogger()->logException($ex, array('app' => 'index')); //show the user a detailed error page OC_Template::printExceptionErrorPage($ex, 503); } catch (\OC\HintException $ex) { try { OC_Template::printErrorPage($ex->getMessage(), $ex->getHint(), 503); } catch (Exception $ex2) { try { \OC::$server->getLogger()->logException($ex, array('app' => 'index')); \OC::$server->getLogger()->logException($ex2, array('app' => 'index')); } catch (Throwable $e) { // no way to log it properly - but to avoid a white page of death we try harder and ignore this one here } //show the user a detailed error page OC_Template::printExceptionErrorPage($ex, 500); } } catch (\OC\User\LoginException $ex) { OC_Template::printErrorPage($ex->getMessage(), $ex->getMessage(), 403); } catch (Exception $ex) { \OC::$server->getLogger()->logException($ex, array('app' => 'index')); //show the user a detailed error page OC_Template::printExceptionErrorPage($ex, 500); } catch (Error $ex) { try { \OC::$server->getLogger()->logException($ex, array('app' => 'index')); } catch (Error $e) { http_response_code(500); header('Content-Type: text/plain; charset=utf-8'); print("Internal Server Error\n\n"); print("The server encountered an internal error and was unable to complete your request.\n"); print("Please contact the server administrator if this error reappears multiple times, please include the technical details below in your report.\n"); print("More details can be found in the webserver log.\n"); throw $ex; } OC_Template::printExceptionErrorPage($ex, 500); } ```

Windyo, 2018-12-18:
So actually it was missing a few packages, and a few instructions.

So if anyone's still stuck
- after installing php 7.2 you need to activate it via `sudo a2enconf php7.2-fpm`
- a few packages are missing from Nacho's list. I got lazy so I just did `sudo apt-get install php7.2* -Y`

If some packages conflict just go the lazy route and script it:
```list=$(apt-cache --names-only search ^php7.2-* | awk '{ print $1 }' | grep -v php-apcu | grep -v php-gmagick | grep -v php-yac)
sudo apt-get install $list```

I'm pretty sure I installed a gazillion things that are unnecessary but now they work.

ovpc, 2018-12-18:
Think they are all listed (maybe you missed the slider below the code?)

> sudo su
apt-get purge -y php7.0* php7.2*
rm -rf /etc/php/7.2
apt-get update
apt-get install -y php7.2 php7.2-curl php7.2-gd php7.2-fpm php7.2-cli php7.2-opcache                         php7.2-mbstring php7.2-xml php7.2-zip php7.2-fileinfo php7.2-ldap
php7.2-intl php7.2-bz2 php7.2-json

Windyo, 2018-12-18:
@ovpc no I actually selected all. I can confirm all the packages above were listed as installed and that the error persisted. got fixed when I did the full yolo install, so something is missing.

One that was missing and I know is mandatory for example was php7.2-mysql

Windyo, 2018-12-18:
Ok so I got it narrowed down. What's missing were :
php7.2-mysql
php-redis
libapache2-mod-php7.2

I'll fully purge PHP again and reinstall to confirm it works with no issues with this.

Windyo, 2018-12-18:
Yeap, confirmed.

This one liner installs everything needed, activates php 7.2, and refreshes what is needed:

`apt-get install -y php7.2 php7.2-curl php7.2-gd php7.2-fpm php7.2-cli php7.2-opcache php7.2-mbstring php7.2-xml php7.2-zip php7.2-fileinfo php7.2-ldap php7.2-intl php7.2-bz2 php7.2-json libapache2-mod-php7.2 php-imagick php-redis php7.2-mysql php7.2-common && a2enconf php7.2-fpm && systemctl reload apache2`

Closing ticket.

Windyo, 2018-12-18:
(Thanks @ovpc and @nachoparker for the pointers and the help)

closed 2018-12-18 by Windyo