nextcloud/nextcloudpi #742: Adminer managing database

feutl, 2018-12-12:
It would be great to have something like Adminer (https://www.adminer.org/) being added to NCP with auto login.
Sometimes it is just handy to have a webui to access the database to figure something out.

And in my case, especially, because I am unable to login to the mysql server because I just do not know the password ;) which was set by nextcloudpi when I installed it month ago.

Thanks

enoch85, 2018-12-12:
https://github.com/nextcloud/vm/blob/master/apps/adminer.sh

enoch85, 2018-12-13:
> Who is to implement and maintain it?

As I'm the maintainer of the [Nextcloud VM](https://github.com/nextcloud/vm) I will maintain all the scripts in that repo. But, I can't promise this will work out of the box in NCP since the VM is running Ubuntu Server 18.04, not Debian.

nachoparker, 2018-12-13:
Hi all,

Thanks for the suggestions and links. In my opinion I don't see this as a very useful addition to NCP.

Not only we would have to be very careful with security, but also the target user of NCP is a person without technical expertise and managing the database I don't see as a feature that adds towards that goal.

What kind of things do you guys use adminer for? I am probably missing something.

enoch85, 2018-12-14:
> What kind of things do you guys use adminer for?

It's much easier to export a database with a GUI than from CLI.

And, since phpPGmyAdmin has many security flaws (since we use PostgreSQL), I decided to move to Adminer which is easy to install, maintain and use.

nachoparker, 2018-12-15:
@enoch85 interesting, thanks for explaining. 

@feutl We already have ncp-tools to backup the NC instance from the web interface, so I don't see much benefit in this.

Thanks for the suggestion

closed 2018-12-15 by nachoparker

feutl, 2018-12-15:
the reason why I have it installed, is not to make a database backup.
I needed to do some troubleshooting on the database and also wanted to have some "insight" how data gets stored (especially passwords for some plugins) and it took me one curl command and a mv to rename it and use it insight the ncp folder which is accessible using 4443.
Using 4443 instead of 443 also made it "easier" to deny access from the internet and limit the attack surface.

Anyhow, for me it was just a curl and mv command to have this up and running, so no big hazzle, but others could find it useful and the effort is quite small to get that integrated.

but as you like, nextcloudpi is still my favorit on my "non-pi" installation :dancer:

nachoparker, 2018-12-15:
> but as you like, nextcloudpi is still my favorit on my "non-pi" installation dancer

thanks!

Yes, that is a pretty special use case, and I think that if somebody wants to troubleshoot the database they are technically savy enough to install this themselves ;)