nextcloud/nextcloudpi #743: Postfix attempting to send email to admin@ownyourbits.com even though NextCloud is configured to email through SMTP (gmail)

dbrandesky, 2018-12-13:
Using NCP 0.67.6 with NextCloud 14.0.4.2 from NextCloudPi_11-22-18 image. I configured email through SMTP in NextCloud and have confirmed that I can get emails from NextCloud. In the relevant ncp-config dialogs, I entered the same email address I used in the NextCloud SMTP settings so I could receive those notifications, however I don't think they're getting sent via SMTP. My syslog has a ton of messages showing failures with postfix to ownyourbits.com:

`Dec 13 06:32:05 pi-cloud postfix/smtp[4716]: connect to ownyourbits.com[185.162.171.236]:25: Connection timed out
Dec 13 06:32:06 pi-cloud postfix/smtp[4716]: F30B21EFCA: to=<admin@ownyourbits.com>, relay=none, delay=29971, delays=29941/0.09/30/0, dsn=4.4.1, status=deferred (connect to ownyourbits.com[185.162.171.236]:25: Connection timed out)
Dec 13 07:02:06 pi-cloud postfix/smtp[5124]: connect to ownyourbits.com[185.162.171.236]:25: Connection timed out
Dec 13 07:02:06 pi-cloud postfix/smtp[5124]: 0C5CD1EF2F: to=<admin@ownyourbits.com>, relay=none, delay=34141, delays=34111/0.1/30/0, dsn=4.4.1, status=deferred (connect to ownyourbits.com[185.162.171.236]:25: Connection timed out)
Dec 13 07:42:06 pi-cloud postfix/smtp[5621]: connect to ownyourbits.com[185.162.171.236]:25: Connection timed out
Dec 13 07:42:06 pi-cloud postfix/smtp[5621]: F30B21EFCA: to=<admin@ownyourbits.com>, relay=none, delay=34171, delays=34141/0.05/30/0, dsn=4.4.1, status=deferred (connect to ownyourbits.com[185.162.171.236]:25: Connection timed out)
Dec 13 08:12:06 pi-cloud postfix/smtp[5931]: connect to ownyourbits.com[185.162.171.236]:25: Connection timed out
Dec 13 08:12:06 pi-cloud postfix/smtp[5931]: 0C5CD1EF2F: to=<admin@ownyourbits.com>, relay=none, delay=38342, delays=38311/0.09/30/0, dsn=4.4.1, status=deferred (connect to ownyourbits.com[185.162.171.236]:25: Connection timed out)
Dec 13 08:52:07 pi-cloud postfix/smtp[6409]: connect to ownyourbits.com[185.162.171.236]:25: Connection timed out
Dec 13 08:52:07 pi-cloud postfix/smtp[6409]: F30B21EFCA: to=<admin@ownyourbits.com>, relay=none, delay=38373, delays=38342/0.1/30/0, dsn=4.4.1, status=deferred (connect to ownyourbits.com[185.162.171.236]:25: Connection timed out)
Dec 13 09:22:07 pi-cloud postfix/smtp[6874]: connect to ownyourbits.com[185.162.171.236]:25: Connection timed out
Dec 13 09:22:07 pi-cloud postfix/smtp[6874]: 0C5CD1EF2F: to=<admin@ownyourbits.com>, relay=none, delay=42543, delays=42512/0.11/30/0, dsn=4.4.1, status=deferred (connect to ownyourbits.com[185.162.171.236]:25: Connection timed out)
Dec 13 10:02:08 pi-cloud postfix/smtp[7885]: connect to ownyourbits.com[185.162.171.236]:25: Connection timed out
Dec 13 10:02:08 pi-cloud postfix/smtp[7885]: F30B21EFCA: to=<admin@ownyourbits.com>, relay=none, delay=42573, delays=42543/0.1/30/0, dsn=4.4.1, status=deferred (connect to ownyourbits.com[185.162.171.236]:25: Connection timed out)
Dec 13 10:32:08 pi-cloud postfix/smtp[8803]: connect to ownyourbits.com[185.162.171.236]:25: Connection timed out
Dec 13 10:32:09 pi-cloud postfix/smtp[8803]: 0C5CD1EF2F: to=<admin@ownyourbits.com>, relay=none, delay=46744, delays=46714/0.1/30/0, dsn=4.4.1, status=deferred (connect to ownyourbits.com[185.162.171.236]:25: Connection timed out)
`

I'm not sure where to start with troubleshooting this, I didn't find much information in the wiki that specified how the actual ncp-config utilities sent emails so it's possible I'm missing something there.

shadowstreik, 2018-12-13:
I'll toss a faithful 'second' on this issue.

shadowstreik, 2018-12-14:
Could you tell me what command brought up those results, whether just a path or a specific command? I believe I have a similar issue and would like to compare.

Thank you.

dbrandesky, 2018-12-14:
I believe I ran
`cat /var/log/syslog | grep ownyourbits`
Or something along those lines. Basically printed the output of syslog to the terminal, but only those lines with "ownyourbits" in them.

andydld, 2019-01-15:
Check or change these lines in "/var/www/nextcloud/config/config.php":

```
  'mail_smtpmode' => 'sendmail',
  'mail_smtpauthtype' => 'LOGIN',
  'mail_from_address' => 'admin',
  'mail_domain' => 'ownyourbits.com',

```

andydld, 2019-01-15:
SMTP is good for the Nextcloud-Mails like welcome for new users or the forgot-password-function and if you want to use an provider-mailserver. 

The lines from my previous post are the defaults. You can change them from the Nextcloud Admin or direct within the config.php. Alternatively you can change the postfix configuration via ssh or console with

`dpkg-reconfigure postfix`

Atm i think checking/changing the postfix-config could be the best solution for you. Because then you can send mails from Nextcloud and the OS/system-wide, too.

You can check the config-files within "/etc/postfix" and check the "/etc/mailname", too.

Based on your log i'm wondering why NCP tries to connect to the server "ownyourbits.com":

`Dec 13 10:32:09 pi-cloud postfix/smtp[8803]: 0C5CD1EF2F: to=<admin@ownyourbits.com>, relay=none, delay=46744, delays=46714/0.1/30/0, dsn=4.4.1, status=deferred (connect to ownyourbits.com[185.162.171.236]:25: Connection timed out)`

I'm new to NCP, but with the current version 1.3.6 and out-of-the-box NCP tries to directly send mail with SMTP to the receipent's server.

You can check the log under "/var/log/mail.log", too.

Where within ncp-config or the ncp-webpanel did you find mail-settings?

nachoparker, 2019-01-16:
That's just the defaults. They can be changed.

Please use the forums in the future for questions

Thanks

closed 2019-01-16 by nachoparker