nextcloud/nextcloudpi #744: set dynDNS update interval

TobiasJu, 2018-12-14:
Hi,
it would be nice if we can set the dynDNS update interval in the NCP Web Panel.

For now i tried it myself but failed, I searched on the RaspPi in the /etc/ddclient.conf and in /etc/cron but i couldn't finde the file which is engaging the update. For example, in the default/ddclient file there is:
```
run_daemon="true"
daemon_interval="300"
```
Which should result in a 5 minutes update interval, but my dynDNS (spdyn.de) only updates once per hour, or in some cases, once all two hours. Which leads to downtimes up to 2 hours per day.

best regards

TomTurnschuh, 2018-12-26:
That's a good issue, thanks.
By now, dynDNS indeed updated only once per hour (at 10 minutes after full hour), set in the cronfile.

It will be changed in one of the next updates to five minutes.
I hope, you can live with this fix update interval.

TomTurnschuh, 2018-12-26:
Be aware that IP updates can take longer when using IPv6. However, this depends on dynDNS.

nachoparker, 2018-12-27:
thanks @TomTurnschuh

closed 2018-12-27 by nachoparker

TobiasJu, 2018-12-28:
> Be aware that IP updates can take longer when using IPv6. However, this depends on dynDNS.
Yeah, but in Germany our Internet sucks pretty hard. So IPv6 is nearly non existent here. So thats not an issue.

> It will be changed in one of the next updates to five minutes.
> I hope, you can live with this fix update interval.
Yes i can, thanks!