nextcloud/nextcloudpi #747: [SMB] New user is not visible in the file manager (nautilus)

prog-amateur, 2018-12-20:
Hello, I have opened a new user account for my Nextcloudpi server. Samba is enabled and I can see all users from Nautilus (Ubuntu file explorer), all except this new user. 

Through SSH, in `/etc/samba`, `nano smb.conf` does not show the `[ncp-user]` part (everybody else is there). 

As a result, I don't have access to his SMB directory and have access to all others (I am the proprietary of all those accounts). If I use the website or the Android app (by using username + password), I have access to the account normally. So only SMB is a problem with this specific user. 

What do you think happened and how can we fix it, please? 

Thank you very much!

nachoparker, 2018-12-21:
Just ran samba again and it should show up.

Please, use the forum for questions!

closed 2018-12-21 by nachoparker

nachoparker, 2018-12-23:
Yes, that is a limitation