nextcloud/nextcloudpi #748: Bug in DDNS_spDYN.sh script

Serviton, 2018-12-21:
NextCloudPi v0.67.8

Hi, I discoverd some problems when I use the NextCloudPi Configuration tool to configure DDNS_spDYN it says:

[badauth] Not a valid username/password

when I try ‘the same’ with the spdnsUpdater.sh script, with the same hostname and updatetoken, everything works fine.

Found the Bug in the etc/ncp-config.d DDNS_spDYN.sh
script.

Line 112: “$INSTALLPATH”/spdnsUpdater.sh “$DOMAIN_” “$TOKEN_” “$IPv6_”
Here the update Script is called, but with 3 Parameters.
In the update Script is a line which checks all parameters, in case of 3 it will expect $DOMAIN $USERNAME $PASSWORD
There is no support for a IPv6 Parameter.

Solution: removed the “IPv6_” Parameter, everything works fine now. :slight_smile:

It would be great if a fix or correct IPv6 implementation will be made to the script.

Greetings
Serviton

nachoparker, 2018-12-22:
cc @TomTurnschuh

TomTurnschuh, 2018-12-26:
You must have an old version of this script. The mentioned `case` section has been removed.
However, the automatic reinstallation has been removed during the latest cleanup of the update script.

@Serviton: Could you please run the following code as root user?:
```
rm -r /usr/local/etc/spdnsupdater
cd /usr/local/etc/ncp-config.d
install_script DDNS_spDYN.sh
```

Serviton, 2018-12-27:
Hi Tom,
I tried to use the 'install_script' command but then I got:
Command not found.
unfortunately I am not that familiar with the use of these scripts

closed 2018-12-27 by Serviton

TomTurnschuh, 2018-12-27:
Hi Serviton

Sorry, my mistake.
The following code should work:
```
rm -r /usr/local/etc/spdnsupdater
cd /usr/local/etc/ncp-config.d
source DDNS_spDYN.sh
install
```

Serviton, 2018-12-27:
No problem, after a quick search my way was a little bit different, I used:

  `source library.sh`
 ` install_script DDNS_spDYN.sh `