nextcloud/nextcloudpi #755: Can't view list of users (SQL error)

benthehuman256, 2018-12-28:
ncp-report output:

NextCloudPi version  v0.67.8
distribution         Debian GNU/Linux 9 \n \l
automount            no
USB devices          none
datadir              /var/www/nextcloud/data
data in SD           yes
data filesystem      ext2/ext3
data disk usage      4.6G/212G
rootfs usage         4.6G/212G
swapfile             /dev/zram3
dbdir                /var/lib/mysql
Nextcloud check      ok
Nextcloud version    14.0.4.2
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            eno1
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               52min

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "1": "192.168.1.5",
            "5": "nextcloudpi.local",
            "7": "nextcloudpi",
            "8": "nextcloudpi.lan",
            "9": "geminicloud.co.uk"
        },
        "datadirectory": "***REMOVED SENSITIVE VALUE***",
        "dbtype": "mysql",
        "version": "14.0.4.2",
        "overwrite.cli.url": "https:\/\/geminicloud.co.uk\/",
        "dbname": "***REMOVED SENSITIVE VALUE***",
        "dbhost": "***REMOVED SENSITIVE VALUE***",
        "dbport": "",
        "dbtableprefix": "oc_",
        "dbuser": "***REMOVED SENSITIVE VALUE***",
        "dbpassword": "***REMOVED SENSITIVE VALUE***",
        "installed": true,
        "instanceid": "***REMOVED SENSITIVE VALUE***",
        "memcache.local": "\\OC\\Memcache\\Redis",
        "memcache.locking": "\\OC\\Memcache\\Redis",
        "redis": {
            "host": "***REMOVED SENSITIVE VALUE***",
            "port": 0,
            "timeout": 0,
            "password": "***REMOVED SENSITIVE VALUE***"
        },
[... 63 lines cut ...]
2018-12-28 17:23:27 139998581432768 [Note] InnoDB: Waiting for purge to start
2018-12-28 17:23:27 139998581432768 [Note] InnoDB:  Percona XtraDB (http://www.percona.com) 5.6.41-84.1 started; log sequence number 1616727
2018-12-28 17:23:27 139998581432768 [Note] Plugin 'FEEDBACK' is disabled.
2018-12-28 17:23:27 139997919311616 [Note] InnoDB: Dumping buffer pool(s) not yet started
2018-12-28 17:23:27 139998581432768 [Note] Server socket created on IP: '127.0.0.1'.
2018-12-28 17:23:27 139998581432768 [Note] /usr/sbin/mysqld: ready for connections.
Version: '10.1.37-MariaDB-0+deb9u1'  socket: '/var/run/mysqld/mysqld.sock'  port: 3306  Debian 9.6

```
</details>

<details>
<summary>Nextcloud logs</summary>

```

{"reqId":"uhcBFWJ2AtEyRjbhna2n","level":3,"time":"2018-12-28T17:30:01+00:00","remoteAddr":"","user":"--","app":"core","method":"","url":"--","message":{"Exception":"Doctrine\\DBAL\\Exception\\DriverException","Message":"An exception occurred while executing 'SELECT `uid`, `displayname` FROM `oc_users` `u` LEFT JOIN `oc_preferences` `p` ON (`userid` = `uid`) AND (`appid` = 'settings') AND (`configkey` = 'email') WHERE (`uid`  COLLATE utf8mb4_general_ci LIKE ?) OR (`displayname`  COLLATE utf8mb4_general_ci LIKE ?) OR (`configvalue`  COLLATE utf8mb4_general_ci LIKE ?) ORDER BY `uid_lower` ASC LIMIT 500 OFFSET 0' with params [\"%%\", \"%%\", \"%%\"]:\n\nSQLSTATE[42000]: Syntax error or access violation: 1253 COLLATION 'utf8mb4_general_ci' is not valid for CHARACTER SET 'utf8'","Code":0,"Trace":[{"file":"\/var\/www\/nextcloud\/3rdparty\/doctrine\/dbal\/lib\/Doctrine\/DBAL\/DBALException.php","line":128,"function":"convertException","class":"Doctrine\\DBAL\\Driver\\AbstractMySQLDriver","type":"->","args":["An exception occurred while executing 'SELECT `uid`, `displayname` FROM `oc_users` `u` LEFT JOIN `oc_preferences` `p` ON (`userid` = `uid`) AND (`appid` = 'settings') AND (`configkey` = 'email') WHERE (`uid`  COLLATE utf8mb4_general_ci LIKE ?) OR (`displayname`  COLLATE utf8mb4_general_ci LIKE ?) OR (`configvalue`  COLLATE utf8mb4_general_ci LIKE ?) ORDER BY `uid_lower` ASC LIMIT 500 OFFSET 0' with params [\"%%\", \"%%\", \"%%\"]:\n\nSQLSTATE[42000]: Syntax error or access violation: 1253 COLLATION 'utf8mb4_general_ci' is not valid for CHARACTER SET 'utf8'",{"errorInfo":["42000",1253,"COLLATION 'utf8mb4_general_ci' is not valid for CHARACTER SET 'utf8'"],"__class__":"Doctrine\\DBAL\\Driver\\PDOException"}]},{"file":"\/var\/www\/nextcloud\/3rdparty\/doctrine\/dbal\/lib\/Doctrine\/DBAL\/Connection.php","line":855,"function":"driverExceptionDuringQuery","class":"Doctrine\\DBAL\\DBALException","type":"::","args":[{"__class__":"Doctrine\\DBAL\\Driver\\PDOMySql\\Driver"},{"errorInfo":["42000",1253,"COLLATION 'utf8mb4_general_ci' is not valid for CHARACTER SET 'utf8'"],"__class__":"Doctrine\\DBAL\\Driver\\PDOException"},"SELECT `uid`, `displayname` FROM `oc_users` `u` LEFT JOIN `oc_preferences` `p` ON (`userid` = `uid`) AND (`appid` = 'settings') AND (`configkey` = 'email') WHERE (`uid`  COLLATE utf8mb4_general_ci LIKE ?) OR (`displayname`  COLLATE utf8mb4_general_ci LIKE ?) OR (`configvalue`  COLLATE utf8mb4_general_ci LIKE ?) ORDER BY `uid_lower` ASC LIMIT 500 OFFSET 0",{"1":"%%","2":"%%","3":"%%"}]},{"file":"\/var\/www\/nextcloud\/lib\/private\/DB\/Connection.php","line":195,"function":"executeQuery","class":"Doctrine\\DBAL\\Connection","type":"->","args":["SELECT `uid`, `displayname` FROM `oc_users` `u` LEFT JOIN `oc_preferences` `p` ON (`userid` = `uid`) AND (`appid` = 'settings') AND (`configkey` = 'email') WHERE (`uid`  COLLATE utf8mb4_general_ci LIKE ?) OR (`displayname`  COLLATE utf8mb4_general_ci LIKE ?) OR (`configvalue`  COLLATE utf8mb4_general_ci LIKE ?) ORDER BY `uid_lower` ASC LIMIT 500 OFFSET 0",{"1":"%%","2":"%%","3":"%%"},{"1":2,"2":2,"3":2},null]},{"file":"\/var\/www\/nextcloud\/3rdparty\/doctrine\/dbal\/lib\/Doctrine\/DBAL\/Query\/QueryBuilder.php","line":206,"function":"executeQuery","class":"OC\\DB\\Connection","type":"->","args":["SELECT `uid`, `displayname` FROM `oc_users` `u` LEFT JOIN `oc_preferences` `p` ON (`userid` = `uid`) AND (`appid` = 'settings') AND (`configkey` = 'email') WHERE (`uid`  COLLATE utf8mb4_general_ci LIKE ?) OR (`displayname`  COLLATE utf8mb4_general_ci LIKE ?) OR (`configvalue`  COLLATE utf8mb4_general_ci LIKE ?) ORDER BY `uid_lower` ASC LIMIT 500 OFFSET 0",{"1":"%%","2":"%%","3":"%%"},{"1":2,"2":2,"3":2}]},{"file":"\/var\/www\/nextcloud\/lib\/private\/DB\/QueryBuilder\/QueryBuilder.php","line":214,"function":"execute","class":"Doctrine\\DBAL\\Query\\QueryBuilder","type":"->","args":[]},{"file":"\/var\/www\/nextcloud\/lib\/private\/User\/Database.php","line":282,"function":"execute","class":"OC\\DB\\QueryBuilder\\QueryBuilder","type":"->","args":[]},{"file":"\/var\/www\/nextcloud\/lib\/private\/User\/Database.php","line":383,"function":"getDisplayNames","class":"OC\\User\\Database","type":"->","args":["",500,0]},{"file":"\/var\/www\/nextcloud\/lib\/private\/User\/Manager.php","line":229,"function":"getUsers","class":"OC\\User\\Database","type":"->","args":["",500,0]},{"file":"\/var\/www\/nextcloud\/apps\/files\/lib\/BackgroundJob\/ScanFiles.php","line":101,"function":"search","class":"OC\\User\\Manager","type":"->","args":["",500,0]},{"file":"\/var\/www\/nextcloud\/lib\/private\/BackgroundJob\/Job.php","line":61,"function":"run","class":"OCA\\Files\\BackgroundJob\\ScanFiles","type":"->","args":[null]},{"file":"\/var\/www\/nextcloud\/lib\/private\/BackgroundJob\/TimedJob.php","line":55,"function":"execute","class":"OC\\BackgroundJob\\Job","type":"->","args":[{"__class__":"OC\\BackgroundJob\\JobList"},{"__class__":"OC\\Log"}]},{"file":"\/var\/www\/nextcloud\/cron.php","line":123,"function":"execute","class":"OC\\BackgroundJob\\TimedJob","type":"->","args":[{"__class__":"OC\\BackgroundJob\\JobList"},{"__class__":"OC\\Log"}]}],"File":"\/var\/www\/nextcloud\/3rdparty\/doctrine\/dbal\/lib\/Doctrine\/DBAL\/Driver\/AbstractMySQLDriver.php","Line":115,"Previous":{"Exception":"Doctrine\\DBAL\\Driver\\PDOException","Message":"SQLSTATE[42000]: Syntax error or access violation: 1253 COLLATION 'utf8mb4_general_ci' is not valid for CHARACTER SET 'utf8'","Code":"42000","Trace":[{"file":"\/var\/www\/nextcloud\/3rdparty\/doctrine\/dbal\/lib\/Doctrine\/DBAL\/Connection.php","line":847,"function":"execute","class":"Doct [... cut ...] hp","line":82,"function":"match","class":"OC\\Route\\Router","type":"->","args":["\/ocsapp\/cloud\/users\/details"]},{"file":"\/var\/www\/nextcloud\/ocs\/v2.php","line":24,"args":["\/var\/www\/nextcloud\/ocs\/v1.php"],"function":"require_once"}],"File":"\/var\/www\/nextcloud\/3rdparty\/doctrine\/dbal\/lib\/Doctrine\/DBAL\/Driver\/PDOStatement.php","Line":105}},"CustomMessage":"--"},"userAgent":"Mozilla\/5.0 (X11; Ubuntu; Linux x86_64; rv:64.0) Gecko\/20100101 Firefox\/64.0","version":"14.0.4.2"}

nachoparker, 2018-12-28:
did you install through the "curl installer"?

benthehuman256, 2018-12-28:
@nachoparker yes I installed through curl installer, I am on Debian 9.6

benthehuman256, 2018-12-28:
It is a custom HP blade server

nachoparker, 2019-01-07:
fixed in `devel` -> https://github.com/nextcloud/nextcloudpi/commit/e58c633ec12457e8eda95d963927781de36aa3e5

It will land in `v1.0.0`

nachoparker, 2019-01-07:
fixed

closed 2019-01-07 by nachoparker