nextcloud/nextcloudpi #756: ERROR: log2ram service not found!

TomTurnschuh, 2018-12-28:
New installation on Raspberry Pi, updated system and NCP.
Trying to activate ramlogs.

<details>
<summary>Log</summary>
```
[ nc-ramlogs ]
ERROR: log2ram service not found!
*** Error in `systemctl': free(): invalid pointer: 0x76cd47c4 ***
./nc-ramlogs.sh: line 53: 12109 Aborted systemctl enable "$UNIT_NAME"
Failed to mangle name: Invalid argument
Failed to expand names: Invalid argument
Logs in RAM. Reboot to take effect
```
</details>

nachoparker, 2018-12-29:
any ideas @theCalcaholic ?

nachoparker, 2019-01-15:
@ovpc would you be so kind to try to activate `nc-ramlogs` on your pi, see if you can reproduce this?

ovpc, 2019-01-15:
> Running nc-ramlogs
Created symlink /etc/systemd/system/sysinit.target.wants/log2ram.service → /etc/systemd/system/log2ram.service.
Logs in RAM. Reboot to take effect
Done. Press any key...

> pi@pibox:~ $ df -h
Filesystem      Size  Used Avail Use% Mounted on
dev             460M     0  460M   0% /dev
none            932G   43G  886G   5% /
tmpfs           469M     0  469M   0% /dev/shm
tmpfs           469M   13M  457M   3% /run
tmpfs           5,0M  4,0K  5,0M   1% /run/lock
tmpfs           469M     0  469M   0% /sys/fs/cgroup
/dev/mmcblk0p1  3,7G   42M  3,7G   2% /boot
tmpfs            94M     0   94M   0% /run/user/1000
log2ram          40M  5,9M   35M  15% /var/log

Seems to work fine, even without a reboot /var/log is mounted to ram.

nachoparker, 2019-01-15:
Thanks @ovpc, must be something like a hardware issue

closed 2019-01-15 by nachoparker