nextcloud/nextcloudpi #761: Fieled to choose IPv4 or IPv6 (spdns)

traori, 2019-01-01:
**Description**  
In the window DDNS_spDYN it have 3 Fields.
1. ACTIVE (YES/NO)
2.DOMAIN
3.TOKEN

In the script spdnsUpdater.sh is at the beginning a param. IPv6=$3.But there are nowwhere to choose.
The problem is IPv6 with DS-Lite.

**Example**  
.
.
3.TOKEN
4.IPv6 (YES/NO)
or
4.Internet Protocol (IPv4/IPv6)

@TomTurnschuh : Is this a problem to implement? I don`t know how to change the GUI of the script.

TomTurnschuh, 2019-01-01:
Hi traori

It actually is implemented – but obviously, no numbers are allowed for `ncp-config` in shell.

Thanks for the report.

Until the implementation of a solution, you can use `ncp-web` instead.

<kbd>[image]</kbd>

TomTurnschuh, 2019-01-01:
@nachoparker: Is this possible to fix this in `ncp-config` by allowing numerics?

traori, 2019-01-02:
Hello,
my problem is, that the ncb-web not work and had never work for me. I don't know why. Also after fresh install it doesn't work.

nachoparker, 2019-01-02:
@TomTurnschuh I am not sure of what you mean. ncp-config and ncp-web accept numbers

TomTurnschuh, 2019-01-02:
> 
> 
> @TomTurnschuh I am not sure of what you mean. ncp-config and ncp-web accept numbers

If you open `DDNS_spDYN` in `ncp-config` (I use SSH with PuTTY), there is no option `IPv6`, as it is supposed to be and as is available in `ncp-web`. If I remove the mumber `6` in the .sh file, the option is also available in `ncp-config`.

nachoparker, 2019-01-02:
ah I see, well just wait for `v1.0.0` to land. after the rework there is no such restriction to the name of the fields.

TomTurnschuh, 2019-01-02:
@traori: As a temporary solution, you can change the respective option to `yes` in `/usr/local/etc/ncp-config.d/DDNS_spDYN.sh` line 15. Afterwards, you can use `ncp-config` for setting the other options and running the script.

nachoparker, 2019-01-16:
`v1.x` has been released

traori, 2019-03-10:
Hi,
for me don`t work. Some solution ?

nachoparker, 2019-07-07:
fixed

closed 2019-07-07 by nachoparker