nextcloud/nextcloudpi #762: Stuck in Maintenance Mode with EVERY update

codac, 2019-01-03:
Every couple of days I'm stuck in maintenance mode.
I have `nc-autoupdate-ncp` and `nc-autoupdate-nc` enabled. It seems like every couple of days an update gets installed and the system enters maintenance mode. 
However, afterwards the maintenance mode doesn't get disabled afterwards. The system is stuck in maintenance mode and so every couple of days I need to log into the system and disable it manually by editing the config.php or by osing the `sudo -u www-data php occ maintenance:mode --off`.

This is really annoying.

Used versions are:
NextCloudPi version  v0.67.11
Nextcloud version    15.0.0.10

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v0.67.11
distribution         Raspbian GNU/Linux 9 \n \l
automount            yes
USB devices          sda
datadir              /media/USBdrive/ncdata
data in SD           no
data filesystem      btrfs
data disk usage      79G/293G
rootfs usage         3.5G/30G
swapfile             /var/swap
dbdir                /var/lib/mysql
Nextcloud check      ok
Nextcloud version    15.0.0.10
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            enxb829eb23fcb3
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               2days

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "1": "192.168.0.3",
            "5": "nextcloudpi.local",
            "7": "nextcloudpi",
            "8": "nextcloudpi.lan",
            "3": "nextcloudpi.myfirewall.org",
            "4": "nextcloudpi.myfirewall.org"
        },
        "datadirectory": "***REMOVED SENSITIVE VALUE***",
        "overwrite.cli.url": "https:\/\/nextcloudpi.myfirewall.org\/",
        "dbtype": "mysql",
        "version": "15.0.0.10",
        "dbname": "***REMOVED SENSITIVE VALUE***",
        "dbhost": "***REMOVED SENSITIVE VALUE***",
[... 67 lines cut ...]
```

{"reqId":"Pjw30YVGjnZsoBPIDaLT","level":4,"time":"2019-01-02T15:46:10+00:00","remoteAddr":"","user":"--","app":"files_antivirus","method":"","url":"--","message":"OCA\\Files_Antivirus\\BackgroundJob\\BackgroundScanner::run, exception: The antivirus executable could not be found at \/usr\/bin\/clamscan","userAgent":"--","version":"15.0.0.10"}
{"reqId":"Pjw30YVGjnZsoBPIDaLT","level":4,"time":"2019-01-02T15:46:10+00:00","remoteAddr":"","user":"--","app":"files_antivirus","method":"","url":"--","message":"OCA\\Files_Antivirus\\BackgroundJob\\BackgroundScanner::run, exception: The antivirus executable could not be found at \/usr\/bin\/clamscan","userAgent":"--","version":"15.0.0.10"}
{"reqId":"Pjw30YVGjnZsoBPIDaLT","level":4,"time":"2019-01-02T15:46:10+00:00","remoteAddr":"","user":"--","app":"files_antivirus","method":"","url":"--","message":"OCA\\Files_Antivirus\\BackgroundJob\\BackgroundScanner::run, exception: The antivirus executable could not be found at \/usr\/bin\/clamscan","userAgent":"--","version":"15.0.0.10"}
{"reqId":"Pjw30YVGjnZsoBPIDaLT","level":4,"time":"2019-01-02T15:46:10+00:00","remoteAddr":"","user":"--","app":"files_antivirus","method":"","url":"--","message":"OCA\\Files_Antivirus\\BackgroundJob\\BackgroundScanner::run, exception: The antivirus executable could not be found at \/usr\/bin\/clamscan","userAgent":"--","version":"15.0.0.10"}
{"reqId":"Pjw30YVGjnZsoBPIDaLT","level":4,"time":"2019-01-02T15:46:10+00:00","remoteAddr":"","user":"--","app":"files_antivirus","method":"","url":"--","message":"OCA\\Files_Antivirus\\BackgroundJob\\BackgroundScanner::run, exception: The antivirus executable could not be found at \/usr\/bin\/clamscan","userAgent":"--","version":"15.0.0.10"}
{"reqId":"Pjw30YVGjnZsoBPIDaLT","level":4,"time":"2019-01-02T15:46:10+00:00","remoteAddr":"","user":"--","app":"files_antivirus","method":"","url":"--","message":"OCA\\Files_Antivirus\\BackgroundJob\\BackgroundScanner::run, exception: The antivirus executable could not be found at \/usr\/bin\/clamscan","userAgent":"--","version":"15.0.0.10"}
{"reqId":"Pjw30YVGjnZsoBPIDaLT","level":4,"time":"2019-01-02T15:46:10+00:00","remoteAddr":"","user":"--","app":"files_antivirus","method":"","url":"--","message":"OCA\\Files_Antivirus\\BackgroundJob\\BackgroundScanner::run, exception: The antivirus executable could not be found at \/usr\/bin\/clamscan","userAgent":"--","version":"15.0.0.10"}
{"reqId":"Pjw30YVGjnZsoBPIDaLT","level":4,"time":"2019-01-02T15:46:10+00:00","remoteAddr":"","user":"--","app":"files_antivirus","method":"","url":"--","message":"OCA\\Files_Antivirus\\BackgroundJob\\BackgroundScanner::run, exception: The antivirus executable could not be found at \/usr\/bin\/clamscan","userAgent":"--","version":"15.0.0.10"}
{"reqId":"Pjw30YVGjnZsoBPIDaLT","level":4,"time":"2019-01-02T15:46:10+00:00","remoteAddr":"","user":"--","app":"files_antivirus","method":"","url":"--","message":"OCA\\Files_Antivirus\\BackgroundJob\\BackgroundScanner::run, exception: The antivirus executable could not be found at \/usr\/bin\/clamscan","userAgent":"--","version":"15.0.0.10"}
{"reqId":"Pjw30YVGjnZsoBPIDaLT","level":4,"time":"2019-01-02T15:46:10+00:00","remoteAddr":"","user":"--","app":"files_antivirus","method":"","url":"--","message":"OCA\\Files_Antivirus\\BackgroundJob\\BackgroundScanner::run, exception: The antivirus executable could not be found at \/usr\/bin\/clamscan","userAgent":"--","version":"15.0.0.10"}
{"reqId":"Pjw30YVGjnZsoBPIDaLT","level":4,"time":"2019-01-02T15:46:10+00:00","remoteAddr":"","user":"--","app":"files_antivirus","method":"","url":"--","message":"OCA\\Files_Antivirus\\BackgroundJob\\BackgroundScanner::run, exception: The antivirus executable could not be found at \/usr\/bin\/clamscan","userAgent":"--","version":"15.0.0.10"}
{"reqId":"Pjw30YVGjnZsoBPIDaLT","level":4,"time":"2019-01-02T15:46:10+00:00","remoteAddr":"","user":"--","app":"files_antivirus","method":"","url":"--","message":"OCA\\Files_Antivirus\\BackgroundJob\\BackgroundScanner::run, exception: The antivirus executable could not be found at \/usr\/bin\/clamscan","userAgent":"--","version":"15.0.0.10"}
{"reqId":"XCzf1n8AAQEAAC@78SEAAA4B","level":2,"time":"2019-01-02T15:59:18+00:00","remoteAddr":"195.200.34.51","user":"codac","app":"core","method":"POST","url":"\/index.php\/login\/confirm","message":"Login failed: 'codac' (Remote IP: '195.200.34.51')","userAgent":"Mozilla\/5.0 (Windows NT 6.1; WOW64; rv:60.0) Gecko\/20100101 Firefox\/60.0","version":"15.0.0.10"}
{"reqId":"XCzf2X8AAQEAAC@78SQAAAkB","level":2,"time":"2019-01-02T15:59:22+00:00","remoteAddr":"195.200.34.51","user":"codac","app":"core","method":"POST","url":"\/index.php\/login\/confirm","message":"Login failed: 'codac' (Remote IP: '195.200.34.51')","userAgent":"Mozilla\/5.0 (Windows NT 6.1; WOW64; rv:60.0) Gecko\/20100101 Firefox\/60.0","version":"15.0.0.10"}
{"reqId":"XC3EKX8AAQEAAGfyelUAAEMA","level":2,"time":"2019-01-03T08:13:29+00:00","remoteAddr":"109.40.130.170","user":"--","app":"core","method":"GET","url":"\/ocs\/v1.php\/cloud\/capabilities?format=json","message":"Login failed: 'iris' (Remote IP: '109.40.130.170')","userAgent":"Mozilla\/5.0 (iOS) Nextcloud-iOS\/2.22.7","version":"15.0.0.10"}
{"reqId":"XC3EKX8AAQEAAGfyelQAAEMS","level":2,"time":"2019-01-03T08:13:29+00:00","remoteAddr":"109.40.130.170","user":"--","app":"core","method":"POST","url":"\/ocs\/v2.php\/apps\/notifications\/api\/v2\/push?format=json&pushTokenHash=4347254b6ede98973c835f8e674f5ddf948b448facdb1e8a14e6becb7cb860c8cf3fc2313366e76b84567f967315b20e95ae039ec3932d62af695c8cd10c8db4&devicePublicKey=-----BEGIN%20PUBLIC%20KEY-----%0AMIIBIjANBgkqhkiG9w0BAQEFAAOCAQ8AMIIBCgKCAQEAoUGqUSWAefS7oLDgVPwG%0ATt1dpRS%2FXKGoS2wt%2FKnPVGgiH2T74A%2BcE4J23tl6re9FaiHE9HARO%2F6VrvLeCl0Q%0AqhZFZUAPJrpYOjG1h70b6oxkC1ILD2r5k1DrGeewgTKI%2BJYq6lCZi7z9uU2KnDgx%0AF5XQbl6XAHpXWcmjPTBCPwlmrC7x30QgmIcHfYortpK8AA6ViKK1sX6U38lCoOOi%0A8sY3xJWawGmJK7ptGKu9ARpReCEy8szIbpvbLpgzD2dqIoQu%2Bx3fGSmgmsT36cO%2F%0AMm4L%2BvFu2Ivt7y7RuqYz4w2SVyTB1ltBaEx7Ml2DpsRBEFA%2B5Lvql81A9UwF1zRW%0AWwIDAQAB%0A-----END%20PUBLIC%20KEY-----%0A&proxyServer=https:\/\/push-notifications.nextcloud.com","message":"Login failed: 'iris [... cut ...]
{"reqId":"XC3EKX8AAQEAAGfyelcAAEMC","level":2,"time":"2019-01-03T08:13:31+00:00","remoteAddr":"109.40.130.170","user":"--","app":"core","method":"PROPFIND","url":"\/remote.php\/webdav","message":"Login failed: 'iris' (Remote IP: '109.40.130.170')","userAgent":"Mozilla\/5.0 (iOS) Nextcloud-iOS\/2.22.7","version":"15.0.0.10"}
{"reqId":"XC3ELH8AAQEAAGfyelsAAFMC","level":2,"time":"2019-01-03T08:13:35+00:00","remoteAddr":"109.40.130.170","user":"--","app":"core","method":"PROPFIND","url":"\/remote.php\/webdav","message":"Login failed: 'iris' (Remote IP: '109.40.130.170')","userAgent":"Mozilla\/5.0 (iOS) Nextcloud-iOS\/2.22.7","version":"15.0.0.10"}

```
</details>

chiyi4488, 2019-01-15:
Do you have auto backup turned on?
This will be the same after I turn it on, but because I have restore, it may be that some part of the system is wrong, and the auto backup will enter maintenance mode.

closed 2019-01-16 by nachoparker

codac, 2019-01-17:
What I was able to figure out is, that the update itself works correctly.
Maintenance Mode goes off and on as requested.

I have auto backup turned on, yes. 
The log files show no information about any activity while the system goes into maintenance mode. 
No Log entries at all in that specific time.

@nachoparker: What information do you need?
Why is this issue closed? Behaviour is still the same.

chiyi4488, 2019-01-17:
Hey friend, it's recommended that you turn off Auto backup first, and after I close, nothing like this has happened.

nachoparker, 2019-01-18:
@codac you could try turning off autobackup, see if it stops happening.

Also you can run a manual backup, see if it goes out of maintenance mode ok

chiyi4488, 2019-01-18:
Actually, I'm pretty sure it's a matter of automatic backup. Manual backups automatically turn off maintenance mode

codac, 2019-02-05:
@nachoparker and @chiyi4488: 
After several weeks I can now confirm, that the root cause is not the auto-update but the auto-backup.
Turning off the auto-update functionality didn't help. After turning off the auto-backup functionality the system stopped getting stuck in maintenance mode.

As I was not able to see any information in the log files, will this be added in future releases?
For me, the auto-backup is quite important. Is there any fix for this planned or at least a work-around available in order to be able to still use an automatic backup?

chiyi4488, 2019-02-07:
@nachoparker
Can you please help test the function of automatic backup? I am pretty sure there are some errors in the middle. So far, I still can't use the function of automatic backup (manual backup can be used)

ovpc, 2019-02-07:
Mine runs every night, receive daily reports by email, without any issues:

```
Cron <root@host> /usr/local/bin/ncp-backup-auto

Maintenance mode enabled
Nextcloud is in maintenance mode - no apps have been loaded

check free space...
clean up old backup /media/snap/ncp-backups/nextcloud-bkp_20190129_1548727501.tar
Maintenance mode already enabled
backup database...
backup files...
backup /media/snap/ncp-backups/nextcloud-bkp_20190207_1549504801.tar generated
Maintenance mode disabled
Maintenance mode already disabled

```

nachoparker, 2019-02-08:
Same here, need more info. Maybe try `sudo /usr/local/bin/ncp-backup-auto` and share the output.

chiyi4488, 2019-02-09:
@ovpc
How can I receive a report via email?

ovpc, 2019-02-09:
I am in habit of installing ssmtp and bsd-mailx, and create a .forward for root to myself.
But if mail is already configured and working on your system; I think, adding "MAILTO"="your@mail.tld" to the cronjob will do the trick also.

codac, 2019-02-10:
I've run `sudo /usr/local/bin/ncp-backup-auto` and again nextcloudpi wont return out of the maintenance mode.
 Here is my root mail.
[/var/mail/root](https://pastebin.com/MTXEdyRx)

I can't find anything at all in the /var/log/syslog file, as if nothing would be logged at all.

The output is:

> Cron <root@host> /usr/local/bin/ncp-backup-auto
> Maintenance mode enabled
> Nextcloud is in maintenance mode - no apps have been loaded

> check free space...
> clean up old backup /media/snap/ncp-backups/nextcloud-bkp_20190129_1548727501.tar
> Maintenance mode already enabled
> backup database...
> backup files...

As the backup takes some time, before I get any more output, the putty connection gets a timeout,

nachoparker, 2019-02-11:
you shouldn't be getting any timeout, maybe that's related to your issue

codac, 2019-04-26:
@nachoparker 
Well, putty itself gets an timeout which is a default setting of putty. As soon as the timeout is there, of course I can't see any progress of the backup anymore.

My girlfriend syncs her photos to my NextCloudPi and that is quite some amount of data. 
Probably the backup process itself gets a timeout after some time due to the 25GB of data that gets backupped plus 7Gb of my own data. 
A rapsberry pi would need quite some time to backup this data because of its low performance.
The backup stops of whatsoever cause and so it never returns back from maintenance mode. 

Could you please further investigate. A backup is quite important and since months I cannot do any full backup of my data. It seems that @chiyi4488 has the same issue.

chiyi4488, 2019-04-27:
> It seems that @chiyi4488 has the same issue.

Yep. And ncp will hang on.
Now I am close automatically backup.