nextcloud/nextcloudpi #763: All passwords are rejected (curl installer)

Charles-IV, 2019-01-04:
I recently installed nextcloud with the nextcloudpi curl installer. It was a bit of a bumpy ride, but after installing all of the missing php modules, the web interface seems all well and good.

However, when I try to log in to the default ncp account, it tells me I have the wrong password. Before trying, I had changed the password in the ncp interface, and discarded the original random one ncp generated for me. I’ve tried resetting the password through the NCP web interface, ncp-config and sudo -u www-data php occ user:resetpassword ncp, and still none of these worked. I then updates to nextcloud 15 and ran nc-init, with still no avail.

I also created a new user with the occ command, and couldn’t log in as this user either. I have attempted to log in through a browser and client application on my computer and phone.

I have also checked that the system time on my Pi is correct through date, as I saw this mentioned online.
Any ideas? Thanks

**System information**

<--! Paste this in GitHub report -->

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v0.67.11
distribution         Raspbian GNU/Linux 9 \n \l
automount            yes
USB devices          sda 
datadir              /var/www/nextcloud/data
data in SD           yes
data filesystem      ext2/ext3
data disk usage      8.7G/13G
rootfs usage         8.7G/13G
swapfile             /var/swap
dbdir                /var/lib/mysql
Nextcloud check      ok
Nextcloud version    15.0.0.10
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            wlan0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               8:16

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "1": "192.168.1.99",
            "5": "nextcloudpi.local",
            "7": "nextcloudpi",
            "8": "nextcloudpi.lan",
            "9": "ncp.charles-iv.uk"
        },
        "datadirectory": "***REMOVED SENSITIVE VALUE***",
        "dbtype": "mysql",
        "version": "15.0.0.10",
        "overwrite.cli.url": "http:\/\/localhost",
[... 90 lines cut ...]
```

{"reqId":"XC@GeH8AAQEAAAew5gYAAAAN","level":3,"time":"2019-01-04T16:14:48+00:00","remoteAddr":"127.0.0.1","user":"--","app":"PHP","method":"GET","url":"\/index.php\/login?user=ncp","message":"You are using a fallback implementation of the intl extension. Installing the native one is highly recommended instead. at \/var\/www\/nextcloud\/3rdparty\/patchwork\/utf8\/src\/Patchwork\/Utf8\/Bootup\/intl.php#18","userAgent":"Mozilla\/5.0 (X11; Linux x86_64; rv:60.0) Gecko\/20100101 Firefox\/60.0","version":"15.0.0.10"}
{"reqId":"XC@Gp38AAQEAAANT@dAAAAAH","level":3,"time":"2019-01-04T16:15:37+00:00","remoteAddr":"127.0.0.1","user":"--","app":"PHP","method":"GET","url":"\/index.php\/login?user=ncp","message":"You are using a fallback implementation of the intl extension. Installing the native one is highly recommended instead. at \/var\/www\/nextcloud\/3rdparty\/patchwork\/utf8\/src\/Patchwork\/Utf8\/Bootup\/intl.php#18","userAgent":"Mozilla\/5.0 (X11; Linux x86_64; rv:60.0) Gecko\/20100101 Firefox\/60.0","version":"15.0.0.10"}
{"reqId":"XC@Gqn8AAQEAAANT@dUAAAAH","level":3,"time":"2019-01-04T16:15:38+00:00","remoteAddr":"127.0.0.1","user":"--","app":"PHP","method":"GET","url":"\/index.php\/apps\/theming\/styles?v=0","message":"You are using a fallback implementation of the intl extension. Installing the native one is highly recommended instead. at \/var\/www\/nextcloud\/3rdparty\/patchwork\/utf8\/src\/Patchwork\/Utf8\/Bootup\/intl.php#18","userAgent":"Mozilla\/5.0 (X11; Linux x86_64; rv:60.0) Gecko\/20100101 Firefox\/60.0","version":"15.0.0.10"}
{"reqId":"XC@Gqn8AAQEAAAerlEgAAAAM","level":3,"time":"2019-01-04T16:15:39+00:00","remoteAddr":"127.0.0.1","user":"--","app":"PHP","method":"GET","url":"\/index.php\/apps\/theming\/js\/theming?v=0","message":"You are using a fallback implementation of the intl extension. Installing the native one is highly recommended instead. at \/var\/www\/nextcloud\/3rdparty\/patchwork\/utf8\/src\/Patchwork\/Utf8\/Bootup\/intl.php#18","userAgent":"Mozilla\/5.0 (X11; Linux x86_64; rv:60.0) Gecko\/20100101 Firefox\/60.0","version":"15.0.0.10"}
{"reqId":"XC@Gqn8AAQEAAANJLqgAAAAF","level":3,"time":"2019-01-04T16:15:39+00:00","remoteAddr":"127.0.0.1","user":"--","app":"PHP","method":"GET","url":"\/index.php\/apps\/theming\/image\/logo?v=0&useSvg=1","message":"You are using a fallback implementation of the intl extension. Installing the native one is highly recommended instead. at \/var\/www\/nextcloud\/3rdparty\/patchwork\/utf8\/src\/Patchwork\/Utf8\/Bootup\/intl.php#18","userAgent":"Mozilla\/5.0 (X11; Linux x86_64; rv:60.0) Gecko\/20100101 Firefox\/60.0","version":"15.0.0.10"}
{"reqId":"XC@Gqn8AAQEAAAexEIAAAAAO","level":3,"time":"2019-01-04T16:15:39+00:00","remoteAddr":"127.0.0.1","user":"--","app":"PHP","method":"GET","url":"\/index.php\/apps\/theming\/image\/logo?v=0","message":"You are using a fallback implementation of the intl extension. Installing the native one is highly recommended instead. at \/var\/www\/nextcloud\/3rdparty\/patchwork\/utf8\/src\/Patchwork\/Utf8\/Bootup\/intl.php#18","userAgent":"Mozilla\/5.0 (X11; Linux x86_64; rv:60.0) Gecko\/20100101 Firefox\/60.0","version":"15.0.0.10"}
{"reqId":"XC@Gqn8AAQEAAAM-RF4AAAAD","level":3,"time":"2019-01-04T16:15:39+00:00","remoteAddr":"127.0.0.1","user":"--","app":"PHP","method":"GET","url":"\/index.php\/apps\/accessibility\/js\/accessibility?v=0","message":"You are using a fallback implementation of the intl extension. Installing the native one is highly recommended instead. at \/var\/www\/nextcloud\/3rdparty\/patchwork\/utf8\/src\/Patchwork\/Utf8\/Bootup\/intl.php#18","userAgent":"Mozilla\/5.0 (X11; Linux x86_64; rv:60.0) Gecko\/20100101 Firefox\/60.0","version":"15.0.0.10"}
{"reqId":"XC@Gqn8AAQEAAANT@dcAAAAH","level":3,"time":"2019-01-04T16:15:39+00:00","remoteAddr":"127.0.0.1","user":"--","app":"PHP","method":"GET","url":"\/index.php\/apps\/theming\/image\/background?v=0","message":"You are using a fallback implementation of the intl extension. Installing the native one is highly recommended instead. at \/var\/www\/nextcloud\/3rdparty\/patchwork\/utf8\/src\/Patchwork\/Utf8\/Bootup\/intl.php#18","userAgent":"Mozilla\/5.0 (X11; Linux x86_64; rv:60.0) Gecko\/20100101 Firefox\/60.0","version":"15.0.0.10"}
{"reqId":"XC@Gtn8AAQEAAAew5ggAAAAN","level":2,"time":"2019-01-04T16:15:52+00:00","remoteAddr":"127.0.0.1","user":"--","app":"core","method":"POST","url":"\/index.php\/login?user=ncp","message":"Login failed: 'ncp' (Remote IP: '127.0.0.1')","userAgent":"Mozilla\/5.0 (X11; Linux x86_64; rv:60.0) Gecko\/20100101 Firefox\/60.0","version":"15.0.0.10"}
{"reqId":"XC@Gtn8AAQEAAAew5ggAAAAN","level":3,"time":"2019-01-04T16:15:52+00:00","remoteAddr":"127.0.0.1","user":"--","app":"PHP","method":"POST","url":"\/index.php\/login?user=ncp","message":"You are using a fallback implementation of the intl extension. Installing the native one is highly recommended instead. at \/var\/www\/nextcloud\/3rdparty\/patchwork\/utf8\/src\/Patchwork\/Utf8\/Bootup\/intl.php#18","userAgent":"Mozilla\/5.0 (X11; Linux x86_64; rv:60.0) Gecko\/20100101 Firefox\/60.0","version":"15.0.0.10"}
{"reqId":"XC@GuX8AAQEAAAew5gkAAAAN","level":3,"time":"2019-01-04T16:15:53+00:00","remoteAddr":"127.0.0.1","user":"--","app":"PHP","method":"GET","url":"\/index.php\/login?user=ncp","message":"You are using a fallback implementation of the intl extension. Installing the native one is highly recommended instead. at \/var\/www\/nextcloud\/3rdparty\/patchwork\/utf8\/src\/Patchwork\/Utf8\/Bootup\/intl.php#18","userAgent":"Mozilla\/5.0 (X11; Linux x86_64; rv:60.0) Gecko\/20100101 Firefox\/60.0","version":"15.0.0.10"}
{"reqId":"XC@Gwn8AAQEAAAM2ioAAAAAC","level":3,"time":"2019-01-04T16:16:02+00:00","remoteAddr":"127.0.0.1","user":"--","app":"PHP","method":"GET","url":"\/index.php\/login\/flow","message":"You are using a fallback implementation of the intl extension. Installing the native one is highly recommended in [... cut ...]
{"reqId":"XC@Gyn8AAQEAAAew5gsAAAAN","level":3,"time":"2019-01-04T16:16:11+00:00","remoteAddr":"127.0.0.1","user":"--","app":"PHP","method":"GET","url":"\/index.php\/js\/core\/merged-login.js?v=7d5229ff-0","message":"You are using a fallback implementation of the intl extension. Installing the native one is highly recommended instead. at \/var\/www\/nextcloud\/3rdparty\/patchwork\/utf8\/src\/Patchwork\/Utf8\/Bootup\/intl.php#18","userAgent":"Mozilla\/5.0 (Linux) mirall\/2.5.1git (Nextcloud)","version":"15.0.0.10"}

```
</details>

nachoparker, 2019-01-15:
See the wiki documentation to see how to restore the password. If still in doubt, please use the forums, this is only for bug tracking.

Thanks

closed 2019-01-15 by nachoparker

Charles-IV, 2019-01-17:
Apologies if i forgot to mention it in the first post, but i have reset the password several times through the NextCloudPi interface, and the issue persists. [I did ask on the forums](https://help.nextcloud.com/t/nextcloud-not-accepting-correct-passwords/44445), and they said this was a bug with the curl installer.
Unfortunately I need to use my raspberry pi for several other things, and setting it up again with a nextcloudpi image would be unfeasible.

nachoparker, 2019-01-18:
Assuming that this is caused by the database bug that was already fixed I would recommend you to just try again over a clean Debian instance.

We know for a  fact that current version works.

Charles-IV, 2019-01-18:
thanks, I will try on a vm or spare sd card and let you know.