nextcloud/nextcloudpi #764: nc-scan-auto add 26 lines in /var/mail/root every minute

ibalabozov, 2019-01-06:
**Description**  
Hi,
At first the nextcloudpi is awesome!

I activated the samba through web interface and also [nc-scan-auto] with auto update interval of 1 minute. After that I found that in /var/mail/root one email is added every minute (26 lines of text). So the /var/mail/root file grow up and up. It is good idea the [nc-scan-auto] to be improved with adding a code that disable posting of mails in the /var/mail/root file or even better adding a option to disable mail in the [nc-scan-auto] section in the web interface.

**Example**  
Solution that work for me is to add > /dev/null to the ncp-scan-auto cron job. I make it in this way:
 - Log in with root account to the terminal and do this:
       nano /etc/cron.d/ncp-scan-auto
 - Change this line:
       */1  *  *  *  *  root /usr/local/bin/ncp-scan
   to look like this:
       */1  *  *  *  *  root /usr/local/bin/ncp-scan > /dev/null

ovpc, 2019-01-06:
Hi @ibalabozov and thanks for input.

Stopping mail, can also be achieved by adding:

> MAILTO=""

to the top of the cronjob

An option for users to enable/disable mail and set interval does make sense tho.

As would the option to set an array of directories to scan. Most of the time only a couple of folders are changed/created externally, so no need to scan all the other ones.....

I modified my /usr/local/bin/ncp-scan to include lines looking like this:
> sudo -u www-data php occ files:scan --path="/ncp/files/somedir"
sleep 1
sudo -u www-data php occ files:scan --path="/ncp/files/someotherdir/srv/folder"

And I just run the cronjob once a day at a fixed time. 
I would certainly limit scan by using --path="/path/to/folder" option if I needed it to run every minute!

I suspect, that unless you have time&skills to create the required code and do a PR to devel, these options may be for the growing ncp wishlist, we welcome anyone wants to contribute. Thank you.

nachoparker, 2019-01-06:
Definitely been wanting to add the option to do a single path instead of everything. Good suggestions

nachoparker, 2019-05-26:
This was fixed in ccb6fc12e5355cdea79d8ed40121a74cfb355cd4

Thanks!

closed 2019-05-26 by nachoparker