nextcloud/nextcloudpi #768: Can't update NCP since 0.67.11

zedocrob, 2019-01-09:
i can't update NCP since 0.67.11:

Tried to update to v1.0.1 through 'ncp-config' or type 'sudo ncp-update' but i always have:
```
root@nextcloudpi:~# sudo ncp-update
Downloading updates
Performing updates
Running nc-autoupdate-nc
/usr/local/etc/library.sh: line 117: echo: write error: No space left on device
root@nextcloudpi:~#
```
same with NextcloudPi Panel (green led but no output in the white window)
same after rebooting Odroid HC2

Tried to free some space with no luck:
```
apt-get autoremove -y
apt-get clean
rm /var/lib/apt/lists/* -r
```

**System information**

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v0.67.11
NextCloudPi image    NextCloudPi_09-29-18
distribution         Debian GNU/Linux 9 \n \l
automount            no
USB devices          sda 
datadir              /media/DATAcloud/ncdata
data in SD           no
data filesystem      btrfs
data disk usage      391G/3.7T
rootfs usage         3.9G/30G
swapfile             /var/swap
dbdir                /var/lib/mysql
Nextcloud check      ok
Nextcloud version    14.0.4.2
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            enx001e0632be42
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               13min

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
[... 71 lines cut ...]
```

{"reqId":"XCjuDX8AAQEAAGU78LEAAAAF","level":3,"time":"2018-12-30T16:10:55+00:00","remoteAddr":"14.98.162.41","user":"--","app":"PHP","method":"GET","url":"\/phpmyadmin","message":"unlink(\/media\/DATAcloud\/ncdata\/appdata_oc65v9pdw1g0\/css\/files_texteditor\/783c-97ca-merged.css.deps): No such file or directory at \/var\/www\/nextcloud\/lib\/private\/Files\/Storage\/Local.php#227","userAgent":"python-requests\/2.21.0","version":"14.0.4.2"}
{"reqId":"XCjuDn8AAQEAADGWDYIAAAAG","level":3,"time":"2018-12-30T16:10:55+00:00","remoteAddr":"14.98.162.41","user":"--","app":"PHP","method":"GET","url":"\/console","message":"unlink(\/media\/DATAcloud\/ncdata\/appdata_oc65v9pdw1g0\/css\/files_texteditor\/783c-97ca-merged.css.gzip): No such file or directory at \/var\/www\/nextcloud\/lib\/private\/Files\/Storage\/Local.php#227","userAgent":"python-requests\/2.21.0","version":"14.0.4.2"}
{"reqId":"XCjuDn8AAQEAADGWDYIAAAAG","level":3,"time":"2018-12-30T16:10:55+00:00","remoteAddr":"14.98.162.41","user":"--","app":"PHP","method":"GET","url":"\/console","message":"unlink(\/media\/DATAcloud\/ncdata\/appdata_oc65v9pdw1g0\/css\/files_trashbin\/1e6d-97ca-trash.css): No such file or directory at \/var\/www\/nextcloud\/lib\/private\/Files\/Storage\/Local.php#227","userAgent":"python-requests\/2.21.0","version":"14.0.4.2"}
{"reqId":"XCjuDn8AAQEAADGWDYIAAAAG","level":3,"time":"2018-12-30T16:10:55+00:00","remoteAddr":"14.98.162.41","user":"--","app":"PHP","method":"GET","url":"\/console","message":"unlink(\/media\/DATAcloud\/ncdata\/appdata_oc65v9pdw1g0\/css\/firstrunwizard\/70e2-97ca-firstrunwizard.css.gzip): No such file or directory at \/var\/www\/nextcloud\/lib\/private\/Files\/Storage\/Local.php#227","userAgent":"python-requests\/2.21.0","version":"14.0.4.2"}
{"reqId":"XCjuDn8AAQEAADGWDYIAAAAG","level":3,"time":"2018-12-30T16:10:55+00:00","remoteAddr":"14.98.162.41","user":"--","app":"PHP","method":"GET","url":"\/console","message":"unlink(\/media\/DATAcloud\/ncdata\/appdata_oc65v9pdw1g0\/css\/icons\/icons-vars.css): No such file or directory at \/var\/www\/nextcloud\/lib\/private\/Files\/Storage\/Local.php#227","userAgent":"python-requests\/2.21.0","version":"14.0.4.2"}
{"reqId":"XCjuDn8AAQEAADGWDYIAAAAG","level":3,"time":"2018-12-30T16:10:55+00:00","remoteAddr":"14.98.162.41","user":"--","app":"PHP","method":"GET","url":"\/console","message":"unlink(\/media\/DATAcloud\/ncdata\/appdata_oc65v9pdw1g0\/css\/notifications\/2c83-97ca-styles.css.gzip): No such file or directory at \/var\/www\/nextcloud\/lib\/private\/Files\/Storage\/Local.php#227","userAgent":"python-requests\/2.21.0","version":"14.0.4.2"}
{"reqId":"XCjuDn8AAQEAADGWDYIAAAAG","level":3,"time":"2018-12-30T16:10:55+00:00","remoteAddr":"14.98.162.41","user":"--","app":"PHP","method":"GET","url":"\/console","message":"unlink(\/media\/DATAcloud\/ncdata\/appdata_oc65v9pdw1g0\/css\/phonetrack\/9eac-97ca-Leaflet.LinearMeasurement.css): No such file or directory at \/var\/www\/nextcloud\/lib\/private\/Files\/Storage\/Local.php#227","userAgent":"python-requests\/2.21.0","version":"14.0.4.2"}
{"reqId":"XCzfQ38AAQEAAAqxLJoAAAAA","level":3,"time":"2019-01-02T15:56:51+00:00","remoteAddr":"91.166.30.12","user":"ncp","app":"PHP","method":"POST","url":"\/index.php\/apps\/files_trashbin\/ajax\/delete.php","message":"RecursiveDirectoryIterator::__construct(\/media\/DATAcloud\/ncdata\/ncp\/files_trashbin\/files\/test.d1546444582\/FLASHABLE): failed to open dir: Permission denied at \/var\/www\/nextcloud\/lib\/private\/Files\/Storage\/Local.php#109","userAgent":"Mozilla\/5.0 (Macintosh; Intel Mac OS X 10_11_6) AppleWebKit\/537.36 (KHTML, like Gecko) Chrome\/71.0.3578.98 Safari\/537.36","version":"14.0.4.2"}
{"reqId":"XCzndn8AAQEAACTFQQUAAAAB","level":3,"time":"2019-01-02T16:31:51+00:00","remoteAddr":"91.166.30.12","user":"ncp","app":"PHP","method":"POST","url":"\/index.php\/apps\/files_trashbin\/ajax\/delete.php","message":"RecursiveDirectoryIterator::__construct(\/media\/DATAcloud\/ncdata\/ncp\/files_trashbin\/files\/test.d1546444582\/FLASHABLE): failed to open dir: Permission denied at \/var\/www\/nextcloud\/lib\/private\/Files\/Storage\/Local.php#109","userAgent":"Mozilla\/5.0 (Macintosh; Intel Mac OS X 10_11_6) AppleWebKit\/537.36 (KHTML, like Gecko) Chrome\/71.0.3578.98 Safari\/537.36","version":"14.0.4.2"}
{"reqId":"XCznh38AAQEAAAqT18YAAAAG","level":3,"time":"2019-01-02T16:32:07+00:00","remoteAddr":"91.166.30.12","user":"ncp","app":"PHP","method":"POST","url":"\/index.php\/apps\/files_trashbin\/ajax\/delete.php","message":"RecursiveDirectoryIterator::__construct(\/media\/DATAcloud\/ncdata\/ncp\/files_trashbin\/files\/test.d1546444582\/FLASHABLE): failed to open dir: Permission denied at \/var\/www\/nextcloud\/lib\/private\/Files\/Storage\/Local.php#109","userAgent":"Mozilla\/5.0 (Macintosh; Intel Mac OS X 10_11_6) AppleWebKit\/537.36 (KHTML, like Gecko) Chrome\/71.0.3578.98 Safari\/537.36","version":"14.0.4.2"}
{"reqId":"XCzn2X8AAQEAAAQPJKUAAAAD","level":3,"time":"2019-01-02T16:33:30+00:00","remoteAddr":"91.166.30.12","user":"ncp","app":"index","method":"POST","url":"\/index.php\/apps\/files_trashbin\/ajax\/delete.php","message":{"Exception":"UnexpectedValueException","Message":"RecursiveDirectoryIterator::__construct(\/media\/DATAcloud\/ncdata\/ncp\/files_trashbin\/files\/test.d1546446718\/APK custom): failed to open dir: Permission denied","Code":0,"Trace":[{"function":"__construct","class":"RecursiveDirectoryIterator","type":"->","args":["\/media\/DATAcloud\/ncdata\/ncp\/files_trashbin\/files\/test.d1546446718\/APK custom",0]},{"function":"getChildren","class":"RecursiveDirectoryIterator","type":"->","args":[]},{"file":"\/var\/www\/nextcloud\/apps\/files_trashbin\/lib\/Trashbin.php","line":950,"function":"next","class":"RecursiveIteratorIterator","type":"->","args":[]},{"file":"\/var\/www\/nextcloud\/apps\/files_trashbin\/lib\/Trashbin.php","line":584,"function":"calculateSize","class":"OCA\\Files_Trashbin\\Trashbin","type":"::","args":[{"__class__":"OC\\Files\\View"}]},{"file":"\/var\/www\/nextcloud\/apps\/files_trashbin\/ajax\/delete.php","line":69,"function":"delete","class":"OCA\\Files_Trashbin\\Trashbin","type":"::","args":["test","ncp","1546446718"]},{"file":"\/var\/www\/nextcloud\/lib\/private\/Route\/Route.php","line":155,"args":["\/var\/www\/nextcloud\/apps\/files_trashbin\/ajax\/delete.php"],"function":"require_once"},{"function":"OC\\Route\\{closure}","class":"OC\\Route\\Route","type":"->","args":["*** sensitive parameters replaced ***"]},{"file":"\/var\/www\/nextcloud\/lib\/private\/Route\/Router.php","line":297,"function":"call_user_func","args":[{"__class__":"Closure"},{"_route":"files_trashbin_ajax_delete"}]},{"file":"\/var\/www\/nextcloud\/lib\/base.p [... cut ...] .0 (Macintosh) mirall\/2.3.3 (build 84) (Nextcloud)","version":"14.0.4.2"}
{"reqId":"XDZxxH8AAQEAAAZI3sIAAAAA","level":3,"time":"2019-01-09T22:12:20+00:00","remoteAddr":"172.91.117.94","user":"--","app":"PHP","method":"PROPFIND","url":"\/remote.php\/webdav\/","message":"Error: Class 'OCA\\DAV\\Connector\\Sabre\\ExceptionLoggerPlugin' not found at \/var\/www\/nextcloud\/remote.php#57","userAgent":"Mozilla\/5.0 (Macintosh) mirall\/2.3.3 (build 84) (Nextcloud)","version":"14.0.4.2"}

```
</details>

nachoparker, 2019-01-10:
You are out of space, please use the forums for support

https://help.nextcloud.com/c/support/appliances-docker-snappy-vm

closed 2019-01-10 by nachoparker

Best-Internet-Tips, 2019-03-25:
APK Custom is one of the cool Android injection apps that is available to download and install on your device for free. This app is better than HTTP injector, kpn tunnel, epoxy because of it’s easy to use layout and responsive design. If you want to know more news about Apk Custom then click the link below
[Download Apk custom](https://ibixion.com/apk-custom.html)