nextcloud/nextcloudpi #769: nc-format-usb conflicts with nc-automount with parititon types

Zackptg5, 2019-01-10:
nc-format-usb formats as btrfs like it always has but nc-automount states that the partition must be ext4.
The status page seems to be totally bugged to as it says that automount is disabled and doesn't show my data directory as being on the usb hard drive like I set with nc-data.
It also gives tops at the top for things that I've already done.

uptime               15min

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "1": "192.168.1.3",
            "5": "nextcloudpi.local",
            "7": "nextcloudpi",
            "8": "nextcloudpi.lan"
        },
        "datadirectory": "***REMOVED SENSITIVE VALUE***",
        "dbtype": "mysql",
        "version": "14.0.4.2",
        "overwrite.cli.url": "http:\/\/localhost",
        "dbname": "***REMOVED SENSITIVE VALUE***",
        "dbhost": "***REMOVED SENSITIVE VALUE***",
        "dbport": "",
        "dbtableprefix": "oc_",
        "mysql.utf8mb4": true,
        "dbuser": "***REMOVED SENSITIVE VALUE***",
        "dbpassword": "***REMOVED SENSITIVE VALUE***",
        "installed": true,
        "instanceid": "***REMOVED SENSITIVE VALUE***",
        "memcache.local": "\\OC\\Memcache\\Redis",
        "memcache.locking": "\\OC\\Memcache\\Redis",
        "redis": {
            "host": "***REMOVED SENSITIVE VALUE***",
            "port": 0,
            "timeout": 0,
            "password": "***REMOVED SENSITIVE VALUE***"
        },
        "tempdirectory": "\/var\/www\/nextcloud\/data\/tmp",
        "mail_smtpmode": "sendmail",
        "mail_smtpauthtype": "LOGIN",
        "mail_from_address": "***REMOVED SENSITIVE VALUE***",
        "mail_domain": "***REMOVED SENSITIVE VALUE***",
        "overwriteprotocol": "https",
        "maintenance": false,
        "logfile": "\/var\/www\/nextcloud\/data\/nextcloud.log"
    }
}

```
</details>

<details>
<summary>HTTPd logs</summary>

```

[Thu Jan 10 22:52:13.859055 2019] [proxy_fcgi:error] [pid 3233:tid 2758140976] [client 192.168.1.189:64772] AH01071: Got error 'PHP message: PHP Warning:  ini_set(): A session is active. You cannot change the session module's ini settings at this time in /var/www/ncp-web/index.php on line 41\nPHP message: PHP Warning:  ini_set(): A session is active. You cannot change the session module's ini settings at this time in /var/www/ncp-web/index.php on line 43\nPHP message: PHP Warning:  fopen(ncp-web.cfg): failed to open stream: No such file or directory in /var/www/ncp-web/L10N.php on line 145\nPHP message: PHP Warning:  fopen(ncp-web.cfg): failed to open stream: No such file or directory in /var/www/ncp-web/L10N.php on line 145\nPHP message: NACHO auto\n - auto\nPHP message: NACHO en\n - auto\nPHP message: NACHO de\n - auto\nPHP message: NACHO es\n - auto\nPHP message: NACHO zh\n - auto\n', referer: https://192.168.1.3:4443/wizard/
[Thu Jan 10 22:52:14.733611 2019] [proxy_fcgi:error] [pid 3233:tid 2774934576] [client 192.168.1.189:64773] AH01071: Got error 'PHP message: PHP Warning:  touch(): Utime failed: Permission denied in /var/www/ncp-web/ncp-output.php on line 81\n'
[Thu Jan 10 22:52:17.516509 2019] [proxy_fcgi:error] [pid 3233:tid 2791728176] [client 192.168.1.189:64773] AH01071: Got error 'PHP message: PHP Warning:  fopen(ncp-web.cfg): failed to open stream: No such file or directory in /var/www/ncp-web/L10N.php on line 145\n'
[Thu Jan 10 22:52:19.405348 2019] [proxy_fcgi:error] [pid 3233:tid 2791728176] [client 192.168.1.189:64773] AH01071: Got error 'PHP message: PHP Warning:  fopen(ncp-web.cfg): failed to open stream: No such file or directory in /var/www/ncp-web/L10N.php on line 145\n'
[Thu Jan 10 22:52:33.325496 2019] [proxy_fcgi:error] [pid 3233:tid 2791728176] [client 192.168.1.189:64773] AH01071: Got error 'PHP message: PHP Warning:  fopen(ncp-web.cfg): failed to open stream: No such file or directory in /var/www/ncp-web/L10N.php on line 145\n'
[... 30 lines cut ...]
2019-01-10 23:08:09 3063418880 [Note] InnoDB: Compressed tables use zlib 1.2.8
2019-01-10 23:08:09 3063418880 [Note] InnoDB: Using Linux native AIO
2019-01-10 23:08:09 3063418880 [Note] InnoDB: Using generic crc32 instructions
2019-01-10 23:08:09 3063418880 [Note] InnoDB: Initializing buffer pool, size = 128.0M
2019-01-10 23:08:09 3063418880 [Note] InnoDB: Completed initialization of buffer pool
2019-01-10 23:08:09 3063418880 [Note] InnoDB: Highest supported file format is Barracuda.
2019-01-10 23:08:09 3063418880 [Note] InnoDB: 128 rollback segment(s) are active.
2019-01-10 23:08:09 3063418880 [Note] InnoDB: Waiting for purge to start
2019-01-10 23:08:09 3063418880 [Note] InnoDB:  Percona XtraDB (http://www.percona.com) 5.6.41-84.1 started; log sequence number 5851853
2019-01-10 23:08:10 2483024704 [Note] InnoDB: Dumping buffer pool(s) not yet started
2019-01-10 23:08:10 3063418880 [Note] Plugin 'FEEDBACK' is disabled.
2019-01-10 23:08:10 3063418880 [Note] Server socket created on IP: '127.0.0.1'.
2019-01-10 23:08:10 3063418880 [Note] /usr/sbin/mysqld: ready for connections.
Version: '10.1.37-MariaDB-0+deb9u1'  socket: '/var/run/mysqld/mysqld.sock'  port: 3306  Debian 9.6

```
</details>

<details>
<summary>Nextcloud logs</summary>

```

```
</details>

Zackptg5, 2019-01-11:
Just an update on this, it appears that it's cosmetic as everything is now working fine

nachoparker, 2019-01-11:
Hi,

Thanks for your feedback

> nc-automount states that the partition must be ext4.

Where do you see this?

Also you have removed all relevant info from your output, so I have no clue if you are on a Raspberry Pi, or x86 or what?

What tips are wrong? do you have nc-automount active?

Please, be more specific so we can fix things if there is anything to fix.

Zackptg5, 2019-01-11:
@nachoparker 
I'm using odroid HC1 - latest img (12-1-18), then ran nc updates to get latest ncpi and nc (nc was already on latest one - 14.0.4-2)

I see the nc-automount message on nc-automount option in ncpi interface:
[image]
I just copy/pasted it. Let me take another log here
`<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v1.0.2
NextCloudPi image    NextCloudPi_12-01-18
distribution         Debian GNU/Linux 9 \n \l
automount            no
USB devices          sda sdb
datadir              /media/myCloudDrive/ncdata
data in SD           no
data filesystem      btrfs
data disk usage      186G/932G
rootfs usage         3.7G/15G
swapfile             /var/swap
dbdir                /media/myCloudDrive/ncdatabase
Nextcloud check      ok
Nextcloud version    14.0.4.2
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            enx001e0630c6f1
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               17:32

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "1": "192.168.1.3",
            "5": "nextcloudpi.local",
            "7": "nextcloudpi",
            "8": "nextcloudpi.lan",
            "2": "nextcloud.zackptg5.com"
        },
        "datadirectory": "***REMOVED SENSITIVE VALUE***",
        "dbtype": "mysql",
        "version": "14.0.4.2",
        "overwrite.cli.url": "https:\/\/nextcloud.zackptg5.com\/",
        "dbname": "***REMOVED SENSITIVE VALUE***",
        "dbhost": "***REMOVED SENSITIVE VALUE***",
        "dbport": "",
        "dbtableprefix": "oc_",
        "mysql.utf8mb4": true,
        "dbuser": "***REMOVED SENSITIVE VALUE***",
[... 92 lines cut ...]
{"reqId":"XDgMPX8AAQEAABhu-aEAAJYG","level":3,"time":"2019-01-11T03:23:41+00:00","remoteAddr":"192.168.1.1","user":"ncp","app":"PHP","method":"POST","url":"\/apps\/theming\/ajax\/updateStylesheet","message":"unlink(\/media\/myCloudDrive\/ncdata\/appdata_ocottpxem0pn\/css\/comments\/1bf6-5998-autocomplete.css): No such file or directory at \/var\/www\/nextcloud\/lib\/private\/Files\/Storage\/Local.php#227","userAgent":"Mozilla\/5.0 (Windows NT 10.0; Win64; x64; rv:64.0) Gecko\/20100101 Firefox\/64.0","version":"14.0.4.2"}
{"reqId":"XDgMPX8AAQEAABhu-aAAAJYF","level":3,"time":"2019-01-11T03:24:54+00:00","remoteAddr":"192.168.1.1","user":"ncp","app":"PHP","method":"POST","url":"\/apps\/theming\/ajax\/updateStylesheet","message":"unlink(\/media\/myCloudDrive\/ncdata\/appdata_ocottpxem0pn\/css\/files_texteditor\/783c-5998-merged.css.deps): No such file or directory at \/var\/www\/nextcloud\/lib\/private\/Files\/Storage\/Local.php#227","userAgent":"Mozilla\/5.0 (Windows NT 10.0; Win64; x64; rv:64.0) Gecko\/20100101 Firefox\/64.0","version":"14.0.4.2"}
{"reqId":"XDgMg38AAQEAABhu-a8AAJgL","level":3,"time":"2019-01-11T03:25:06+00:00","remoteAddr":"192.168.1.1","user":"--","app":"PHP","method":"GET","url":"\/apps\/theming\/styles?v=4","message":"unlink(\/media\/myCloudDrive\/ncdata\/appdata_ocottpxem0pn\/css\/firstrunwizard\/70e2-5998-colorbox.css.gzip): No such file or directory at \/var\/www\/nextcloud\/lib\/private\/Files\/Storage\/Local.php#227","userAgent":"Mozilla\/5.0 (Windows NT 10.0; Win64; x64; rv:64.0) Gecko\/20100101 Firefox\/64.0","version":"14.0.4.2"}
{"reqId":"XDgMS38AAQEAABhu-aMAAJYB","level":3,"time":"2019-01-11T03:25:06+00:00","remoteAddr":"192.168.1.1","user":"ncp","app":"PHP","method":"POST","url":"\/apps\/theming\/ajax\/updateStylesheet","message":"unlink(\/media\/myCloudDrive\/ncdata\/appdata_ocottpxem0pn\/css\/firstrunwizard\/70e2-5998-colorbox.css.gzip): No such file or directory at \/var\/www\/nextcloud\/lib\/private\/Files\/Storage\/Local.php#227","userAgent":"Mozilla\/5.0 (Windows NT 10.0; Win64; x64; rv:64.0) Gecko\/20100101 Firefox\/64.0","version":"14.0.4.2"}
{"reqId":"XDgMg38AAQEAABhu-a8AAJgL","level":3,"time":"2019-01-11T03:25:09+00:00","remoteAddr":"192.168.1.1","user":"--","app":"PHP","method":"GET","url":"\/apps\/theming\/styles?v=4","message":"unlink(\/media\/myCloudDrive\/ncdata\/appdata_ocottpxem0pn\/css\/firstrunwizard\/70e2-73d5-firstrunwizard.css): No such file or directory at \/var\/www\/nextcloud\/lib\/private\/Files\/Storage\/Local.php#227","userAgent":"Mozilla\/5.0 (Windows NT 10.0; Win64; x64; rv:64.0) Gecko\/20100101 Firefox\/64.0","version":"14.0.4.2"}
{"reqId":"XDgMS38AAQEAABhu-aMAAJYB","level":3,"time":"2019-01-11T03:25:09+00:00","remoteAddr":"192.168.1.1","user":"ncp","app":"PHP","method":"POST","url":"\/apps\/theming\/ajax\/updateStylesheet","message":"unlink(\/media\/myCloudDrive\/ncdata\/appdata_ocottpxem0pn\/css\/firstrunwizard\/70e2-73d5-firstrunwizard.css): No such file or directory at \/var\/www\/nextcloud\/lib\/private\/Files\/Storage\/Local.php#227","userAgent":"Mozilla\/5.0 (Windows NT 10.0; Win64; x64; rv:64.0) Gecko\/20100101 Firefox\/64.0","version":"14.0.4.2"}
[... cut ...] ndows NT 10.0; Win64; x64) AppleWebKit\/537.36 (KHTML, like Gecko) Chrome\/71.0.3578.98 Safari\/537.36","version":"14.0.4.2"}

```
</details>
`

I have nc-automount active, and successfully moved the nc-data and databases to the first usb harddrive which is why I think it's cosmetic. The incorrect info is shown on the status page of ncpi:
[image]
It shows automount as disabled when it's enabled and shows NAT loopback as no when it's actually on (I can access it remotely and I couldn't otherwise)

Haraade, 2019-01-11:
Have the same problem on my Odroid Xu4

zedocrob, 2019-01-12:
Same here on Odroid HC2 and NCP 1.0.1
I have nc-automount active and i see "_you should enable automount to use your plugged in USB drive_"
i've never activated NAT Loopback, i can access it from outside.

Zackptg5, 2019-01-14:
Just confirming that the issues are still present after nc and ncp updated to 15.0.2 and 1.3.0

nachoparker, 2019-01-15:
fixed automount description and the entry in the dashboard.

NAT loopback is not required to access from outside, but from inside your house wifi

Thanks for reporting! missed those bugs during the big rewrite

Cheers

closed 2019-01-15 by nachoparker

Zackptg5, 2019-01-15:
Great, thanks!