nextcloud/nextcloudpi #770: Access Web-UI via ipv6

lenzls, 2019-01-12:
Hello,

it seems to me that the nextcloudpi web-ui does not work with ipv6.

After some tinkering around I noticed that I was greeted by a "no permission" error when accessing "https://nextcloudpi:4443"

After a quick check I noticed that I was trying to connect via ipv6 but the configuration has this not on the whitelist.

When I add my ipv6 ip to 
`/etc/apache2/sites-enabled/ncp.conf`
```
[..]
<RequireAny>
      Require host localhost
      Require local
      Require ip 192.168
      Require ip 172
      Require ip 10

      # added by me
      Require ip fd00::4077:0027:d823:a11b/64
</RequireAny>
[..]
```

I can access the web-ui without any problems.

I don't know what could be done about that but it seems to me like an issue not only I would experience.

regards

nachoparker, 2019-01-15:
Hello,

Good catch, could you try doing `Require ip fd00::/8` instead of your rule, and see if it works for you?

Thanks!

lenzls, 2019-01-15:
Thanks for the quick response.

Your suggested rule seems to work as well :)

nachoparker, 2019-01-15:
awesome, thanks

>Will the 1.3.7 update then just override my ncp.conf?

yes

closed 2019-01-15 by nachoparker

mrkskwsnck, 2019-02-02:
I ran into exactly that same issue with a fresh install from the Raspberry Pi image [NextCloudPi_RPi_01-13-19](https://ownyourbits.com/downloads/NextCloudPi_RPi_01-13-19/). The NCP version was 1.4.x something. Luckily, I found this bug report and so I was able to apply the fix manually. Thanks a lot!

nachoparker, 2019-02-05:
@mrkskwsnck if you update to the latest version that should work without manual intervention

mrkskwsnck, 2019-02-05:
@nachoparker, in fact I already updated to NCP version 1.5.x while writing my above comment. It didn't solve the problem.

nachoparker, 2019-02-06:
@mrkskwsnck how does your  `/etc/apache2/sites-enabled/ncp.conf` look?

mrkskwsnck, 2019-02-06:
My current `ncp.conf` file looks like:

```

nachoparker, 2019-02-07:
thanks, does changing that line for `Require ip fd00::/8` work for you?

mrkskwsnck, 2019-02-09:
Unfortunately no, it does not work with

```apache
Require ip fd00::/8
```

for me :-1: Neither via my domain nor using the IPv6 address within the browser.

n-b, 2019-06-22:
Shouldn’t we allow link-local addresses as well? (i.e.`Require ip fe80::/10`) This was needed in my case.

nachoparker, 2019-06-24:
thanks, changed to `fe80::/10`

n-b, 2019-06-24:
I suspect both `fe80::/10ˋ (link-local) and `fd00::/8` (“Unique local”) are necessary depending of the local network configuration.