nextcloud/nextcloudpi #772: In the NextCloudPi Docker container, use the IP passed into the container to resolve the $DOMAIN, rather than the container's private IP

necopinus, 2019-01-14:
**Description**  
Right now, enabling `dnsmasq` within a NextCloudPi Docker container associates your `$DOMAIN` with the container's _private_ IP address. This means that Docker images cannot [use `dnsmasq` to redirect systems on your home network to your NextCloudPi server](https://ownyourbits.com/2017/03/09/dnsmasq-as-dns-cache-server-for-nextcloudpi-and-raspbian/).

Instead, `dnsmasq` should either use the IP passed into the Docker container, or allow the IP address to be specified in `ncp-config` as part of the `dnsmasq` configuration.

**Example**  
Currently, `/etc/dnsmasq.conf` has a line that looks like:

```
address=/ncp.mydomain.tld/<private_ip_of_the_docker_container>  # This is optional if we add it to /etc/hosts
```

After this change, this line should instead be:

```
address=/ncp.mydomain.tld/<trusted_ip_passed_to_docker>  # This is optional if we add it to /etc/hosts
```

**Work-Around**
This is more of a wishlist/papercut feature request, as the work-around is pretty simple (if a bit technical): After enabling the `dnsmasq` service, `dnsmasq.conf` will be accessible from the host system in the docker `ncdata` volume directory as `/var/lib/docker/volumes/ncdata/_data/etc/dnsmasq.conf`. Stop the container, edit the file and replace the internal IP with the correct IP of your system, and then restart the container.

ovpc, 2019-01-17:
There might be another small bug:
When I first launch  a container, I use:
 > docker run -d -p 4443:4443 -p 443:443 -p 80:80 -v /media/ncdata:/data --name nextcloudpi ownyourbits/nextcloudpi-x86 $**1**192.168.0.41

I have to add an extra digit to the $DOMAIN. The first one gets discarded, so without the extra **1** the added trusted domain becomes 92.168.0.41

nachoparker, 2019-01-18:
@ovpc it would be

```
docker run -d -p 4443:4443 -p 443:443 -p 80:80 -v /media/ncdata:/data --name nextcloudpi ownyourbits/nextcloudpi-x86 192.168.0.41
```

, without the dollar

necopinus, 2019-01-21:
Pull request submitted.

Note that this only covers the case where an IP address has been passed to the docker container (if you pass a domain, or nothing at all, then the previous behavior will be used).

I _think_ I understand how NextCloudPi works well enough now that I can write something that allows the user to manually configure this using `ncp-config`, but the more I think about this the less useful it seems: If you're using the docker image, then you're probably passing in a trusted IP, and if you're running NextCloudPi directly then the existing method is probably what you want. Making the user configure the same information twice seems like it's probably making the default use case more complicated just to make a set of edge cases (which can be addressed by directly editing `dnsmasq.conf`) a little easier.

That said, if there's a preference to add this to `ncp-config`, I can submit a second pull request to do so.

Note that there's a potential edge-case here where `truseted_domains 6` winds up with an IP address that doesn't correspond to something locally addressable because some other process has added something to that slot and no IP address has been passed into the docker image (if that's how NextCloudPi is being run). It's not clear to me how likely this is, or how to test for it.

nachoparker, 2019-01-22:
Thanks!

This should work for domains too, as they get stored in trusted domains 6.

In NCP we have trusted domains 6 reserved for that, so this sounds like a good idea to me :)

Actually initially I had the option to specify the IP, but I decided that it could be automated, so I got rid of it -> [link](https://ownyourbits.com/2017/03/09/dnsmasq-as-dns-cache-server-for-nextcloudpi-and-raspbian/)

Many thanks for your PRs and suggestions

closed 2019-01-22 by nachoparker