nextcloud/nextcloudpi #774: odroid HDD 'clack noise' parking  fix from Odroid

GreyMarvel, 2019-01-14:
Hi all,
I'm not sure if this is useful or not but I stumbled across this on odroid's website. It's a workaround fix, for the hardrive parking 'clack noise ' on shutdown and I believe on the forum there is a workaround script for a reboot as well . 
I haven't run the the scripts myself as I'm not really an expert, but and would prefer a

nachoparker, 2019-01-15:
wow thank you

zedocrob, 2019-01-15:
i use this script since a few month without issue on Odroid HC2, no "clack noise" on shutdown.
But i also had to set the firmware spindown timing to "never" like that:
`root@odroid:~# ./JMS578FwUpdate -d /dev/sda -f ./JMS578_Hardkernel_v173.01.00.01.bin -b ./backup.bin -t 0`
Mean reflashing the firmware with:` -t 0` (no spindown)
Otherwise the firmware is set to 3min before spindown by default wich was causing high Power-Off_Retract_Count!

Then check if the shutdown script is still there (or reapply it).

check this thread about the firmware 3min default spindown problem!:
https://forum.odroid.com/viewtopic.php?f=97&t=31985

GreyMarvel, 2019-01-15:
Is this something that can be incorporated into a NextcloudPi update /image ?

nachoparker, 2019-01-16:
It could, but it's low priority for me. It might happen if I receive a PR

zedocrob, 2019-01-16:
That'll be cool
Also be careful with the firmware flashing, maybe detach the drive before...I've experienced a corruption after flashing the Odroid HC2 firmware, and after rebooting:
( ncdata input/output error... ). :(

funkmastergst, 2019-01-21:
Not quite sure if this is helpful, but my WDred installed in ODroid HC2 since several months shows "192 Power-Off_Retract_Count 0x0032 200 200 000 Old_age Always - 0" in Check HDD health.

nachoparker, 2019-06-01:
Assuming this fix has been included in the latest Armbian images. If it is not, then we should report it to the Armbian team.

closed 2019-06-01 by nachoparker

GreyMarvel, 2019-06-01:
It is now on the new images,  I remember seeing the updated files recently but forgot to close the issue.
Thanks nacho for your great work .