nextcloud/nextcloudpi #778: Change backup script to use multiple cores for compression

XDA-Bam, 2019-01-16:
The [backup script](https://github.com/nextcloud/nextcloudpi/blob/master/bin/ncp/BACKUPS/nc-backup.sh#L93) is currently only using one core for compressing backups. With nextcloudpi running on a SOC, even small installs can take over an hour to compress the backup (database + base files, no user data).

I propose to switch the compression from tar/gzip to tar/[pigz](https://packages.debian.org/en/stretch/pigz). pigz is a parallelized version of gzip and resulting files can still be decompressed by gzip if desired. [Benchmarks](https://community.centminmod.com/threads/compression-comparison-benchmarks-zstd-vs-brotli-vs-pigz-vs-bzip2-vs-xz-etc.12764/) suggest, that compression speed scales nicely with the number of cores. Assuming that I/O isn't a limiting factor, typical compression times on four core SOCs should be reduced by a factor of two to three.

Usage would only change very little from
`tar -cf /tmp/myFile /tmp/compressedFile`
to
`tar -I pigz -cf /tmp/myFile /tmp/compressedFile`

Additionally, the pigz package would have to be installed during the nextcloud setup process and during the update procedure.

nachoparker, 2019-01-16:
Hi,

LOL that's good timing, I am just refactoring the backup script -> https://github.com/nextcloud/nextcloudpi/commit/2419e578b600fa9d1905102bfc02aba39c1b8334

I have thought about it, I personally like `pbzip2` but for the sake of compatibility with most users I thought about using some standard "zip" format.

Good idea, I'll see into adding it

nachoparker, 2019-01-17:
Done https://github.com/nextcloud/nextcloudpi/commit/84ac075831ce3379618556d274f7bc31acd04695

THanks

closed 2019-01-17 by nachoparker

XDA-Bam, 2019-01-17:
Cool :grinning:  Thanks!