nextcloud/nextcloudpi #780: Restoring backup doesnt work

LukasWerfel, 2019-01-18:
I am using docker and I want to restore a backup as described [here](https://github.com/nextcloud/nextcloudpi/wiki/How-to-backup-and-restore-a-NextCloudPi-instance-using-ncp-config).

But I get the error message:
> Can only restore from ext/btrfs filesystems

Not sure what I am doing wrong.

ovpc, 2019-01-18:
Hi @LukasWerfel, thanks for reporting,
but would you please use the [forum](https://help.nextcloud.com/c/support/appliances-docker-snappy-vm) for questions and getting help. Thank you for including ncp-report (thru pastbin or similar).

closed 2019-01-19 by nachoparker

LukasWerfel, 2019-01-24:
Thanks, but I moved to another solution, since flaky backup is a deal breaker for my requirements.

facorazza, 2020-05-12:
Is there a solution to this? I'm trying to restore a database from a raspberry image to a docker image but I get the same error message.

azurefreecovid, 2020-05-14:
Me too! Inside a Docker volume that is running on an XFS file system. I'd really like a solution to this, even if it is a manual one at the moment

facorazza, 2020-05-14:
I was able to make it work by moving the backup file inside `/var/www/nextcloud/data` but I wasn't able to successfully restore it due to another error. I ended up restoring everything following the docs on nextcloud and it kinda worked

azurefreecovid, 2020-05-14:
Thanks for letting me know and glad you got it sorted.

I'm in the process of trying a restore using a modified version of nc-restore where I just commented out the line that does the ext/btrfs file system check

``` bash
grep -q -e ext -e btrfs <( stat -fc%T "$TMPDIR" ) || { echo "Can only restore from ext/btrfs filesystems"     >&2; exit 1; }
```

To be honest I'm not exactly clear why the check is there.

I'll post back here if it turns out to be successful (or not).

aleistah, 2021-04-05:
> I'm in the process of trying a restore using a modified version of nc-restore where I just commented out the line that does the ext/btrfs file system check

> 
> To be honest I'm not exactly clear why the check is there.

Tried that and found out, that restoring is impossible since the file has not been chown.
I did the following:
1. remove ext/btrfs file system check in ncp-backup as suggested by User:azurefreecovid to save backup on remote nfs share
2. in case of restoring copy the Backup file to local folder on nextcloudpi 
3. " chown :www-data" the backup file before restoring
4. restore the smoothly prepared backup file in nextcloudpi panel

worked like a charm when i just tried that.
[image]
[image]

fl0f, 2023-07-19:
Hi, 

restoring from nextcloud 26 to 25 on nextcloudpi 1.52.1 on a Debian Raspi

I found that commenting the line ([[ "$(stat -fc%T "${BASEDIR}")" =~ ext|btrfs|zfs ]] || { echo "Can only restore from ext/btrfs/zfs filesystems (found '$(stat -fc%T "${TMPDIR}")"     >&2; exit 1; }) in the /usr/local/bin/ncp-restore file let the process run peacefully. 

Unfortunately datat path was not backed up in this backup I I am still trying to find where my data has gone...

fl0f, 2023-07-19:
My files were backed up in /var/www/recovery/ncp-data.xxxxx directory. I moved them manually into the /var/www/nextcloud/data directory and performed nc-fix-permissions and nc-scan.

I recovered everything

The fact is that I tried to upgrade NC to version 26 from a php 7.4 debian based Raspi 4. At the end of the upgrade, the console stated that NC 26 requires php 8 + and tried to restore from backup. Then I got stuck by this filesystem check (see previous posts) and the restoration of the backup failed stating "Rollback failed! Data left at /var/www/recovery/ncp-data.f8NCny". This is the location where my files where stored for the upgrade. Then When I commented the filesystem check line in the ncp-restore script, the restoration was sucessful but my files where still in the recovery directory. The restore script performed a scan that erased all data from the database because the /var/www/nextcloud/data directory was then empty. 

In fact my file system is ext4. Maybe in future versions this filesystem check could include ext4 filesystems so that the automatic restore performed by the updater works !

fl0f, 2023-07-21:
Hi @theCalcaholic and thank you for your reply. 
Could you tell me what the latest nextcloudpi version is because I thought I had the latest currently running (1.52.1).

Thanks !

theCalcaholic, 2023-07-21:
> Hi @theCalcaholic and thank you for your reply. 
> Could you tell me what the latest nextcloudpi version is because I thought I had the latest currently running (1.52.1).
> 
> Thanks !

The latest version is v1.52.2 (see https://github.com/nextcloud/nextcloudpi/releases/tag/v1.52.2)