nextcloud/nextcloudpi #784: Update to v1.4.5 via ncp-update fails when cron.hourly is empty

necopinus, 2019-01-23:
When `/etc/cron.hourly` is empty (which can happen in docker containers), the update to v1.4.5 fails (because `chmod 755 /etc/cron.daily/* /etc/cron.hourly/*` on line 276 of `update.sh` takes an error). To account for this case, `update.sh` should test if various `/etc/cron.*` contain any files of interest before applying the fix.

Pull request incoming momentarily.

ovpc, 2019-01-23:
I can confirm this also happens on Debian9 (curl installed)
[ncp-update 1.4.1 fails to update to 1.4.5](https://pastebin.com/Xb8LstPy)

nachoparker, 2019-01-24:
Thanks for the fix!

closed 2019-01-24 by nachoparker

ovpc, 2019-01-25:
Cool, thanks for fix
When exiting ncp-config v1.4.6 I now get following output:

> sudo ncp-config
/usr/local/bin/ncp-config: line 18: local: can only be used in a function
/usr/local/bin/ncp-config: line 19: local: can only be used in a function
/usr/local/bin/ncp-config: line 20: local: can only be used in a function

ovpc, 2019-01-25:
Thanks @necopinus 
@nachoparker let me know when/if want me to run  **ncp-update devel**

PR #785 (2019-01-24): Proposed fix for #784
Test if `/etc/cron.*` directories have files before running `chmod` on them to prevent upgrade errors. Should fix #784.