nextcloud/nextcloudpi #786: Russian translation

BETEP-ok, 2019-01-24:
**Description**  
I made a translation for the json values described in the guide. But not a developer, so could not merge it properly. maybe someone with a bit more experience could help to make a pull request ?

archive with translations is here: 
[https://www.my-personal.link/index.php/s/q88GscDAAbGR8wC](url)

nachoparker, 2019-01-26:
Hello,

Many thanks for that! I would like to improve the translations process, and make it more automated, but I am pretty busy and that is low priority.

I would like to wait until then to add translations, I'll ping you then.

Cheers

BETEP-ok, 2019-01-28:
> Hello,
> 
> Many thanks for that! I would like to improve the translations process, and make it more automated, but I am pretty busy and that is low priority.
> 
> I would like to wait until then to add translations, I'll ping you then.
> 
> Cheers

thnx a lot. please let me know when a more straightforward process is in place. I`d like to see as many as possible of our Russian "comrades" using this wonderful software.
especially in the light of latest changes in government enforcement policies )))

nachoparker, 2019-05-26:
Hi, that link does not seem to work for me right now.

nachoparker, 2019-06-01:
Please, send again when you have a minute. I'll close until then

BETEP-ok, 2019-06-02:
here is the file 

[l10n_templates RU-ru.zip](https://github.com/nextcloud/nextcloudpi/files/3245133/l10n_templates.RU-ru.zip)

nachoparker, 2019-06-02:
got it, thanks!

closed 2021-10-21 by nachoparker