nextcloud/nextcloudpi #789: nc-snapshot gets wrong path /media/USBdrive/../ncp-snapshots

langfingaz, 2019-01-28:
I have my data harddrive mounted at `/media/myCloudDrive` with the snapshot folder at its default location `/media/myCloudDrive/ncp-snapshots`.
When I try to create create a snapshot with **nc-snapshot** the following error is returned:

```
Running nc-snapshot
Maintenance mode enabled
Nextcloud is in maintenance mode - no apps have been loaded

ERROR: not a btrfs filesystem: /media/myCloudDrive/../ncp-snapshots
Maintenance mode disabled
Done. Press any key...
```

The nc-snapshot script somehow does not get the correct path of my snapshot folder. If the "/../" part of the path would be "/" instead everything should work fine.

I just installed the latest Odroid image today and restored my previous nextcloud files.

**System information**
```
Running nc-info
Gathering information...
NextCloudPi version  v1.4.9
NextCloudPi image    NextCloudPi_01-12-19
distribution         Debian GNU/Linux 9 \n \l
automount            yes
USB devices          sda sdb
datadir              /media/myCloudDrive/ncdata
data in SD           no
data filesystem      btrfs
data disk usage      1.6T/2.8T
rootfs usage         4.1G/29G
swapfile             /var/swap
dbdir                /var/lib/mysql
Nextcloud check      ok
Nextcloud version    15.0.2.0
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   192.168.178.39
gateway              192.168.178.1
interface            enx001e0632a720
certificates         cloud.privacy1st.de-0001
ncp.spdns.eu-0001
NAT loopback         no
uptime               1:02
```

Any ideas, how this could be fixed? Where is the script located, can I maybe run it manually?

Greetings,
Daniel

langfingaz, 2019-01-28:
`nc-fix-permissions` doesn't get my nextcloud path neither:

```
Running nc-fix-permissions
data dir not found
Done. Press any key...
```

Nextcloud itself is up and running without problems.

langfingaz, 2019-01-28:
Ok, sorry for this false positive. The nc-snapshot script was not the problem.
After I run 
+ sudo chown -R www-data:www-data /media/myCloudDrive/ncdata/ && sudo chmod -R 0750 /media/myCloudDrive/ncdata/
+ sudo chown root:www-data /media/myCloudDrive/ncdata/.htaccess && sudo chmod 0644 /media/myCloudDrive/ncdata/.htaccess
the `nc-fix-permissions`  and the `nc-snapshot` scripts do find the datadir again.

But this raises the question of changing the `nc-fix-permissions script` to be able to handle this permission problem. Does anybody think that would be an improvement?

closed 2019-01-31 by ovpc

TobjasR, 2021-01-20:
> Ok, sorry for this false positive. The nc-snapshot script was not the problem.
> After I run
>
>     * sudo chown -R www-data:www-data /media/myCloudDrive/ncdata/ && sudo chmod -R 0750 [...]

I'm having same ERROR, now done the same fixes (commands), and still getting same ERROR...