nextcloud/nextcloudpi #790: Outdated certificate of Let’s Encrypt? Pages don't load.

nullproblemo, 2019-01-29:
My NextCloudPi worked very fine. Yesterday after "sudo ncp-update" my NextCloudPi crashed. So I installed NextCloudPi_RPi_01-13-19.img and updated it. But I can’t load the page. My browsers (Pale Moon and Firefox) told me that there was an outdated certificate of Let’s Encrypt. I can’t load the page locally either. The browsers told me the page was in a loop.  Perhaps it might be the problem I just read about and which is discussed currently with Debian server and Let’s Encrypt. During the installation there were problems with old Let’s Encrypt archives. NP couldn't find some files either. So I deleted the my sd-card waiting for a new version of NextCloudPi.
[Look](https://www.pro-linux.de/news/1/26688/debian-server-zertifikate-gef%C3%83%C2%A4hrdet-lets-encrypt-schaltet-tls-sni-01-ab.html).

ovpc, 2019-01-29:
Hi 
Thank you for using [forum](https://help.nextcloud.com/tags/ncp) for help and questions/answers.

Issue with Apache/letsencrypt have been reported fixed with v1.4.11 [here](https://help.nextcloud.com/t/apache-stops-after-update-to-nextcloudpi-v1-4-9-2019-01-25-letsencrypt/46230/12)

nullproblemo, 2019-01-29:
No, it is not fixed with version 1.4.11:
Please have look at the screenshots (browser: Pale Moon, german language) and the content of the files.
[image]
[image]
[image]

After installing NCP again:

ncp: cannot stat'/etc/letsencrypt-old/archive: no such file or directory

Path etc/letsencrypt/live is empty

Content of /etc/apache2/sites-available/nextcloud.conf

<IfModule mod_ssl.c>
  <VirtualHost _default_:443>
    DocumentRoot /var/www/nextcloud
    CustomLog /var/log/apache2/nc-access.log combined
    ErrorLog  /var/log/apache2/nc-error.log
    SSLEngine on
    SSLCertificateFile      /etc/ssl/certs/ssl-cert-snakeoil.pem
    SSLCertificateKeyFile /etc/ssl/private/ssl-cert-snakeoil.key
  </VirtualHost>
  <Directory /var/www/nextcloud/>
    Options +FollowSymlinks
    AllowOverride All
    <IfModule mod_dav.c>
      Dav off
    </IfModule>
    LimitRequestBody 0
    SSLRenegBufferSize 10486000
  </Directory>
</IfModule>

Content of /etc/apache2/sites-available/ncp.conf

Listen 4443
<VirtualHost _default_:4443>
  DocumentRoot /var/www/ncp-web
  SSLEngine on
  SSLCertificateFile      /etc/ssl/certs/ssl-cert-snakeoil.pem
  SSLCertificateKeyFile /etc/ssl/private/ssl-cert-snakeoil.key

  # 2 days to avoid very big backups requests to timeout
  TimeOut 172800

  <IfModule mod_authnz_external.c>
    DefineExternalAuth pwauth pipe /usr/sbin/pwauth
  </IfModule>

</VirtualHost>
<Directory /var/www/ncp-web/>

  AuthType Basic
  AuthName "ncp-web login"
  AuthBasicProvider external
  AuthExternal pwauth

  SetEnvIf Request_URI "^" noauth
  SetEnvIf Request_URI "^index\.php$" !noauth
  SetEnvIf Request_URI "^/$" !noauth
  SetEnvIf Request_URI "^/wizard/index.php$" !noauth
  SetEnvIf Request_URI "^/wizard/$" !noauth

  <RequireAll>

   <RequireAny>
      Require host localhost
      Require local
      Require ip 192.168
      Require ip 172
      Require ip 10
   </RequireAny>

   <RequireAny>
      Require env noauth
      Require user ncp
   </RequireAny>

  </RequireAll>

</Directory>

ovpc, 2019-01-29:
Did you make a backup, before installing a new image?
That output from terminal would be normal on a fresh install, there are no old certificates. Please use the [forum](https://help.nextcloud.com/tags/ncp).

closed 2019-01-29 by ovpc

nullproblemo, 2019-01-29:
> Did you make a backup, before installing a new image?
> That output from terminal would be normal on a fresh install, there are no old certificates. Please use the [forum](https://help.nextcloud.com/tags/ncp).

No, I didn't make a backup. Normally I only export my calendar and my contacts periodically so that I can import them into a fresh installation. I installed NCP several times before but never had such problems.

nullproblemo, 2019-01-29:
I had a look in my Fritzbox. With a click at the link of the PI I got the page with the Passwords I had to accept. Afterwards I could load the local page to get in to receive a new certificate of Let's encrypt. I never had this before.
Why did you tell me to got to the forum to delete my account afterwards?

ovpc, 2019-01-29:
Delete account? on forum is not within my power, and had nothing to do with me.

nullproblemo, 2019-01-29:
> Delete account? on forum is not within my power, and had nothing to do with me.

Ok, I thought so. Sorry!  Then it must have been someone else and I don't know why. But you answered my post, didn't you?

ovpc, 2019-01-29:
yes

nullproblemo, 2019-01-29:
Ok. I can't log in there although I received a new password. What a miracle. Nevertheless the problem is solved. Thanks for your support!

nachoparker, 2019-01-30:
Thanks @ovpc