nextcloud/nextcloudpi #791: Apache wont start after ncp-update

fosBHY, 2019-01-29:
Hello.
I installed nextcloudpi via the install skript yesterday evening.

At first it didn't work because I had nginx running, after stopping nginx and starting apache2 it worked. 
I went through the configuration in the ncp-webUI and everything went fine. At the end I noticed that ncp version in top left corner said 0.0.0 - so I clicked on update ncp. The web-UI gave some output as usual, saying that it now updated.
After that the services were not reachable.

Apache stopped.

**APACHE ERRORS**

```
sudo systemctl status apache2.service

● apache2.service - The Apache HTTP Server
   Loaded: loaded (/lib/systemd/system/apache2.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Tue 2019-01-29 10:17:14 CET; 8min ago
  Process: 18433 ExecStart=/usr/sbin/apachectl start (code=exited, status=1/FAILURE)

systemd[1]: Starting The Apache HTTP Server...
apachectl[18433]: AH00526: Syntax error on line 1 of /etc/apache2/mods-enabled/authnz_external.load:
apachectl[18433]: Invalid command 'L\xefa$Module', perhaps misspelled or defined by a module not included in the server configuration
apachectl[18433]: Action 'start' failed.
apachectl[18433]: The Apache error log may have more information.
systemd[1]: apache2.service: Control process exited, code=exited status=1
systemd[1]: Failed to start The Apache HTTP Server.
systemd[1]: apache2.service: Unit entered failed state.
systemd[1]: apache2.service: Failed with result 'exit-code'.
```

**/etc/apache2/mods-enabled/authnz_external.load is saying**
`Lïa$Module authnz_external_module /usr/lib/apakhe2/modules/mod_authnz_ezternal.so`

**are these --three-- typos of ncp?**
I corrected to 
`LoadModule authnz_external_module /usr/lib/apache2/modules/mod_authnz_external.so`
 
Now apache throws me a apachectl[5618]: Segmentation fault

I will look further into it but I have very limited knowledge of this computermagic. 

I don't want to start troubleshooting apache2 here, but this was surely caused by ncp or the way the script set things up.
 I would also like to have it fixed (and help with it) as I want to start using nextcloud, preferably with ncp and without completely reinstalling everything.

**NCP-REPORT** as it was asked for

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v1.4.11
distribution         Raspbian GNU/Linux 9 \n \l
automount            no
USB devices          sda 
datadir              /mnt/DATEN/nextcloud/data
data in SD           no
data filesystem      btrfs
data disk usage      9.7G/59G
rootfs usage         5.3G/15G
swapfile             /var/swap
dbdir                /mnt/DATEN/nextcloud/database
Nextcloud check      error
HTTPD service        down
PHP service          up
MariaDB service      down
Redis service        up
Postfix service      up
internet check       ok
port check 80        closed
[... 140 lines cut ...]
2019-01-28 23:32:10 1989165056 [Note] Server socket created on IP: '127.0.0.1'.
2019-01-28 23:32:10 1989165056 [Note] /usr/sbin/mysqld: ready for connections.
Version: '10.1.37-MariaDB-0+deb9u1'  socket: '/var/run/mysqld/mysqld.sock'  port: 3306  Raspbian 9.0
2019-01-29  0:25:22 1639437120 [Note] /usr/sbin/mysqld: Normal shutdown
2019-01-29  0:25:23 1639437120 [Note] Event Scheduler: Purging the queue. 0 events
2019-01-29  0:25:24 1463808832 [Note] InnoDB: FTS optimize thread exiting.
2019-01-29  0:25:24 1639437120 [Note] InnoDB: Starting shutdown...
2019-01-29  0:25:25 1639437120 [Note] InnoDB: Waiting for page_cleaner to finish flushing of buffer pool
2019-01-29  0:25:27 1639437120 [Note] InnoDB: Shutdown completed; log sequence number 4379520
2019-01-29  0:25:27 1639437120 [Note] /usr/sbin/mysqld: Shutdown complete

```
</details>

<details>
<summary>Nextcloud logs</summary>

```

{"reqId":"nvbuAfPT7EUjUCYrYU7b","level":3,"time":"2019-01-28T20:00:09+00:00","remoteAddr":"","user":"--","app":"cron","method":"","url":"--","message":{"Exception":"Exception","Message":"Not installed","Code":0,"Trace":[{"file":"\/var\/www\/nextcloud\/lib\/base.php","line":660,"function":"checkInstalled","class":"OC","type":"::","args":[]},{"file":"\/var\/www\/nextcloud\/lib\/base.php","line":1068,"function":"init","class":"OC","type":"::","args":[]},{"file":"\/var\/www\/nextcloud\/cron.php","line":41,"args":["\/var\/www\/nextcloud\/lib\/base.php"],"function":"require_once"}],"File":"\/var\/www\/nextcloud\/lib\/base.php","Line":277,"CustomMessage":"--"},"userAgent":"--","version":""}
{"reqId":"g26OTHLTgAA8c0UDRaHk","level":3,"time":"2019-01-28T20:16:00+00:00","remoteAddr":"","user":"--","app":"cron","method":"","url":"--","message":{"Exception":"Exception","Message":"Not installed","Code":0,"Trace":[{"file":"\/var\/www\/nextcloud\/lib\/base.php","line":660,"function":"checkInstalled","class":"OC","type":"::","args":[]},{"file":"\/var\/www\/nextcloud\/lib\/base.php","line":1068,"function":"init","class":"OC","type":"::","args":[]},{"file":"\/var\/www\/nextcloud\/cron.php","line":41,"args":["\/var\/www\/nextcloud\/lib\/base.php"],"function":"require_once"}],"File":"\/var\/www\/nextcloud\/lib\/base.php","Line":277,"CustomMessage":"--"},"userAgent":"--","version":""}
{"reqId":"cLA28swWtfpuUy3rTYoE","level":3,"time":"2019-01-28T20:30:01+00:00","remoteAddr":"","user":"--","app":"cron","method":"","url":"--","message":{"Exception":"Exception","Message":"Not installed","Code":0,"Trace":[{"file":"\/var\/www\/nextcloud\/lib\/base.php","line":660,"function":"checkInstalled","class":"OC","type":"::","args":[]},{"file":"\/var\/www\/nextcloud\/lib\/base.php","line":1068,"function":"init","class":"OC","type":"::","args":[]},{"file":"\/var\/www\/nextcloud\/cron.php","line":41,"args":["\/var\/www\/nextcloud\/lib\/base.php"],"function":"require_once"}],"File":"\/var\/www\/nextcloud\/lib\/base.php","Line":277,"CustomMessage":"--"},"userAgent":"--","version":"15.0.2.0"}

```
</details>

fosBHY, 2019-01-29:
Might be something with the automatic setup of https ?

this was when it stopped working:

```
[Tue Jan 29 00:10:06.678782 2019] [mpm_event:notice] [pid 3725:tid 1992093696] AH00493: SIGUSR1 received.  Doing graceful restart                                               │
[Tue Jan 29 00:10:06.739477 2019] [ssl:warn] [pid 3725:tid 1992093696] AH01909: localhost:4443:0 server certificate does NOT include an ID which matches the server name        │
[Tue Jan 29 00:10:06.739809 2019] [ssl:error] [pid 3725:tid 1992093696] AH02217: ssl_stapling_init_cert: can't retrieve issuer certificate! [subject: CN=pi / issuer: CN=pi / se│
[Tue Jan 29 00:10:06.739846 2019] [ssl:error] [pid 3725:tid 1992093696] AH02604: Unable to configure certificate localhost:4443:0 for stapling                                  │
[Tue Jan 29 00:10:06.746523 2019] [unique_id:alert] [pid 3725:tid 1992093696] (EAI 2)Name or service not known: AH01564: unable to find IPv4 address of "pi"                    │
[Tue Jan 29 00:10:06.747204 2019] [:emerg] [pid 3725:tid 1992093696] AH00020: Configuration Failed, exiting  
```

ovpc, 2019-01-29:
Please use the [forum](https://help.nextcloud.com/tags/ncp) to ask questions and get help.  Thank you for including output of **ncp-report** and **cat /var/log/ncp.log** thru pastbin or similar and post link on forum, if you are looking for further assistance. (Github is for Devs, bugs and wiki discussions only)
For simple yes/no or short questions, and view/take part in discussions, you can also join the NCP's [telegram](https://t.me/NextCloudPi) group.

> Segmentation fault
Can be indication of corrupt sd card, or bad power source/cable

Suggest look at output of **dmesg** for more clues and paste output to https://lmddgtfy.net/ or forum

fosBHY, 2019-01-29:
this is not a question, it is a bug, isn't it? At least to me it looks like the install script and/or ncp is screwing things up.

nachoparker, 2019-01-30:
You seem to have some corrupted files. NCP doesn't touch those files at all. You would probably get more help in the forums, but I would consider replacing your SD card first.

Thanks for reporting

closed 2019-01-30 by nachoparker