nextcloud/nextcloudpi #792: NextCloudPi Wifi Masquarading

Etheras, 2019-02-02:
**Description**  
A script to allow the use of either built-in RPI3B+ wifi adapter or external USB adapters to be used to provide internet for LAN PC(s) through the RPI

**Example**  
https://raspberrypi.stackexchange.com/questions/48307/sharing-the-pis-wifi-connection-through-the-ethernet-port

or a couple of other methods i've looked around and found through IPTABLES

nachoparker, 2019-02-05:
hi, is this a feature request? I don't get it

Etheras, 2019-02-05:
Yes i posted this as a feature request, not sure why it didn't show up as one.

nachoparker, 2019-02-06:
I don't quite see the use for this. Doesn't seem like a typical use case. Care to elaborate?

Thank you for your suggestion,

Etheras, 2019-02-11:
Closing for now as i don't have time to elaborate.
The idea is to use DNSmasq to use the RPI as a router, and i believe the configs are being overwritten on bootup by ncp, so the suggestion was for a built-in solution with a script to route traffic from eth0 to wifi0, thus providing the PC/Switch attached to eth0 with internet provided by wifi0.
Will reconsider and come back with a solution or a stronger case. Thanks for your time :)

closed 2019-02-11 by Etheras