nextcloud/nextcloudpi #794: nc-snapshot trying to create snapshots outside of data filesystem

theCalcaholic, 2019-02-06:
This [line](https://github.com/nextcloud/nextcloudpi/blob/master/bin/ncp/BACKUPS/nc-snapshot.sh#L37) in nc-snapshot.sh tries to set a directory outside of the src filesystem as btrfs-snapshot target (usually in the root FS). This will fail if the parent file system is not in btrfs format (which it usually is not).

Could it be that the variabel which was intended to be used here is `$DATADIR` instead of `$MOUNTPOINT`?

theCalcaholic, 2019-02-06:
The same would apply to [nc-snapshot-auto.sh](https://github.com/nextcloud/nextcloudpi/blob/master/bin/ncp/BACKUPS/nc-snapshot-auto.sh#L42)

theCalcaholic, 2019-02-06:
Ah, I suppose I'm missing a btrfs subvolume

closed 2019-02-06 by theCalcaholic

ekorn, 2019-05-19:
I have the same problem could you explain in more detail what you did to solve the Problem ?

Itchyfantastico, 2022-10-07:
I have the same issue.
It does work with the btrfs-snp command via SSH and leaving out the ../ directory jump

milianw, 2023-03-02:
I'm also running into this issue. what is the purpose of the relative path, or snapshotting the whole disk instead of just the datadir?