nextcloud/nextcloudpi #798: Nextcloud upgrade failed and deleted files

Letterus, 2019-02-10:
I'm running  ncp 1.6.0 on a Odroid HC-1. Update to NCP 1.6.0 worked well. But then I tried to update NC from 15.0.2 to 15.0.4 using NCP using web interface. The update failed. I don't know why, but at some time it said "unknown command upgrade". Then a rollback of the update process happened. It said the rollback was successful, but after the rollback I had a working Nextcloud installation but no data anymore. Which led to the consequence each client was asked if he whished to keep his files or delete it. In case one kept it, clients immediately started uploading all stuff back.

After that upgrading NC would not work anymore because NCP would not detect a correct version. Now I updated NC manually, set NC to maintanance mode and am now restoring data from my backup…

So I'd be really thankful if you could do some more testing on the upgrade process. Losing all data is really bad stuff. I hope it will work out now as I have a current backup (which I set up manually using rsnapshot).

nachoparker, 2019-02-11:
Hello,

Thank you for the feedback. The upgrade is tested by several people before going public and it has been working fine for months without any code change.

No data is removed during the upgrade and a backup is taken. Depending on whether you moved your datadir, your data might be somewhere else.

In any case, I would need logs and the output of the upgrade in order to investigate the issue. 

Please reopen if somebody else runs into this and if there are logs to look at

Cheers,

theCalcaholic, 2019-02-12:
Same here. Update failed, data directory is empty on x86-docker

Letterus, 2019-02-12:
Yep. I did not find the logs yet, but I do remember that it said something like "data dir is not backed up/excluded from backup" during the update process. probably it did restore NC from this backup without data dir during rollback and using that it emptied the data dir.

theCalcaholic, 2019-02-12:
I was able to reproduce this on a newly created docker container. Here's the output of nc-update-nexctcloud:

```
Running nc-update-nextcloud
Current   Nextcloud version 15.0.2.0
Available Nextcloud version 15.0.4
Download Nextcloud 15.0.4...
Back up current instance...
check free space...
Maintenance mode enabled
backup database...
backup files...
backup /var/www//nextcloud-bkp_20190212_1549989076.tar.gz generated
Maintenance mode disabled
Install Nextcloud 15.0.4...
Restore apps...
 * admin_notifications
 * audioplayer
 * bruteforcesettings
 * calendar
 * contacts
 * deck
 * drawio
 * files_clipboard
 * groupfolders
 * nextcloudpi
 * ownpad
 * polls
 * previewgenerator
 * qownnotesapi
 * sentry
 * tasks
 * user_external
Fix permissions...
Upgrade...
PHP Warning:  fileperms(): stat failed for /var/www/nextcloud/data/nextcloud.log in /data/nextcloud/lib/private/Log/File.php on line 140
{"reqId":"j2oDl9gbfiWaQStr2CaQ","level":3,"time":"2019-02-12T16:35:42+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"--","message":"fileperms(): stat failed for \/var\/www\/nextcloud\/data\/nextcloud.log at \/data\/nextcloud\/lib\/private\/Log\/File.php#140","userAgent":"--","version":"15.0.2.0"}
{"reqId":"j2oDl9gbfiWaQStr2CaQ","level":2,"time":"2019-02-12T16:35:42+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Temporary directory \/var\/www\/nextcloud\/data\/tmp is not present or writable","userAgent":"--","version":"15.0.2.0"}
Nextcloud or one of the apps require upgrade - only a limited number of commands are available
You may use your browser or the occ upgrade command to do the upgrade
Set log level to debug
Turned on maintenance mode
Updating database schema
Updated database
Checking for update of app accessibility in appstore
Checking for update of app activity in appstore
Checking for update of app audioplayer in appstore
Checking for update of app bruteforcesettings in appstore
Checking for update of app calendar in appstore
Checking for update of app cloud_federation_api in appstore
Checking for update of app comments in appstore
Checking for update of app contacts in appstore
Checking for update of app dav in appstore
Checking for update of app federatedfilesharing in appstore
Checking for update of app federation in appstore
Checking for update of app files in appstore
Checking for update of app files_pdfviewer in appstore
Checking for update of app files_sharing in appstore
Checking for update of app files_texteditor in appstore
Checking for update of app files_trashbin in appstore
Checking for update of app files_versions in appstore
Checking for update of app files_videoplayer in appstore
Checking for update of app firstrunwizard in appstore
Checking for update of app gallery in appstore
Checking for update of app groupfolders in appstore
Checking for update of app logreader in appstore
Checking for update of app lookup_server_connector in appstore
Checking for update of app nextcloud_announcements in appstore
Checking for update of app nextcloudpi in appstore
Checking for update of app notifications in appstore
Checking for update of app oauth2 in appstore
Checking for update of app password_policy in appstore
Checking for update of app previewgenerator in appstore
Checking for update of app provisioning_api in appstore
Checking for update of app qownnotesapi in appstore
Checking for update of app serverinfo in appstore
Checking for update of app sharebymail in appstore
Checking for update of app support in appstore
Checking for update of app survey_client in appstore
Checking for update of app systemtags in appstore
Checking for update of app tasks in appstore
Checking for update of app theming in appstore
Checking for update of app twofactor_backupcodes in appstore
Checking for update of app updatenotification in appstore
Checking for update of app workflowengine in appstore
Repair warning: Unable to clear the frontend cache
Starting code integrity check...
Finished code integrity check
Update successful
Turned off maintenance mode
Reset log level
Check indices of the share table.
Done.

Backup stored at /var/www/nextcloud/data/ncp-update-backups/nextcloud-bkp_20190212_1549989076-15.0.2.0.tar.gz
Clean up...
rm: cannot remove '/data/nextcloud-old/data': Operation not permitted
Abort
Clean up...
rm: cannot remove '/data/nextcloud-old/data': Operation not permitted
Rolling back to backup /var/www/nextcloud/data/ncp-update-backups/nextcloud-bkp_20190212_1549989076-15.0.2.0.tar.gz...
extracting backup file /var/www/nextcloud/data/ncp-update-backups/nextcloud-bkp_20190212_1549989076-15.0.2.0.tar.gz...
restore files...
mv: cannot stat '/var/www/nextcloud/data/ncp-update-backups/ncp-restore.Z5vuxp/nextcloud': No such file or directory
Error restoring base files
Cleanup...
Rollback failed! Data left at /var/www/ncp-data.WC1J6T
Done. Press any key...
```

/var/www/ncp-data.WC1J6T is empty and so is /var/www/nextcloud-old/data

theCalcaholic, 2019-02-12:
It should be noted that the updated seemed to time out - between that 'Done.' and the 'Backup stored...' passed a few minutes

theCalcaholic, 2019-02-12:
I think it might have to do with the data directory being a btrfs subvolume... :thinking:

theCalcaholic, 2019-02-12:
Okay I learned a few things:

1. Docker does not support btrfs a s volume file system
2. ncp still tries creates a btrfs subvolume when the underlying file system of the mounted docker volume is btrfs
3. Deleting a btrfs subvolume fails with 'Operation not permitted'

However this can't have been my issue the first time this happened, because the datadir wasn't in a btrfs fs back then.

@nachoparker Is there a way to have ncp in docker with a btrfs file system without anything breaking, while setting up snapshots and on the host?

nachoparker, 2019-02-13:
@Letterus please check `/var/log/ncp.log`

nachoparker, 2019-02-13:
@theCalcaholic yours is a different case, I 

1. would say docker should support BTRFS for the volume, but `nc-update-nextcloud` should be adapted.
2. yes, that's good
3. because it's doing `rm- r`, and it should be doing `btrfs subvolume delete`, that's really the issue you had

nachoparker, 2019-02-14:
@theCalcaholic ok, for now I have added a check for a BTRFS `$BASEDIR`

nachoparker, 2019-02-15:
please reopen if somebody else sees this, and attach logs

closed 2019-02-15 by nachoparker