nextcloud/nextcloudpi #801: Nextcloudpi problem when I reconect

pruna1971, 2019-02-11:
good morning . I followed the tutorial on internet from PcMac and it worked very well , the problem was when I disconnected the raspberry pi from the electric current to change it. There is no way to enter via web. neither with my IP nor adding: 4443. Any idea of what happened?﻿

this is te tutoria lhttps://www.youtube.com/watch?v=FKmyvXFwwtY&t=1078s

my ip is static

nachoparker, 2019-02-11:
You must never unplug the pi without shutting it down cleanly. You might have damaged your SD card.

Please, use the forums, it is very clearly stated in the instructions that this is not for support, but for bugs.

Cheers

closed 2019-02-11 by nachoparker

pruna1971, 2019-02-16:
Good afternoon . I bought a new 64GB card and I have made a new installation of nextcloud through docker. all perfect . It works very well . but when I restart the raspberry pi then there is no way to connect again. I never unplug the raspberry directly. What could be the problem? It seems like some configuration. Thank you ...