nextcloud/nextcloudpi #803: docker run failing consistently

theCalcaholic, 2019-02-12:
When trying to create a new docker container form ownyourbits/nextcloudpi-x86 it failes when trying to setup nc-admin (after clicking on activate in the activation page).

ncp.log says:

```
[ nc-admin ]
Cannot load Zend OPcache - it was already loaded
Nextcloud is not installed - only a limited number of commands are available

                                                          
  There are no commands defined in the "user" namespace.
```

Apparently `occ maintenance:install has never been executed`

ovpc, 2019-02-12:
Strange mine works fine with:

 > docker run -d -p 4443:4443 -p 443:443 -p 80:80 -v /media/ncdata:/data --name nextcloudpi ownyourbits/nextcloudpi-x86 192.168.x.xx

The
 > Cannot load Zend OPcache - it was already loaded

Is a [bug](https://github.com/nextcloud/nextcloudpi/issues/416) already reported multiple times

I think unrelated, anyway that it is harmless, just 2 opcache start files, and does not affect anything.

 > Nextcloud is not installed - only a limited number of commands are available
   There are no commands defined in the "user" namespace.

I would try running nc-init, if its a fresh install, and watch for **init done** in ncp.log

theCalcaholic, 2019-02-12:
Hm interestingly, mysql seems to be failing to start - no idea why

closed 2019-02-12 by theCalcaholic