nextcloud/nextcloudpi #804: Nfs service fails to start on system startup

manofthepeace, 2019-02-13:
Hi, 
I am using NCP 1.6.0/NC 15.0.4, I also did setup nfs via ncp-web. At first it all worked fine. But once I did a reboot I could see the nfs service failing to start. I see `ncp-delay-automount.conf` in `/etc/systemd/system/nfs-server.service.d`. but I do not see the wait happening at all during boot.

journalctl -b;
```
Feb 13 16:42:23 rock64 systemd[1]: nfs-server.service: Control process exited, code=exited status=1
Feb 13 16:42:23 rock64 systemd[1]: Failed to start NFS server and services.
Feb 13 16:42:24 rock64 systemd[1]: nfs-server.service: Unit entered failed state.
Feb 13 16:42:24 rock64 systemd[1]: nfs-server.service: Failed with result 'exit-code'
```

systemctl status;
```
root@rock64:system/nfs-server.service.d # systemctl status nfs-server.service
● nfs-server.service - NFS server and services
   Loaded: loaded (/lib/systemd/system/nfs-server.service; enabled; vendor preset: enabled)
  Drop-In: /etc/systemd/system/nfs-server.service.d
           └─ncp-delay-automount.conf
   Active: failed (Result: exit-code) since Wed 2019-02-13 16:42:23 EST; 10min ago
  Process: 894 ExecStartPre=/usr/sbin/exportfs -r (code=exited, status=1/FAILURE)

Feb 13 16:42:23 rock64 systemd[1]: Starting NFS server and services...
Feb 13 16:42:23 rock64 exportfs[894]: exportfs: Failed to stat /media/myCloudDrive/ncdata/bob/files/shared: No such file or directory
Feb 13 16:42:23 rock64 systemd[1]: nfs-server.service: Control process exited, code=exited status=1
Feb 13 16:42:23 rock64 systemd[1]: Failed to start NFS server and services.
Feb 13 16:42:24 rock64 systemd[1]: nfs-server.service: Unit entered failed state.
Feb 13 16:42:24 rock64 systemd[1]: nfs-server.service: Failed with result 'exit-code'.
```

Only way to make it work is to automatically run `systemctl start nfs-server.service` Which will then wait for 30 sec (which is not needed at this point).

Maybe this systemd unit param coulr be used `RequiresMountsFor` otherwise maybe udiskie could be set in dependencies (but still could fail if mount is not yet mounted even is service is started?).

**System information**
Cannot load Zend OPcache - it was already loaded
{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "11": "***REMOVED SENSITIVE VALUE***",
            "1": "192.168.1.201",
            "5": "nextcloudpi.local",
            "7": "nextcloudpi",
            "8": "nextcloudpi.lan",
        },
        "datadirectory": "***REMOVED SENSITIVE VALUE***",
        "dbtype": "mysql",
        "version": "15.0.4.0",
        "overwrite.cli.url": "https:\/\/bmart.ddns.net\/",
        "dbname": "***REMOVED SENSITIVE VALUE***",
        "dbhost": "***REMOVED SENSITIVE VALUE***",
        "dbport": "",
        "dbtableprefix": "oc_",
        "mysql.utf8mb4": true,
        "dbuser": "***REMOVED SENSITIVE VALUE***",
        "dbpassword": "***REMOVED SENSITIVE VALUE***",
        "installed": true,
        "instanceid": "***REMOVED SENSITIVE VALUE***",
        "mail_smtpmode": "smtp",
        "mail_smtpauthtype": "LOGIN",
        "mail_from_address": "***REMOVED SENSITIVE VALUE***",
        "mail_domain": "***REMOVED SENSITIVE VALUE***",
        "overwriteprotocol": "https",
        "maintenance": false,
        "tempdirectory": "\/media\/myCloudDrive\/ncdata\/tmp",
        "logfile": "\/media\/myCloudDrive\/ncdata\/nextcloud.log",
        "htaccess.RewriteBase": "\/",
        "mail_sendmailmode": "smtp",
        "mail_smtpsecure": "tls",
        "mail_smtpauth": 1,
        "mail_smtphost": "***REMOVED SENSITIVE VALUE***",
        "mail_smtpport": "587",
        "mail_smtpname": "***REMOVED SENSITIVE VALUE***",
        "mail_smtppassword": "***REMOVED SENSITIVE VALUE***",
        "loglevel": 2,
        "memcache.local": "\\OC\\Memcache\\Redis",
        "memcache.locking": "\\OC\\Memcache\\Redis",
        "redis": {
            "host": "***REMOVED SENSITIVE VALUE***",
            "port": 0,
            "timeout": 0,
            "password": "***REMOVED SENSITIVE VALUE***"
        }
    }
}

nachoparker, 2019-02-14:
Hello,

in `/usr/lib/systemd/system/nc-automount.service` you have 

```
Before=mysqld.service dphys-swapfile.service fail2ban.service smbd.service nfs-server.service
```
, so it is already in the dependencies, and in `/etc/systemd/system/nfs-server.service.d/ncp-delay-automount.conf` we have

```
ExecStartPre=/bin/bash -c "/bin/sleep 30; /usr/sbin/exportfs -r"
```

Now that I look at it, it should only be

```
ExecStartPre=/usr/sbin/exportfs -r
```

It seems to be enough to do a sleep, but if you find a better solution please let me know. I would try that change in `/etc/systemd/system/nfs-server.service.d/ncp-delay-automount.conf` first

manofthepeace, 2019-02-14:
Hi,
thanks for your quick response.

So I tried modifying `/etc/systemd/system/nfs-server.service.d/ncp-delay-automount.conf` by just having 
`ExecStartPre=/usr/sbin/exportfs -r` but same thing happens.

So a bit more I can add; 

From journalctl -b,

I see nc-automount running at : ***21:47:04***  then right after during the same second I see NFS startup failing.Then few seconds later I see;
```
Feb 13 21:47:09 rock64 udisksd[1741]: Cleaning up mount point /media/myCloudDrive (device 8:1 is not mounted)  
```
Then looks like it gets mounted a bit later
```
Feb 13 21:47:09 rock64 nc-automount-links-mon[831]: myCloudDrive CREATE,ISDIR
Feb 13 21:47:10 rock64 kernel: BTRFS info (device sda1): disk space caching is enabled
Feb 13 21:47:10 rock64 kernel: BTRFS: has skinny extents
Feb 13 21:47:10 rock64 systemd[1]: Reloading OpenBSD Secure Shell server.
Feb 13 21:47:10 rock64 sshd[916]: Received SIGHUP; restarting.
Feb 13 21:47:10 rock64 systemd[1]: Reloaded OpenBSD Secure Shell server.                                                Feb 13 21:47:10 rock64 sshd[916]: Server listening on 0.0.0.0 port 22.
Feb 13 21:47:10 rock64 sshd[916]: Server listening on :: port 22.
Feb 13 21:47:10 rock64 udisksd[1741]: Mounted /dev/sda1 at /media/myCloudDrive on behalf of uid 0 
```

So it looks like the automount is still running while nfs tries to start, or that its kind of async.

Without the modification to the .conf, ith the 30 sec wait, I am still unsure why on boot it does not was for 30 sec.

nachoparker, 2019-02-14:
The sleep should be on `/etc/systemd/system/nfs-server.service.d/ncp-delay-automount.conf`, sorry, my mistake

manofthepeace, 2019-02-14:
ok looks like I have a simple fix.

It seems like `ExecStart*` units are additive. the one for nfs-service already contains 
`ExecStartPre=/usr/sbin/exportfs -r`

So we need to reset it, so ncp-delay-automount.conf becomes;
```
[Service]                                                                                                               
ExecStartPre=
ExecStartPre=/bin/bash -c "/bin/sleep 30; /usr/sbin/exportfs -r"  
```

This would ba valid for all the ones where ExecStartPre is already set in the service itself

manofthepeace, 2019-02-14:
Just having `/bin/sleep 30` in ncp-delay-automount.conf causes the same issue since only ExecStartPre from the service is taken; `Process: 893 ExecStartPre=/usr/sbin/exportfs -r `

when using
[Service]                                                                                                               
ExecStartPre=
ExecStartPre=/bin/bash -c "/bin/sleep 30; /usr/sbin/exportfs -r"

I do get;
```
Process: 2161 ExecStart=/usr/sbin/rpc.nfsd $RPCNFSDARGS (code=exited, status=0/SUCCESS)
Process: 912 ExecStartPre=/bin/bash -c /bin/sleep 30; /usr/sbin/exportfs -r (code=exited, status=0/SUCCESS)  
```

nachoparker, 2019-02-14:
good job, so the fix is to have `ncp-delay-automount.conf` as 

```
[Service]
ExecStartPre=
ExecStartPre=/bin/bash -c "/bin/sleep 30; /usr/sbin/exportfs -r"
```
? 

Does that solve the issue?

manofthepeace, 2019-02-15:
Yes that would be it! Thank you.

For info I looked at where  `ncp-delay-automount.conf` was used also. Looks like the same behaviour could happen for the mariadb service, if using external storage (not my case) since it defines a ExecStartPre already. Maybe not if the service can start without the mount (which was not the case for nfs) and deal with it later.

Maybe better to not touch the others for now until something similar is reported.

closed 2019-02-15 by manofthepeace

nachoparker, 2019-02-15:
You're right.

well, thank you very much. I will add the fix for everyone

cheers