nextcloud/nextcloudpi #807: Update to 1.7.0 from 1.6.3 on rock64 does not work

gadric, 2019-02-20:
Hi, 

I tried to update my rock64 system to 1.7.0 from 1.6.3. The update stopped with the following error: 

> root@xxxxxxx:~# ncp-update
Downloading updates
Performing updates
Installing nc-backup
Hit:1 http://security.debian.org stretch/updates InRelease
Ign:3 http://cdn-fastly.deb.debian.org/debian stretch InRelease             
Get:5 http://cdn-fastly.deb.debian.org/debian stretch-updates InRelease [91.0 kB]
Hit:2 https://apt.armbian.com stretch InRelease           
Get:6 http://cdn-fastly.deb.debian.org/debian stretch-backports InRelease [91.9 kB]
Hit:7 http://cdn-fastly.deb.debian.org/debian stretch Release               
Get:4 https://cdn.sury.org/php stretch InRelease [6934 B]
Get:9 https://cdn.sury.org/php stretch/main arm64 Packages [104 kB]
Err:9 https://cdn.sury.org/php stretch/main arm64 Packages
  Writing more data than expected (104419 > 103705)
  Hashes of expected file:
   - Filesize:103705 [weak]
   - SHA256:cc879f82cca5e3d1b50b21736c26132426e1c3ea6bf890a686ec86b12b58a6f1
   - SHA1:88752152f3012e72a3025d3571af029b6ba2bb8e [weak]
   - MD5Sum:c12063ef26d72ec861440150b2e21b8b [weak]
  Release file created at: Tue, 19 Feb 2019 09:12:38 +0000
Get:10 https://cdn.sury.org/php stretch/main armhf Packages [105 kB]
Fetched 295 kB in 2s (99.8 kB/s)    
Reading package lists... Done
E: Failed to fetch https://cdn.sury.org/php/dists/stretch/main/binary-arm64/Packages.gz  Writing more data than expected (104419 > 103705)
   Hashes of expected file:
    - Filesize:103705 [weak]
    - SHA256:cc879f82cca5e3d1b50b21736c26132426e1c3ea6bf890a686ec86b12b58a6f1    - SHA1:88752152f3012e72a3025d3571af029b6ba2bb8e [weak]
    - MD5Sum:c12063ef26d72ec861440150b2e21b8b [weak]
   Release file created at: Tue, 19 Feb 2019 09:12:38 +0000
E: Some index files failed to download. They have been ignored, or old ones used instead.

suxlex, 2019-02-20:
I get the same error...

shadowstreik, 2019-02-20:
A link for support is available on the NextCloudPi Mastodon page...

https://mastodon.social/@NextCloudPi

Thank you.

gadric, 2019-02-20:
Seems to be fixed now ...

closed 2019-02-26 by gadric