nextcloud/nextcloudpi #808: FreeDNS update IPV4 and IPV6

pehebo, 2019-02-20:
At the moment you can only update the ipv4 OR ipv6 record of FREEDNS, as there is only one entry possible in ncp-config.
But FREEDNS offers an new update service (sync.afraid.org) with a randomly generated update token that supports autodetection of ipv4 and ipv6 addresses. You need to enter two URL-strings, separated by <blank> to update both records.
Is it possible to implement this into nextcloudpi?
Thank you very much, NCP is still a very very welldone project!
Best regards, Peter

shadowstreik, 2019-02-20:
A link for support is available on the NextCloudPi Mastodon page...

https://mastodon.social/@NextCloudPi

Thank you.

TinoPfleiderer, 2019-11-09:
Hey,
I am a newbie but I am interested to contribute and have already a script running on a raspberry with the new update service. Although it doesn`t autodetect if ipv4 or ipv6 is in use. You have to make separate calls.  
The only issues I have is that I am not sure what to do in order to contribute. Give me a hint to a documentation and I will get all things running.
Best Regards
Tino

nachoparker, 2019-11-10:
Hi,

thanks for your offer! our documentation is a bit outdated

https://docs.nextcloudpi.com/en/ncp-app-developer-guide/

but you can look for examples in the current code. Basically each ncp-app is a shell file + a json config file that defines variables that will be editable from the web and accessible from the shell script.