nextcloud/nextcloudpi #811: Update from v1.6.3 to v1.7.0 on Rasbpi 3B+ doesn not work (compile error)

tim-peters, 2019-02-21:
Updating ncp from version 1.6.3 to 1.7.0 via the `ncp-update` command does not work on my Raspberry Pi 3 Model B+. The update process stops with the following error:
```
Downloading updates
Performing updates
jq: error: Invalid escape at line 1, column 4 (while parsing '"\'"') at <top-level>, line 1:
.params[1].value = "u=G]bQ42I'\''Tgg^(3"
jq: 1 compile error
```

The updates before worked well and I did not change anything on my setup.

**System information**

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v1.6.3
NextCloudPi image    NextCloudPi_01-14-19
distribution         Raspbian GNU/Linux 9 \n \l
automount            yes
USB devices          sda
datadir              /media/USBdrive/ncdata
data in SD           no
data filesystem      btrfs
data disk usage      299G/1.9T
rootfs usage         2.1G/15G
swapfile             /var/swap
dbdir                /var/lib/mysql
Nextcloud check      ok
Nextcloud version    15.0.2.0
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        closed
port check 443       closed
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            wlan0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               1min

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "5": "nextcloudpi.local",
            "7": "nextcloudpi",
            "8": "nextcloudpi.lan",
            "1": "192.168.178.21",
            "3": "***REMOVED SENSITIVE VALUE***"
        },
        "datadirectory": "***REMOVED SENSITIVE VALUE***",
        "dbtype": "mysql",
        "version": "15.0.2.0",
        "overwrite.cli.url": "***REMOVED SENSITIVE VALUE***",
        "dbname": "***REMOVED SENSITIVE VALUE***",
        "dbhost": "***REMOVED SENSITIVE VALUE***",
        "dbport": "",
        "dbtableprefix": "oc_",
        "mysql.utf8mb4": true,
        "dbuser": "***REMOVED SENSITIVE VALUE***",
        "dbpassword": "***REMOVED SENSITIVE VALUE***",
        "installed": true,
        "instanceid": "***REMOVED SENSITIVE VALUE***",
        "memcache.local": "\\OC\\Memcache\\Redis",
        "memcache.locking": "\\OC\\Memcache\\Redis",
        "redis": {
            "host": "***REMOVED SENSITIVE VALUE***",
            "port": 0,
            "timeout": 0,
            "password": "***REMOVED SENSITIVE VALUE***"
        },
        "tempdirectory": "\/media\/USBdrive\/ncdata\/tmp",
        "mail_smtpmode": "smtp",
        "mail_smtpauthtype": "LOGIN",
        "mail_from_address": "***REMOVED SENSITIVE VALUE***",
        "mail_domain": "***REMOVED SENSITIVE VALUE***",
        "overwriteprotocol": "https",
        "loglevel": "2",
        "log_type": "file",
        "maintenance": false,
        "logfile": "\/media\/USBdrive\/ncdata\/nextcloud.log",
        "mail_sendmailmode": "smtp",
        "mail_smtpsecure": "tls",
        "mail_smtpauth": 1,
        "mail_smtphost": "***REMOVED SENSITIVE VALUE***",
        "mail_smtpport": "587",
        "mail_smtpname": "***REMOVED SENSITIVE VALUE***",
        "mail_smtppassword": "***REMOVED SENSITIVE VALUE***"
    }
}

```
</details>

nachoparker, 2019-02-22:
Seems like you have some unsupported characters, probably in one of your passwords.

Since you just shared said password with us it will be better to change it anyway ;)

Then you should be able to update.

tim-peters, 2019-02-22:
Thanks for your quick reply!

I can confirm that this was my admin password (didn't realize since I use a password manager). After changing it the update went without issues.

I still believe this to be a bug. Using a secure password (or whatever password) should not prevent the system from updating. Don't you think?

nachoparker, 2019-02-22:
Nice

Ideally, but I don't consider this to be a big deal. I am happy to accept PRs with fixes

Cheers,

closed 2019-02-22 by nachoparker