nextcloud/nextcloudpi #812: Doctrine\DBAL\DBALException: Failed to connect to the database

piexad, 2019-02-22:
Hello, 

Three days ago I made some updates on my RPi and I remember that there was like 70 updates available (even if I update my RPi really often). 
Since this time, my NCP is down. When I try to connect to it I get a white page and HTTP 500 error with DAVx5 and NEWS app. 

The issue seems to be this : 
Doctrine\DBAL\DBALException: Failed to connect to the database: An exception occured in driver: SQLSTATE[HY000] [1698] Access denied for user 'ncadmin'@'localhost' in /var/www/nextcloud/lib/private/DB/Connection.php:64

Unfortunately I really don't understand what it means...

Thank you for your help !

**System information**

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v1.7.0
NextCloudPi image    NextCloudPi_02-06-18
distribution         Raspbian GNU/Linux 9 \n \l
automount            yes
USB devices          sda
datadir              /media/USBdrive/ncdata
data in SD           no
data filesystem      btrfs
data disk usage      304G/932G
rootfs usage         3.8G/30G
swapfile             /var/swap
dbdir                /media/USBdrive/ncdatabase
Nextcloud check      error
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            enxb827eb5031b9
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         yes
uptime               25min

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

An unhandled exception has been thrown:
Doctrine\DBAL\DBALException: Failed to connect to the database: An exception occured in driver: SQLSTATE[HY000] [1698] Access denied for user 'ncadmin'@'localhost' in /var/www/nextcloud/lib/private/DB/Connection.php:64
Stack trace:
#0 /var/www/nextcloud/3rdparty/doctrine/dbal/lib/Doctrine/DBAL/Connection.php(429): OC\DB\Connection->connect()
#1 /var/www/nextcloud/3rdparty/doctrine/dbal/lib/Doctrine/DBAL/Connection.php(389): Doctrine\DBAL\Connection->getDatabasePlatformVersion()
#2 /var/www/nextcloud/3rdparty/doctrine/dbal/lib/Doctrine/DBAL/Connection.php(328): Doctrine\DBAL\Connection->detectDatabasePlatform()
#3 /var/www/nextcloud/3rdparty/doctrine/dbal/lib/Doctrine/DBAL/Connection.php(623): Doctrine\DBAL\Connection->getDatabasePlatform()
#4 /var/www/nextcloud/lib/private/DB/Connection.php(151): Doctrine\DBAL\Connection->setTransactionIsolation(2)
#5 /var/www/nextcloud/3rdparty/doctrine/dbal/lib/Doctrine/DBAL/DriverManager.php(172): OC\DB\Connection->__construct(Array, Object(Doctrine\DBAL\Driver\PDOMySql\Driver), Object(Doctrine\DBAL\Configuration), Object(Doctrine\Common\EventManager))
#6 /var/www/nextcloud/lib/private/DB/ConnectionFactory.php(158): Doctrine\DBAL\DriverManager::getConnection(Array, Object(Doctrine\DBAL\Configuration), Object(Doctrine\Common\EventManager))
#7 /var/www/nextcloud/lib/private/Server.php(641): OC\DB\ConnectionFactory->getConnection('mysql', Array)
#8 /var/www/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php(118): OC\Server->OC\{closure}(Object(OC\Server))
#9 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(117): Pimple\Container->offsetGet('OCP\\IDBConnecti...')
#10 /var/www/nextcloud/lib/private/ServerContainer.php(132): OC\AppFramework\Utility\SimpleContainer->query('OCP\\IDBConnecti...')
#11 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(165): OC\ServerContainer->query('OCP\\IDBConnecti...')
[... 109 lines cut ...]
```

{"reqId":"XGugiX8AAQEAAD7U8i4AAAUF","level":0,"time":"2019-02-19T06:22:02+00:00","remoteAddr":"192.168.1.1","user":"--","app":"core","method":"GET","url":"\/index.php\/apps\/news\/api\/v1-2\/user","message":"Scss is disabled for \/var\/www\/nextcloud\/core\/css\/css-variables.scss, ignoring","userAgent":"okhttp\/3.8.0","version":"15.0.2.0"}
{"reqId":"XGugiX8AAQEAAD7U8i4AAAUF","level":0,"time":"2019-02-19T06:22:02+00:00","remoteAddr":"192.168.1.1","user":"--","app":"core","method":"GET","url":"\/index.php\/apps\/news\/api\/v1-2\/user","message":"Scss is disabled for \/var\/www\/nextcloud\/core\/css\/jquery.ocdialog.scss, ignoring","userAgent":"okhttp\/3.8.0","version":"15.0.2.0"}
{"reqId":"XGugiX8AAQEAAD7U8i4AAAUF","level":0,"time":"2019-02-19T06:22:02+00:00","remoteAddr":"192.168.1.1","user":"--","app":"core","method":"GET","url":"\/index.php\/apps\/news\/api\/v1-2\/user","message":"Scss is disabled for \/var\/www\/nextcloud\/core\/search\/css\/results.scss, ignoring","userAgent":"okhttp\/3.8.0","version":"15.0.2.0"}
{"reqId":"XGugiX8AAQEAAD7V@KYAAEAE","level":0,"time":"2019-02-19T06:22:03+00:00","remoteAddr":"192.168.1.1","user":"--","app":"core","method":"GET","url":"\/index.php\/apps\/news\/api\/v1-2\/feeds","message":"Scss is disabled for \/var\/www\/nextcloud\/core\/css\/jquery-ui-fixes.scss, ignoring","userAgent":"okhttp\/3.8.0","version":"15.0.2.0"}
{"reqId":"XGugiX8AAQEAAD7V@KYAAEAE","level":0,"time":"2019-02-19T06:22:03+00:00","remoteAddr":"192.168.1.1","user":"--","app":"core","method":"GET","url":"\/index.php\/apps\/news\/api\/v1-2\/feeds","message":"Scss is disabled for \/var\/www\/nextcloud\/core\/css\/server.scss, ignoring","userAgent":"okhttp\/3.8.0","version":"15.0.2.0"}
{"reqId":"XGugiX8AAQEAAD7V@KYAAEAE","level":0,"time":"2019-02-19T06:22:03+00:00","remoteAddr":"192.168.1.1","user":"--","app":"core","method":"GET","url":"\/index.php\/apps\/news\/api\/v1-2\/feeds","message":"Scss is disabled for \/var\/www\/nextcloud\/core\/css\/css-variables.scss, ignoring","userAgent":"okhttp\/3.8.0","version":"15.0.2.0"}
{"reqId":"XGugiX8AAQEAAD7V@KYAAEAE","level":0,"time":"2019-02-19T06:22:03+00:00","remoteAddr":"192.168.1.1","user":"--","app":"core","method":"GET","url":"\/index.php\/apps\/news\/api\/v1-2\/feeds","message":"Scss is disabled for \/var\/www\/nextcloud\/core\/css\/jquery.ocdialog.scss, ignoring","userAgent":"okhttp\/3.8.0","version":"15.0.2.0"}
{"reqId":"XGugiX8AAQEAAD7V@KYAAEAE","level":0,"time":"2019-02-19T06:22:03+00:00","remoteAddr":"192.168.1.1","user":"--","app":"core","method":"GET","url":"\/index.php\/apps\/news\/api\/v1-2\/feeds","message":"Scss is disabled for \/var\/www\/nextcloud\/core\/search\/css\/results.scss, ignoring","userAgent":"okhttp\/3.8.0","version":"15.0.2.0"}
{"reqId":"XGugcX8AAQEAAD7U8iwAAA0g","level":1,"time":"2019-02-19T06:22:09+00:00","remoteAddr":"192.168.1.1","user":"--","app":"updater","method":"GET","url":"\/core\/ajax\/update.php?requesttoken=DCHkmfw3qBlF8H0UDdcvdFmyO4%2FhK50tJCZvKdlfyGM%3D%3ASWzS3K4Hw3MdnS1QYPwaOTzodLetSs9JXnAHHLMUnSE%3D","message":"\\OC\\Repair::info: Repair info: JS cache cleared","userAgent":"Mozilla\/5.0 (X11; Ubuntu; Linux x86_64; rv:65.0) Gecko\/20100101 Firefox\/65.0","version":"15.0.2.0"}
{"reqId":"XGugcX8AAQEAAD7U8iwAAA0g","level":1,"time":"2019-02-19T06:22:09+00:00","remoteAddr":"192.168.1.1","user":"--","app":"updater","method":"GET","url":"\/core\/ajax\/update.php?requesttoken=DCHkmfw3qBlF8H0UDdcvdFmyO4%2FhK50tJCZvKdlfyGM%3D%3ASWzS3K4Hw3MdnS1QYPwaOTzodLetSs9JXnAHHLMUnSE%3D","message":"\\OC\\Repair::step: Repair step: Clear every generated avatar on major updates","userAgent":"Mozilla\/5.0 (X11; Ubuntu; Linux x86_64; rv:65.0) Gecko\/20100101 Firefox\/65.0","version":"15.0.2.0"}
{"reqId":"XGugcX8AAQEAAD7U8iwAAA0g","level":1,"time":"2019-02-19T06:22:09+00:00","remoteAddr":"192.168.1.1","user":"--","app":"updater","method":"GET","url":"\/core\/ajax\/update.php?requesttoken=DCHkmfw3qBlF8H0UDdcvdFmyO4%2FhK50tJCZvKdlfyGM%3D%3ASWzS3K4Hw3MdnS1QYPwaOTzodLetSs9JXnAHHLMUnSE%3D","message":"\\OC\\Repair::step: Repair step: Add preview background cleanup job","userAgent":"Mozilla\/5.0 (X11; Ubuntu; Linux x86_64; rv:65.0) Gecko\/20100101 Firefox\/65.0","version":"15.0.2.0"}
{"reqId":"XGugcX8AAQEAAD7U8iwAAA0g","level":1,"time":"2019-02-19T06:22:09+00:00","remoteAddr":"192.168.1.1","user":"--","app":"updater","method":"GET","url":"\/core\/ajax\/update.php?requesttoken=DCHkmfw3qBlF8H0UDdcvdFmyO4%2FhK50tJCZvKdlfyGM%3D%3ASWzS3K4Hw3MdnS1QYPwaOTzodLetSs9JXnAHHLMUnSE%3D","message":"\\OC\\Repair::step: Repair step: Queue a one-time job to cleanup old backups of the updater","userAgent":"Mozilla\/5.0 (X11; Ubuntu; Linux x86_64; rv:65.0) Gecko\/20100101 Firefox\/65.0","version":"15.0.2.0"}
[... cut ...] }
{"reqId":"XGugcX8AAQEAAD7U8iwAAA0g","level":1,"time":"2019-02-19T06:22:39+00:00","remoteAddr":"192.168.1.1","user":"--","app":"updater","method":"GET","url":"\/core\/ajax\/update.php?requesttoken=DCHkmfw3qBlF8H0UDdcvdFmyO4%2FhK50tJCZvKdlfyGM%3D%3ASWzS3K4Hw3MdnS1QYPwaOTzodLetSs9JXnAHHLMUnSE%3D","message":"\\OC\\Updater::resetLogLevel: Reset log level to Warning(2)","userAgent":"Mozilla\/5.0 (X11; Ubuntu; Linux x86_64; rv:65.0) Gecko\/20100101 Firefox\/65.0","version":"15.0.4.0"}

```
</details>

shadowstreik, 2019-02-23:
A link for support is available on the NextCloudPi Mastodon page...

https://mastodon.social/@NextCloudPi

Thank you.

Medvosteri, 2019-02-23:
You mean going to mastodon, just to click on a link to go to the nextcloud/ncp [Forum](https://help.nextcloud.com/c/support/appliances-docker-snappy-vm)?

closed 2019-03-09 by nachoparker