nextcloud/nextcloudpi #813: Configuration Panel is still empty on any menu points

kpleines, 2019-02-23:
**System information**
<--! Paste this in GitHub report -->

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v1.7.0
NextCloudPi image    NextCloudPi_docker_11-30-18
distribution         Debian GNU/Linux 9 \n \l
automount            no
USB devices          sdc 
datadir              /data/nextcloud/data
data in SD           yes
data filesystem      ext2/ext3
data disk usage      1.2T/1.8T
rootfs usage         1.2T/1.8T
swapfile             /share/CACHEDEV1_DATA/.swap/qnap_swap
dbdir                /data/database
Nextcloud check      ok
Nextcloud version    15.0.4.0
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        open
port check 443       closed
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               21days

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "1": "192.168.178.3",
            "5": "nextcloudpi.local",
            "4": "kpleines.spdns.de",
            "3": "nextcloudpi.lan",
            "2": "kpleines.spdns.de",
            "20": "kpleines.spdns.de",
            "21": "lanqnapdockernextcloudpi.fritz.box",
            "22": "192.168.178.3"
        },
        "datadirectory": "***REMOVED SENSITIVE VALUE***",
        "dbtype": "mysql",
        "version": "15.0.4.0",
        "overwrite.cli.url": "https:\/\/kpleines.spdns.de\/",
        "dbname": "***REMOVED SENSITIVE VALUE***",
        "dbhost": "***REMOVED SENSITIVE VALUE***",
        "dbport": "",
        "dbtableprefix": "oc_",
        "mysql.utf8mb4": true,
        "dbuser": "***REMOVED SENSITIVE VALUE***",
[... 100 lines cut ...]
{"reqId":"XG04acCosgMAAABeJ3UAAAAP","level":2,"time":"2019-02-20T11:22:55+00:00","remoteAddr":"109.250.157.251","user":"--","app":"core","method":"PROPFIND","url":"\/remote.php\/dav\/addressbooks\/users\/pleinesk\/SolaMA\/","message":"Login failed: 'pleinesk ' (Remote IP: '109.250.157.251')","userAgent":"Thunderbird CardBook\/34.0 Lightning\/6.2.5","version":"15.0.4.0"}
{"reqId":"XG04acCosgMAAABeJ3kAAAAX","level":2,"time":"2019-02-20T11:22:55+00:00","remoteAddr":"109.250.157.251","user":"--","app":"core","method":"PROPFIND","url":"\/remote.php\/dav\/addressbooks\/users\/pleinesk\/MGMAdressen_shared_by_mgm\/","message":"Login failed: 'pleinesk ' (Remote IP: '109.250.157.251')","userAgent":"Thunderbird CardBook\/34.0 Lightning\/6.2.5","version":"15.0.4.0"}
{"reqId":"XG04acCosgMAAABeJ3gAAAAV","level":2,"time":"2019-02-20T11:23:20+00:00","remoteAddr":"109.250.157.251","user":"--","app":"core","method":"PROPFIND","url":"\/remote.php\/dav\/addressbooks\/users\/pleinesk\/MGMFreunde_shared_by_mgm\/","message":"Login failed: 'pleinesk ' (Remote IP: '109.250.157.251')","userAgent":"Thunderbird CardBook\/34.0 Lightning\/6.2.5","version":"15.0.4.0"}
{"reqId":"XG04acCosgMAAABeJ3cAAAAT","level":2,"time":"2019-02-20T11:23:20+00:00","remoteAddr":"109.250.157.251","user":"--","app":"core","method":"PROPFIND","url":"\/remote.php\/dav\/addressbooks\/users\/pleinesk\/Bva\/","message":"Login failed: 'pleinesk ' (Remote IP: '109.250.157.251')","userAgent":"Thunderbird CardBook\/34.0 Lightning\/6.2.5","version":"15.0.4.0"}
{"reqId":"XG04acCosgMAAABeJ3YAAAAR","level":2,"time":"2019-02-20T11:23:21+00:00","remoteAddr":"109.250.157.251","user":"--","app":"core","method":"PROPFIND","url":"\/remote.php\/dav\/addressbooks\/users\/pleinesk\/Kontakte\/","message":"Login failed: 'pleinesk ' (Remote IP: '109.250.157.251')","userAgent":"Thunderbird CardBook\/34.0 Lightning\/6.2.5","version":"15.0.4.0"}
{"reqId":"XG04acCosgMAAABeJ4AAAAAb","level":2,"time":"2019-02-20T11:24:12+00:00","remoteAddr":"109.250.157.251","user":"--","app":"core","method":"PROPFIND","url":"\/remote.php\/dav\/addressbooks\/users\/pleinesk\/Service\/","message":"Login failed: 'pleinesk ' (Remote IP: '109.250.157.251')","userAgent":"Thunderbird CardBook\/34.0 Lightning\/6.2.5","version":"15.0.4.0"}
{"reqId":"XG043cCosgMAAABf3DgAAAAP","level":2,"time":"2019-02-20T11:25:03+00:00","remoteAddr":"109.250.157.251","user":"--","app":"core","method":"PROPFIND","url":"\/remote.php\/dav\/addressbooks\/users\/pleinesk\/SolaLk\/","message":"Login failed: 'pleinesk ' (Remote IP: '109.250.157.251')","userAgent":"Thunderbird CardBook\/34.0 Lightning\/6.2.5","version":"15.0.4.0"}
{"reqId":"XG043cCosgMAAABf3DkAAAAR","level":2,"time":"2019-02-20T11:25:28+00:00","remoteAddr":"109.250.157.251","user":"--","app":"core","method":"PROPFIND","url":"\/remote.php\/dav\/addressbooks\/users\/pleinesk\/SolaMA\/","message":"Login failed: 'pleinesk ' (Remote IP: '109.250.157.251')","userAgent":"Thunderbird CardBook\/34.0 Lightning\/6.2.5","version":"15.0.4.0"}
{"reqId":"XG043cCosgMAAABf3DwAAAAX","level":2,"time":"2019-02-20T11:25:28+00:00","remoteAddr":"109.250.157.251","user":"--","app":"core","method":"PROPFIND","url":"\/remote.php\/dav\/addressbooks\/users\/pleinesk\/Bva\/","message":"Login failed: 'pleinesk ' (Remote IP: '109.250.157.251')","userAgent":"Thunderbird CardBook\/34.0 Lightning\/6.2.5","version":"15.0.4.0"}
{"reqId":"XG043cCosgMAAABf3DoAAAAT","level":2,"time":"2019-02-20T11:25:53+00:00","remoteAddr":"109.250.157.251","user":"--","app":"core","method":"PROPFIND","url":"\/remote.php\/dav\/addressbooks\/users\/pleinesk\/MGMAdressen_shared_by_mgm\/","message":"Login failed: 'pleinesk ' (Remote IP: '109.250.157.251')","userAgent":"Thunderbird CardBook\/34.0 Lightning\/6.2.5","version":"15.0.4.0"}
{"reqId":"XG043cCosgMAAABf3DsAAAAV","level":2,"time":"2019-02-20T11:25:53+00:00","remoteAddr":"109.250.157.251","user":"--","app":"core","method":"PROPFIND","url":"\/remote.php\/dav\/addressbooks\/users\/pleinesk\/MGMFreunde_shared_by_mgm\/","message":"Login failed: 'pleinesk ' (Remote IP: '109.250.157.251')","userAgent":"Thunderbird CardBook\/34.0 Lightning\/6.2.5","version":"15.0.4.0"}
{"reqId":"XG043cCosgMAAABf3D0AAAAZ","level":2,"time":"2019-02-20T11:25:53+00:00","remoteAddr":"109.250.157.251","user":"--","app":"core","method":"PROPFIND","url":"\/remote.php\/dav\/addressbooks\/users\/pleinesk\/Kontakte\/","message":"Login failed: 'pleinesk ' (Remote IP: '109.250.157.251')","userAgent":"Thunderbird CardBook\/34.0 Lightning\/6.2.5","version":"15.0.4.0"}
{"reqId":"4B0ZQqwvjpRQWVaakeBW","level":3,"time":"2019-02-20T19:15:01+00:00","remoteAddr":"","user":"--","app":"cron","method":"","url":"--","message":{"Exception":"TypeError","Message":"Argument 1 passed to OCA\\GroupFolders\\Versions\\VersionsBackend::getVersionsFolder() must be of the type integer, string given, called in \/data\/nextcloud\/apps\/groupfolders\/lib\/Versions\/VersionsBackend.php on line 154","Code":0,"Trace":[{"file":"\/data\/nextcloud\/apps\/groupfolders\/lib\/Versions\/VersionsBackend.php","line":154,"function":"getVersionsFolder","class":"OCA\\GroupFolders\\Versions\\VersionsBackend","type":"->","args":["1"]},{"file":"\/data\/nextcloud\/apps\/groupfolders\/lib\/Versions\/GroupVersionsExpireManager.php","line":53,"function":"getAllVersionedFiles","class":"OCA\\GroupFolders\\Versions\\VersionsBackend","type":"->","args":[{"id":"1","mount_point":"GMgmLeitung","groups":{"MgmLeitung":"31"},"quota":"1073741274","size":0}]},{"file":"\/data\/nextcloud\/apps\/groupfolders\/lib\/Versions\/GroupVersionsExpireManager.php","line":48,"function":"expireFolder","class":"OCA\\GroupFolders\\Versions\\GroupVersionsExpireManager","type":"->","args":[{"id":"1","mount_point":"GMgmLeitung","groups":{"MgmLeitung":"31"},"quota":"1073741274","size":0}]},{"file":"\/data\/nextcloud\/apps\/groupfolders\/lib\/BackgroundJob\/ExpireGroupVersions.php","line":40,"function":"expireAll","class":"OCA\\GroupFolders\\Versions\\GroupVersionsExpireManager","type":"->","args":[]},{"file":"\/data\/nextcloud\/lib\/private\/BackgroundJob\/Job.php","line":61,"function":"run","class":"OCA\\GroupFolders\\BackgroundJob\\ExpireGroupVersions","type":"->","args":[null]},{"file":"\/data\/nextcloud\/lib\/private\/BackgroundJob\/TimedJob.php","line":55,"function":"execute","class":"OC\\BackgroundJob\\Job","type":"->","args":[{"__class__":"OC\\BackgroundJob\\JobList"},{"__class__":"OC\\Log"}]},{" [... cut ...] s__":"OC\\BackgroundJob\\JobList"},{"__class__":"OC\\Log"}]}],"File":"\/data\/nextcloud\/apps\/groupfolders\/lib\/Versions\/VersionsBackend.php","Line":189,"CustomMessage":"--"},"userAgent":"--","version":"15.0.4.0"}

```
</details>

The Problem is, if I open the NextcloudPi-Web-Panel I see the System-Info correct. Also I can call the links from the upper menu bar.
But if I call any link from the left menu bar e.g. nc-preview or nc-update, the main window stays empty.
Any ideas whats went wrong?

nachoparker, 2019-02-23:
first time I hear this. Do you see any errors in apache logs or in the javascript console?

kpleines, 2019-02-23:
I use Firefox --> Web-Developer --> Debugger-> Console ?

Here the output:
Content Security Policy: Die Einstellungen der Seite haben das Laden einer Ressource auf https://192.168.178.3:4443/core/fonts/OpenSans-Regular.woff blockiert ("default-src").

[image]

nachoparker, 2019-02-23:
yes, that looks ok, how about apache `error.log`?

kpleines, 2019-02-23:
Heres apaches error.log

[error.log](https://github.com/nextcloud/nextcloudpi/files/2897147/error.log)

nachoparker, 2019-02-23:
You have a permissions problem, `www-data` needs to be able to access files in `/usr/local/etc/ncp-config.d/`

It must be something specific with your setup, as it works for everyone else

Cheers!

closed 2019-02-23 by nachoparker