nextcloud/nextcloudpi #815: Blank Webpage after install

Rustymage, 2019-02-25:
Evening all,

I have an issue.

I installed Nextcloudpi using this method

`curl -sSL https://raw.githubusercontent.com/nextcloud/nextcloudpi/master/install.sh | sudo bash
`

It all went well. I then figured out that I needed to enable ncp-web. I can now login but then I get a blank webpage. The tab shows 'NextCloudPi Panel' but that's it. See attached screenshot.

This is running on a RPi3B+ with a 15G SD card (reasonably new) and 1TB HDD. The Pi is running Syncthing. It also has rclone and youtube-dl workloads but these are sporadic.

Please assist - I will try to explain more when you ask questions.

Thanks in advance.

Rustymage

PS: I wonder if it has something to do with the following?

`[Mon Feb 25 22:27:02.515184 2019] [authz_host:error] [pid 4773:tid 1682572336] [client 192.168.1.122:53991] AH01753: access check of 'localhost' to /index.php failed, reason: unable to get the remote host name`

[image]

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v1.8.4
distribution         Raspbian GNU/Linux 9 \n \l
automount            yes
USB devices          sda 
datadir              /var/www/nextcloud/data
data in SD           yes
data filesystem      ext2/ext3
data disk usage      2.8G/15G
rootfs usage         2.8G/15G
swapfile             /var/swap
dbdir                /media/USBdrive/ncdatabase
Nextcloud check      ok
Nextcloud version    15.0.4.0
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               47min

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```
{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
[... 82 lines cut ...]
2019-02-25 22:14:05 1989115696 [Note] InnoDB: Using generic crc32 instructions
2019-02-25 22:14:05 1989115696 [Note] InnoDB: Initializing buffer pool, size = 370.0M
2019-02-25 22:14:05 1989115696 [Note] InnoDB: Completed initialization of buffer pool
2019-02-25 22:14:05 1989115696 [Note] InnoDB: Highest supported file format is Barracuda.
2019-02-25 22:14:06 1989115696 [Note] InnoDB: 128 rollback segment(s) are active.
2019-02-25 22:14:06 1989115696 [Note] InnoDB: Waiting for purge to start
2019-02-25 22:14:06 1989115696 [Note] InnoDB:  Percona XtraDB (http://www.percona.com) 5.6.41-84.1 started; log sequence number 4095989
2019-02-25 22:14:06 1989115696 [Note] Plugin 'FEEDBACK' is disabled.
2019-02-25 22:14:06 1107292992 [Note] InnoDB: Dumping buffer pool(s) not yet started
2019-02-25 22:14:06 1989115696 [Note] Server socket created on IP: '127.0.0.1'.
2019-02-25 22:14:06 1989115696 [Note] /usr/sbin/mysqld: ready for connections.
Version: '10.1.37-MariaDB-0+deb9u1'  socket: '/var/run/mysqld/mysqld.sock'  port: 3306  Raspbian 9.0

```
</details>

<details>
<summary>Nextcloud logs</summary>

```

{"reqId":"dN6yeiwEBulDVHQrUA5n","level":3,"time":"2019-02-25T21:00:02+00:00","remoteAddr":"","user":"--","app":"cron","method":"","url":"--","message":{"Exception":"Exception","Message":"Not installed","Code":0,"Trace":[{"file":"\/var\/www\/nextcloud\/lib\/base.php","line":660,"function":"checkInstalled","class":"OC","type":"::","args":[]},{"file":"\/var\/www\/nextcloud\/lib\/base.php","line":1068,"function":"init","class":"OC","type":"::","args":[]},{"file":"\/var\/www\/nextcloud\/cron.php","line":41,"args":["\/var\/www\/nextcloud\/lib\/base.php"],"function":"require_once"}],"File":"\/var\/www\/nextcloud\/lib\/base.php","Line":277,"CustomMessage":"--"},"userAgent":"--","version":""}

```
</details>

nachoparker, 2019-02-25:
it should be fixed now. Please try again from scratch

closed 2019-02-25 by nachoparker

Rustymage, 2019-02-25:
Thanks - remove and reinstall with the first code insert I posted?

What was the issue? How can I debug myself next time?

:)

Rustymage, 2019-02-26:
Can you also tell me how to remove Nextcloud please.

Rustymage, 2019-02-26:
I've uninstalled mysqld and associated components and it still doesn't work. I'm restoring an old image without NCPi installed to try once more.

Rustymage, 2019-02-26:
[image]

Hmm - is this meant to happen?

Rustymage, 2019-02-27:
All working now. Thanks.

nachoparker, 2019-03-01:
Hi, I was away for a few days. I am glad it works

nachoparker, 2019-03-02:
I am not sure yet, I rolled back the last changes. I will look into it soon