nextcloud/nextcloudpi #816: Constant reload on first open of webui

Judoguy12, 2019-02-26:
When I attempt to load the webui for the first time after install it just constantly reloads and never appears after accepting the browser exception is this a bug?

**System information**

<--! Paste this in GitHub report -->

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v1.8.0
distribution         Debian GNU/Linux 9 \n \l
automount            no
USB devices          none
datadir              /var/www/nextcloud/data
data in SD           yes
data filesystem      ext2/ext3
data disk usage      5.7G/289G
rootfs usage         5.7G/289G
swapfile             /var/swap
dbdir                /var/lib/mysql
Nextcloud check      ok
Nextcloud version    15.0.4.0
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               1:21

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "11": "2a00:23c5:bd2a:7700:a852:ba1a:8dea:7cd1",
            "1": "192.168.1.181",
            "5": "nextcloudpi.local",
            "7": "nextcloudpi",
            "8": "nextcloudpi.lan"
        },
        "datadirectory": "***REMOVED SENSITIVE VALUE***",
        "dbtype": "mysql",
        "version": "15.0.4.0",
        "overwrite.cli.url": "http:\/\/localhost",
        "dbname": "***REMOVED SENSITIVE VALUE***",
        "dbhost": "***REMOVED SENSITIVE VALUE***",
        "dbport": "",
        "dbtableprefix": "oc_",
        "mysql.utf8mb4": true,
        "dbuser": "***REMOVED SENSITIVE VALUE***",
        "dbpassword": "***REMOVED SENSITIVE VALUE***",
        "installed": true,
        "instanceid": "***REMOVED SENSITIVE VALUE***",
        "memcache.local": "\\OC\\Memcache\\Redis",
  "memcache.locking": "\\OC\\Memcache\\Redis",
        "redis": {
            "host": "***REMOVED SENSITIVE VALUE***",
            "port": 0,
            "timeout": 0,
            "password": "***REMOVED SENSITIVE VALUE***"
        },
        "tempdirectory": "\/var\/www\/nextcloud\/data\/tmp",
        "mail_smtpmode": "sendmail",
        "mail_smtpauthtype": "LOGIN",
        "mail_from_address": "***REMOVED SENSITIVE VALUE***",
        "mail_domain": "***REMOVED SENSITIVE VALUE***",
        "overwriteprotocol": "https"
    }
}

```
</details>

<details>
<summary>HTTPd logs</summary>

```

[Tue Feb 26 19:59:56.000345 2019] [mpm_event:notice] [pid 10580:tid 4148238080] AH00489: 
Apache/2.4.25 (Debian) OpenSSL/1.0.2l configured -- resuming normal operations
[Tue Feb 26 19:59:56.000382 2019] [core:notice] [pid 10580:tid 4148238080] AH00094: Command line: '/usr/sbin/apache2'
[Tue Feb 26 20:05:16.380081 2019] [mpm_event:notice] [pid 10580:tid 4148238080] AH00493: SIGUSR1 received.  Doing graceful restart
[Tue Feb 26 20:05:16.394326 2019] [ssl:warn] [pid 10580:tid 4148238080] AH01909: localhost:4443:0 server certificate does NOT include an ID which matches the server name
[Tue Feb 26 20:05:16.394445 2019] [ssl:error] [pid 10580:tid 4148238080] AH02217: ssl_stapling_init_cert: can't retrieve issuer certificate! [subject: CN=raspberry.home / issuer: CN=raspberry.home / serial: 806C4B1EABF83C04 / notbefore: Feb 26 19:38:07 2019 GMT / notafter: Feb 23 19:38:07 2029 GMT]
[Tue Feb 26 20:05:16.394455 2019] [ssl:error] [pid 10580:tid 4148238080] AH02604: Unable to configure certificate localhost:4443:0 for stapling
[Tue Feb 26 20:05:16.394711 2019] [ssl:warn] [pid 10580:tid 4148238080] AH01909: localhost:443:0 server certificate does NOT include an ID which matches the server name
[Tue Feb 26 20:05:16.394802 2019] [ssl:error] [pid 10580:tid 4148238080] AH02217: ssl_stapling_init_cert: can't retrieve issuer certificate! [subject: CN=raspberry.home / issuer: CN=raspberry.home / serial: 806C4B1EABF83C04 / notbefore: Feb 26 19:38:07 2019 GMT / notafter: Feb 23 19:38:07 2029 GMT]
[Tue Feb 26 20:05:16.394826 2019] [ssl:error] [pid 10580:tid 4148238080] AH02604: Unable to configure certificate localhost:443:0 for stapling
[Tue Feb 26 20:05:17.000731 2019] [mpm_event:notice] [pid 10580:tid 4148238080] AH00489: Apache/2.4.25 (Debian) OpenSSL/1.0.2l configured -- resuming normal operations
[Tue Feb 26 20:05:17.000775 2019] [core:notice] [pid 10580:tid 4148238080] AH00094: Command line: '/usr/sbin/apache2'
[Tue Feb 26 20:24:17.976964 2019] [mpm_event:notice] [pid 10580:tid 4148238080] AH00493: 
SIGUSR1 received.  Doing graceful restart
[Tue Feb 26 20:24:17.991218 2019] [ssl:warn] [pid 10580:tid 4148238080] AH01909: localhost:4443:0 server certificate does NOT include an ID which matches the server name
[Tue Feb 26 20:24:17.991335 2019] [ssl:error] [pid 10580:tid 4148238080] AH02217: ssl_stapling_init_cert: can't retrieve issuer certificate! [subject: CN=raspberry.home / issuer: CN=raspberry.home / serial: 806C4B1EABF83C04 / notbefore: Feb 26 19:38:07 2019 GMT / notafter: Feb 23 19:38:07 2029 GMT]
[Tue Feb 26 20:24:17.991346 2019] [ssl:error] [pid 10580:tid 4148238080] AH02604: Unable to configure certificate localhost:4443:0 for stapling
[Tue Feb 26 20:24:17.991603 2019] [ssl:warn] [pid 10580:tid 4148238080] AH01909: localhost:443:0 server certificate does NOT include an ID which matches the server name
[Tue Feb 26 20:24:17.991691 2019] [ssl:error] [pid 10580:tid 4148238080] AH02217: ssl_stapling_init_cert: can't retrieve issuer certificate! [subject: CN=raspberry.home / issuer: CN=raspberry.home / serial: 806C4B1EABF83C04 / notbefore: Feb 26 19:38:07 2019 GMT / notafter: Feb 23 19:38:07 2029 GMT]
[Tue Feb 26 20:24:17.991700 2019] [ssl:error] [pid 10580:tid 4148238080] AH02604: Unable to configure certificate localhost:443:0 for stapling
[Tue Feb 26 20:24:19.001047 2019] [mpm_event:notice] [pid 10580:tid 4148238080] AH00489: Apache/2.4.25 (Debian) OpenSSL/1.0.2l configured -- resuming normal operations
[Tue Feb 26 20:24:19.001078 2019] [core:notice] [pid 10580:tid 4148238080] AH00094: Command line: '/usr/sbin/apache2'

```
</details>

shadowstreik, 2019-02-27:
A link for support is available on the NextCloudPi Mastodon page...

https://mastodon.social/@NextCloudPi

Thank you.

Judoguy12, 2019-02-27:
Fixed i used a different browser safari was playing up

closed 2019-02-27 by Judoguy12

nachoparker, 2019-03-01:
Interesting, I am glad it works now