nextcloud/nextcloudpi #817: NextCloudPi & Reverse Proxy

Rustymage, 2019-02-27:
Hi all,

I have the NCP behind an NGINX reverse proxy raspberry pi. This pi serves up several services in the house and has an SSL cert from Let's Encrypt.

When I request a connection to the NCP from outside the lan I get a URL rewrite that trips up the reverse proxy. 

I type in domain.com/ncp and that get rewritten as domain.com/index.php

This leads me to suspect that the reverse proxy forwards the request but once it's rewritten it searches itself for index.php and fails with the error 404 not found.

What I think I need is it to remain as domain.com/ncp/index.php...

How do I do this? I can find my way around Linux so can help with logs etc but I need a little guidance. I expect I've missed some vital configs you want to view so please shout at me for them!

I should add that the other services behind the proxy work fine.

Thanks in advance!

Rusty

shadowstreik, 2019-02-27:
A link for support is available on the NextCloudPi Mastodon page...

https://mastodon.social/@NextCloudPi

Thank you.

Rustymage, 2019-02-27:
Thanks, I've taken a brief look and can't seem to find anything that's too similar in problem.

My issue is the URL rewrite that trips the proxy up.

Rustymage, 2019-02-27:
Checked again, nothing I can see that can help me. Anyone here got any suggestions? I've performed Google Fu for hours. About to give up...

lucasff, 2019-02-27:
I use subdomains instead of subfolders. Way easier. You can have multiple subdomains pointing to a single CNAME that gets updated with the real public IP. I have a real domain tho (FQDN).

nachoparker, 2019-03-01:
Hello, please use the forums for this kind of support. Github is for bug reports

help.nextcloud.com

closed 2019-03-01 by nachoparker